pyvisa/pyvisa #112: Handle Prologix USB to GPIB

hgrecco, 2015-01-12:
Prologix sells a [GPIB-USB Controller](http://prologix.biz/gpib-usb-controller.html) which does not handle transparently the connections to the instrument (i.e. it is not just a GPIB device as it needs some extra commands).

The proposal is to create a custom Resource that handles deals with the prologix device without requiring extra commands from the user. 

PyVISA will:
- Provide a Resource Class based on `MessageBasedResource` that appears as an standard resource but internally does all the things that prologix needs.
- Provide a prefix to indicate a custom resource name (For example `+`)
- In `open_resource`, dispatch this custom prefix to a function that should instantiate the new ResourceClass and return it.

We should also patch other functions like:
- `resource_info`

The idea is that something like this would just work
```python
import visa
rm = visa.ResourceManager()
inst = rm.open_resource('+PROLOGIX::/dev/bla/bla::GPIB::2')
inst.query('*IDN?')
```

ignamv, 2015-01-13:
With the NI backend, you get a serial port and no way to tell it's an adapter (I checked all the attributes with pyvisa shell). So the user would have to tell the library about this.

hgrecco, 2015-01-13:
Then I think we could do what I suggested before. Open questions from my side: 
- should we subclass or embed the underlying resource
- what is the underlying resource. A USB RAW port or a serial port, or this depend on the OS/Backend?

hgrecco, 2015-01-13:
My suggestion for the resource name is: 

`+<NAME OF THE ADAPTER>::<REAL RESOURCE NAME>+<FAKE RESOURCE NAME>`

For example, talking to a devices in serial port 1 via a prologix adapater that simulates a device in GPIB 2 is:

'+PROLOGIX::ASRL1+GPIB::2`

This is really easy to parse and dispatch. I think that the `+` sign is not a valid character in a resource name

stadi1987, 2015-01-23:
'''
#!/usr/bin/python
# -*- coding: utf-8 -*-

'''
Module description here

Author: Daniel Stadelmann
$Revision: 13700 $
$Date: 2014-12-02 17:02:58 +0100 (Di, 02 Dez 2014) $
Copyright: Comlab AG, CH-3063 Ittigen. All rights reserved. 
'''

import serial, visa

class PrologixInstrument(object):
    '''
    Class for instruments connected via prologix USB-GPIB adapter
    '''
    serFd = None
    lineEnd="\n"
        
    def __init__(self, gpibAddr, comPort, baud_rate=921600, timeout=0.25, silent=True):
        comPort = int(comPort[3:])-1
        self._silent=silent
        self.gpibAddr = gpibAddr
        
        if timeout==None:
            timeout=2
        if PrologixInstrument.serFd==None:
            PrologixInstrument.serFd=serial.Serial(comPort, baudrate=baud_rate, timeout=timeout)
            PrologixInstrument.serFd.write("++mode 1"+self.lineEnd )
            PrologixInstrument.serFd.write("++ifc"+PrologixInstrument.lineEnd )
            #serialInterface.serFd.write("++auto 1"+serialInterface.lineEnd )
            PrologixInstrument.serFd.write("++read_tmo_ms 200"+PrologixInstrument.lineEnd )
            ver = tuple(( int(i) for i in self.ask('++ver').split()[-1].split('.')))
#            print(ver)
            if ver != (6, 107):
                raise(RuntimeError("Prologix is of version: %d.%d\n Please Update or ask Tobias Badertscher." % ver))
        #self.identity=self.ask('*IDN?')
        
    def ask(self,cmd):
        PrologixInstrument.serFd.write('++addr %d' %self.gpibAddr+PrologixInstrument.lineEnd)
        if isinstance(cmd,list):
            for i in cmd:
                if not self._silent:
                    print("Command: \'%s\'" % cmd)
                PrologixInstrument.serFd.write(i+PrologixInstrument.lineEnd)
        else:
            if not self._silent:
                print("Command: \'%s\'" % cmd)
            PrologixInstrument.serFd.write(cmd+PrologixInstrument.lineEnd)
        #serialInterface.serFd.write("++read eoi"+serialInterface.lineEnd)
        PrologixInstrument.serFd.write("++read"+PrologixInstrument.lineEnd)
        retry=1
        while retry>0:
            res=PrologixInstrument.serFd.readlines()
            if len(res)>0:
                if not self._silent:
                    print("Obtained result:")
                retry=0
                if res[0][0]=='#':
                    sizeLen=int(res[0][1])
                    binSize=int(res[0][2:2+sizeLen])
                    oRes=res[0][2+sizeLen:]+res[1]+res[2][0:-1]
                    if binSize!=len(oRes):
                        oRes=None
                    res=oRes
                    if not self._silent:
                        print("Obtained binary data of size %d" % binSize)  
[... 81 lines cut ...]
        '''
        Get device ID
        '''
        return self.ask( '*IDN?' )
        
    def wait( self, timeout=None ):
        '''
        Wait for operation to complete
        ''' 
        if timeout!= None:
            to = self._inst.timeout
            self._inst.timeout = timeout        
            self.ask('*OPC?')
            self._inst.timeout = to
        else:
            self.ask('*OPC?') 
        
    @property
    def timeout(self,):
        return self._inst.timeout
        
    @timeout.setter        
    def timeout(self, timeout):        
        self._inst.timeout = timeout
'''

hgrecco, 2015-01-24:
@stadi1987 your proposal disappeared from the tracker but was interesting to discuss. Feel free to repost it if you feel like doing it.

stadi1987, 2015-01-26:
Sorry I had to discuss with my line manager before I post some code.

Here is what we implemented. The class ScpiInstrumentWrapper is the top level wrapper I use for all SCPI controlled instruments. It instantiates either a VISA instrument or a PrologixInstrument, which is the other class in the module.

```python
#!/usr/bin/python
# -*- coding: utf-8 -*-

'''
Instrument wrapper module

Author: Daniel Stadelmann, Tobias Badertscher
'''

import serial, visa

class PrologixInstrument(object):
    '''
    Class for instruments connected via prologix USB-GPIB adapter
    '''
    serFd = None
    lineEnd="\n"
        
    def __init__(self, gpibAddr, comPort, baud_rate=921600, timeout=0.25, silent=True):
        comPort = int(comPort[3:])-1
        self._silent=silent
        self.gpibAddr = gpibAddr
        
        if timeout==None:
            timeout=2
        if PrologixInstrument.serFd==None:
            PrologixInstrument.serFd=serial.Serial(comPort, baudrate=baud_rate, timeout=timeout)
            PrologixInstrument.serFd.write("++mode 1"+self.lineEnd )
            PrologixInstrument.serFd.write("++ifc"+PrologixInstrument.lineEnd )
            #serialInterface.serFd.write("++auto 1"+serialInterface.lineEnd )
            PrologixInstrument.serFd.write("++read_tmo_ms 200"+PrologixInstrument.lineEnd )
            ver = tuple(( int(i) for i in self.query('++ver').split()[-1].split('.')))
            if ver != (6, 107):
                raise(RuntimeError("Prologix is of version: %d.%d\n Please Update." % ver))
                
    def ask(self, cmd):
        return self.query(cmd)
        
    def query(self,cmd):
        PrologixInstrument.serFd.write('++addr %d' %self.gpibAddr+PrologixInstrument.lineEnd)
        if isinstance(cmd,list):
            for i in cmd:
                if not self._silent:
                    print("Command: \'%s\'" % cmd)
                PrologixInstrument.serFd.write(i+PrologixInstrument.lineEnd)
        else:
            if not self._silent:
                print("Command: \'%s\'" % cmd)
            PrologixInstrument.serFd.write(cmd+PrologixInstrument.lineEnd)
        #serialInterface.serFd.write("++read eoi"+serialInterface.lineEnd)
        PrologixInstrument.serFd.write("++read"+PrologixInstrument.lineEnd)
        retry=1
        while retry>0:
            res=PrologixInstrument.serFd.readlines()
            if len(res)>0:
                if not self._silent:
                    print("Obtained result:")
                retry=0
                if res[0][0]=='#':
                    sizeLen=int(res[0][1])
                    binSize=int(res[0][2:2+sizeLen])
                    oRes=res[0][2+sizeLen:]+res[1]+res[2][0:-1]
                    if binSize!=len(oRes):
                        oRes=None
                    res=oRes
[... 92 lines cut ...]
            self.ask('*OPC?')
            self._inst.timeout = to
        else:
            self.ask('*OPC?') 
        
    @property
    def timeout(self,):
        return self._inst.timeout
        
    @timeout.setter        
    def timeout(self, timeout):        
        self._inst.timeout = timeout
```

Here is an example:
```python
import ScpiInstrumentWrapper

myPrologicInst = ScpiInstrumentWrapper('PROLOGIX::COM12::GPIB::20')
myLanVisaInst = ScpiInstrumentWrapper('TCPIP::192.168.4.103')
print(myPrologicInst.query('*IDN?'))
print(myLanVisaInst.query('*IDN?'))
```

Hope this helps!

hgrecco, 2015-01-26:
It helps a lot. Thanks. @ignamv what do you think?

I do think that my proposal to write an adapter included in PyVISA will be a good thing. Because then any  program or library that uses PyVISA will profit automatically just using the right resource name.

ignamv, 2015-01-27:
That's great, I'll give it a shot with my adapter.

roamingryan, 2015-05-29:
The code @stadi1987 posted is a very useful starting point.  One immediate limitation is the lack of support for multiple instruments downstream from a single Prologix adapter.  Migration of the idea to a "proper" PyVISA backend should solve this though.  Has anyone made progress on this front?

ignamv, 2015-05-29:
Unfortunately I no longer have access to the adapter so I can't test the
code.

Thanks for the help.

stadi1987, 2015-05-29:
@roamingryan: You are right. Actually I never used this adapter with more than one instrument. We also have the expensive NI adapters for that purpose. We control all new instruments over LAN and LXI which is much cheaper :)
However I think to control multiple instruments, it is necessary to instantiate the serial.Serial object in a higher level s.t. it can be used for all instruments. E.g. the pyvisa resource manager may be the right place for that?
Another possibility could be to provide the Serial object to the wrapper as an optional parameter. if it is not provided, it is created. Otherwise the already created serial object is forwarded to the Prologix instrument class. Maybe I can provide the some code next week. Feel free to try it yourself :)

stadi1987, 2015-05-30:
I checked the code and actually it would support multiple instruments but there is an indentation bug. The serial.Serial object is now created globally and reused, if another instrument is instantiated.
Here is the corrected and verified code:

```python
#!/usr/bin/python
# -*- coding: utf-8 -*-

'''
Instrument wrapper module

Author: Daniel Stadelmann, Tobias Badertscher
$Revision: 16330 $
$Date: 2015-05-18 21:39:30 +0200 (Mo, 18 Mai 2015) $
Copyright: Comlab AG, CH-3063 Ittigen. All rights reserved. 
'''

import serial, visa, string

class PrologixInstrument(object):
    '''
    Class for instruments connected via prologix USB-GPIB adapter
    '''
    serFd = None
    lineEnd="\n"
        
    def __init__(self, gpibAddr, comPort, baud_rate=921600, timeout=0.25, silent=True):
        comPort = int(comPort[3:])-1
        self._silent=silent
        self.gpibAddr = gpibAddr
        
        if timeout==None:
            timeout=2
        if PrologixInstrument.serFd==None:
            PrologixInstrument.serFd=serial.Serial(comPort, baudrate=baud_rate, timeout=timeout)
        PrologixInstrument.serFd.write("++mode 1"+self.lineEnd )
        PrologixInstrument.serFd.write("++ifc"+PrologixInstrument.lineEnd )
        #serialInterface.serFd.write("++auto 1"+serialInterface.lineEnd )
        PrologixInstrument.serFd.write("++read_tmo_ms 200"+PrologixInstrument.lineEnd )
        ver = tuple(( int(i) for i in self.query('++ver').split()[-1].split('.')))
        if ver != (6, 107):
            raise(RuntimeError("Prologix is of version: %d.%d\n Please Update." % ver))
                
    def ask(self, cmd):
        return self.query(cmd)
        
    def query(self,cmd):
        PrologixInstrument.serFd.write('++addr %d' %self.gpibAddr+PrologixInstrument.lineEnd)
        if isinstance(cmd,list):
            for i in cmd:
                if not self._silent:
                    print("Command: \'%s\'" % cmd)
                PrologixInstrument.serFd.write(i+PrologixInstrument.lineEnd)
        else:
            if not self._silent:
                print("Command: \'%s\'" % cmd)
            PrologixInstrument.serFd.write(cmd+PrologixInstrument.lineEnd)
        #serialInterface.serFd.write("++read eoi"+serialInterface.lineEnd)
        PrologixInstrument.serFd.write("++read"+PrologixInstrument.lineEnd)
        retry=1
        while retry>0:
            res=PrologixInstrument.serFd.readlines()
            if len(res)>0:
                if not self._silent:
                    print("Obtained result:")
                retry=0
                if res[0][0]=='#':
                    sizeLen=int(res[0][1])
                    binSize=int(res[0][2:2+sizeLen])
                    oRes=res[0][2+sizeLen:]+res[1]+res[2][0:-1]
                    if binSize!=len(oRes):
[... 102 lines cut ...]
        
    def wait( self, timeout=None ):
        '''
        Wait for operation to complete
        ''' 
        if timeout!= None:
            to = self._inst.timeout
            self._inst.timeout = timeout        
            self.ask('*OPC?')
            self._inst.timeout = to
        else:
            self.ask('*OPC?') 
        
    @property
    def timeout(self,):
        return self._inst.timeout
        
    @timeout.setter        
    def timeout(self, timeout):        
        self._inst.timeout = timeout
```

Please try it out and let me know if this works for you.

Best regards

murtyns, 2015-07-10:
If I am using a prologix usb-gpib adapter, how do I then access the devices on the gpib bus?

Is there a sample program that you can share?

Murty

soundsc, 2015-11-10:
I have just tested the code posted above by stadi1987 on May 30 with a Prologix GPIB-USB v6.101 controller on Debian 8, successfully.

On Linux, the virtual com port (VCP) is at `/dev/ttyUSBx`, so the line `comPort = int(comPort[3:])-1` needs to be commented out in this case. The user needs to be in the `dialout` group to access the VCP. Finally, you may have to modify the version check so it doesn't throw an exception.

Working code:

```python
import stadi1987prologix
inst = stadi1987prologix.ScpiInstrumentWrapper('PROLOGIX::/dev/ttyUSB0::GPIB::1')
print inst.query('*IDN?')
```

(which results in the instrument returning identifying information)

My question to the pyivsa developers is whether you are interested in supporting this as part of pyvisa. One reason against would be that the interface is VCP and not actually VISA. One reason for would be that people who use this often the same people who use pyvisa (like myself) and might appreciate being able to access it via the same interface. What do you think?

SinghNidhi, 2015-12-29:
Hi,

I am using Prologix GPIB-USB adapter to connect SR810 lockin amplifier to my PC. As mentioned in the post the adapter appears with a Resource name as '+PROLOGIX::ASRL1+GPIB::2' but when I connect my adapter and locate it in NI MAX all I can see is 'ASRL5::INSTR' as a COM port.

I am facing problems in reading data from SR810, can there be a problem cuz my port is appearing as a COM port and not as GPIB-USB adapter?  What could be the probable reason of my adapter appearing as a COM port?

stadi1987, 2015-12-30:
@SinghNidhi: The prologix adapter is not supported by VISA, so it will never appear in NI-MAX. This thread is about a self-made interface for pyvisa (python) which makes it possible to use a device connected via prologix adapter the same way as a true VISA resource. The prefix "PROLOGIX::" is not a real VISA resource name. The prologix adapter offers nothing more than a COM-port. With the corresponding prefixes to the instrument commands, the parameters (like GPIB address)of the adapter can be changed. This implementation (see above) will only work if you control your instrument using python (you need VISA, pyvisa and the classes posted above). If you linke to control it with a dedicated application, it will be very difficult as they may only support true VISA resources. To implement some code in other languages, the above example will help you.

mbirthelmer, 2015-12-30:
All, is anyone working on incorporating ProLogix into pyvisa natively? If not, I might take a crack at it, but I might need a bit of direction as to what the final design should look like. I have a Prologix USB device in hand (and access to an ethernet one, if that's of interest) and an HP 3488 with GPIB interface.

hgrecco, 2016-01-06:
I will be happy to merge something in the direction of the first post in this thread

FTdx401, 2016-02-04:
I've found this thread a few times via Google (from different searches) it definitely looks like adding Prologix support into PyVISA is of interest to some people and from the comments above looks pretty feasible.

I'd be interested in trying to lend a hand as well.  I say "trying" as I'm just starting to learn python (and git).

I do have a Prologix USB-GPIB dongle and a few GPIB instruments here I'd like to control.  I can certainly act as tester.  I'll see if I can get everything installed on my machine as a starter.

greysAcademicCode, 2016-09-15:
I've got a PROLOGIX GPIB-ETHERNET controller and I'd very much like to see support for it here.
How much of what's in this issue applies to the ethernet flavor of the prologix adapter?
I'm happy to help any way I can here (especially testing)!

cweickhmann, 2017-09-06:
Hi everyone!
This thread has been idle for a year now. Is there any development going on?
I have just got a Prologix GPIB-ETHERNET adapter and managed to get it to work using `pyvisa-py`, but it is only for one instrument.
Do I would be happy to see some features implemented. I'm happy to contribute.
Just out of curiosity, @greysAcademicCode, have you had the chance to play around with it?

greyltc, 2017-09-07:
Nope! But I do still have a Prologix GPIB-ETHERNET adapter sitting around somewhere.

tianshiz, 2017-09-25:
@cweickhmann can you elaborate a bit on how you got your GPIB-Ethernet adapter working with pyvisa-py?

Did you just modify the code from the earlier discussion to work with TCP instead of serial?

Thanks

cweickhmann, 2017-09-27:
@tianshiz , it is not straight-forward, but easier than I thought.

You open the connection using a SOCKET. The port to use is 1234 according to documentation (unless changed). There, you can select the GPIB address to talk to and configure things like automatic read.

Here's an example:
```python
import visa
rm = visa.ResourceManager()
gpib = rm.open_resource("TCPIP::192.168.0.123::1234::SOCKET") # Or whatever IP you have
gpib.write("++addr 9\n") # if GPIB device's number is 9
gpib.write("++auto 0\n") # if you want to make the device "listen" (i.e. accept commands); use value 1 instead to make it "talk" (i.e. request a read automatically after write)
gpib.write("++eoi 1\n") # Check docs for details
gpib.write("++eos 3\n") # Check docs for line ending behaviour
# ++eot_enable and ++eot_char allow automatic adding of line ending characters on write
```

If your code above is set correctly for your device, you can query it. E.g.
```python
# ... see above
gpib.query("*IDN?")
>>> "ANRITSU,37297C"
```

I did not have time yet to fully understand how the "++" commands could be hidden in a pyVISA-ic way. Someone suggested to have something like a "PROLOGIX::XXX::GPIBn" identifier. Maybe someone can help me understand where I have to dig in the code?

zamora18, 2018-09-19:
It seems that there is a lot of interest in having pyvisa functional with Prologix GPIB controllers, and maybe there has been some development already made (since the original issue began 3 years ago and the thread has been idle for ~1 year). 
I recently acquired a GPIB-Ethernet controller and would also be willing to contribute and test. 

I came across a [Python wrapper for the Prologix GPIB-to-Ethernet adapter](https://github.com/nelsond/prologix-gpib-ethernet) already implemented. Maybe there can be some collaboration between the developers? Again, I am willing to contribute what I can, but this would be awesome to have integrated into pyvisa (pyvisa-py).

MatthieuDartiailh, 2018-09-19:
Indeed but since I do not have such an adapter and limited time I never looked into it.
I believe the suggestions made by @hgrecco in the beginning of that thread are still valid and I would be happy to provide feedback on a PR moving in that direction.

cweickhmann, 2019-03-19:
Here's a little follow up on the Prologix GPIBoE adapter.
We've been using it in the company quite regularly now. I have recently got rid of the imho most annoying shortcoming: The lack of native read after write functionality.
To circumvent this, I've created a little class `Prologix` which basically encapsulates the `TCPIPSocket` class of pyvisa and extends the queries with a small decorator:

```python
class Prologix(visa.resources.TCPIPSocket):
    """
    Encapsulating class providing write("++auto 1") before and
    write("++auto 0") after queries. Every TCPIPSocket.query*() method
    is wrapped in a decorator implementing this functionality.
    Furthermore, configure() and a setGpibAddr() methods are added
    to work around limitations of the Prologix controller.
    """
    
    @decorator.decorator
    def AutoFetchResponse(fun, self, *args, **kargs):
        self.write("++auto 1")
        resp = fun(self, *args, **kargs)
        self.write("++auto 0")
        return resp
    
    def __init__(self, *args, **kargs):
        super().__init__(*args, **kargs)
    
    def configure(self, vna_gpib_address=6):
        """Set up the Prologix controller to use:
            * set Prologix into Controller mode (++mode 1)
            * not automatic read after write (++auto 0)
            * read timeout 500 ms (++read_tmo_ms 500)
            * enable EOI assertion (++eoi 1)
            * termination character "\n" (++eos 2)
            * Disable character appending (++eot_enable 0)
            * Set device GPIB address to `vna_gpib_address`, defaults to 6
        """
        self.write_termination = "\n"
        self.read_termination = "\n"
        self.write("++mode 1")
        self.write("++addr %d" % vna_gpib_address)
        self.write("++auto 0") # No automatic read after write!
        self.write("++read_tmo_ms 500")
        self.write("++eos 2")
        self.write("++eoi 1")
        self.write("++eot_enable 0")
    
    def setGpibAddr(self, vna_gpib_address=6):
        """Set GPIB address. Defaults to 6"""
        self.write("++addr %d" % vna_gpib_address)
    
    @AutoFetchResponse
    def query(self, message, delay=None):
        return super().query(message, delay=delay)
    
    @AutoFetchResponse
    def query_ascii_values(self, message, converter='f', separator=',',
                           container=list, delay=None):
        return super().query_ascii_values(message, converter=converter,
                                          separator=separator,
                                          container=container,
                                          delay=delay)
    
    @AutoFetchResponse
    def query_binary_values(self, message, datatype='f', is_big_endian=False,
                            container=list, delay=None, header_fmt='ieee',
                            expect_termination=True):
        return super().query_binary_values(message, datatype=datatype,
                                           is_big_endian=is_big_endian,
                                           container=container,
                                           delay=delay, header_fmt=header_fmt,
                                           expect_termination=expect_termination)
    
    @AutoFetchResponse
    def query_values(self, message, delay=None):
        return super().query_values(message, delay=delay)
```

That's certainly not perfect, but it is practical and working. The only thing that is still ugly now is how to use `pyvisa.ResourceManager` with it (but a lot better than before):

```python
import pyvisa
import vnautil.Prologix

rm = pyvisa.ResourceManager()
vna = rm.open_resource("TCPIP::192.168.168.24::1234::SOCKET", resource_pyclass=vnautil.Prologix)
vna.configure(vna_gpib_address=9)
vna.query("*IDN?")
```

The `resource_pyclass` parameter is used to override the default class for this resource locator (TCPIP::___::SOCKET is handled with TCPIPSocket).

Hope this helps.

greyltc, 2019-03-27:
@cweickhmann This looks great! I'll test it out with my Prologix GPIB-ETHERNET CONTROLLER

cweickhmann, 2019-04-11:
@greyltc , let me know if it worked for you.

rprr, 2019-05-03:
@cweickhmann -- I tried to setup using your python file but was unable to get it to work. I first saved your initial class into a file called vnautil.py

When I try to import vnautil.Prologix, I get name 'visa' is not defined. 

```
>>> import pyvisa
>>> import vnautil.Prologix
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/private/tmp/prologixtest/vnautil.py", line 1, in <module>
    class Prologix(visa.resources.TCPIPSocket):
NameError: name 'visa' is not defined
```

So I added a line to the first class file.
```
import pyvisa as visa
```
But when I do import vnautil.Prologix get the error
```
>>> import vnautil.Prologix
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/private/tmp/prologixtest/vnautil.py", line 3, in <module>
    class Prologix(visa.resources.TCPIPSocket):
  File "/private/tmp/prologixtest/vnautil.py", line 12, in Prologix
    @decorator.decorator
NameError: name 'decorator' is not defined
```

Not sure what is going on. Thanks for your help. I'm using python3.

rprr, 2019-05-03:
> @tianshiz , it is not straight-forward, but easier than I thought.
>
> You open the connection using a SOCKET. The port to use is 1234 according to [...]

I tried to use your code with a spectrum analyzer HP8562E but no luck. The address is correct as I set it over telnet. I got a timeout.

cweickhmann, 2019-05-05:
@tianshiz, I think the issue is with the import statement: It should read indeed
```python
import pyvisa as visa
```
for my code. Sorry.

You should install the `decorator` module. https://pypi.org/project/decorator/
This allows wrapping up the annoying `write("++auto 1")` (before) and `write("++auto 0")` (after) statements around all `query...()` calls.

rprr, 2019-05-06:
> @tianshiz, I think the issue is with the import statement: It should read indeed
>
> ```python
> import pyvisa as visa
> ```
>
> for my code. Sorry.
>
> You should install the `decorator` module. [...]

Many thanks. Now I have gotten your test code to work. To your class definition code, I added the following.
```
import pyvisa as visa
import decorator
```

paulgoulain, 2021-03-12:
Hi, I'd like to reopen the discussion about the prologix GPIB-USB Controller. I got one recently and when I use list_resources, it only appears as a a COM device. What would be needed to be developed in order to make it appear the way @hgrecco proposed ? I may try to tackle this problem but I'll need some guidance from people that are familiar with the internals of Pyvisa.
Cheers

cweickhmann, 2021-03-12:
Well, @paulgoulain, that's intended behaviour.
Under Windows, your GPIB-USB adapter will get a COM port assigned (e.g. COM9). The resource may be named ASRL*.
Use `rm.list_resources()` to get a list.

```python
import pyvisa as visa
rm = visa.ResourceManager()
print(rm.list_resources())
```

Then
```python
gpib = rm.open_resource("ASRL9") # Or whatever yours is called.
```

See also the docs:
* https://pyvisa.readthedocs.io/en/1.8/tutorial.html
* https://pyvisa.readthedocs.io/en/1.8/names.html

paulgoulain, 2021-03-12:
Ok, I can understand that, but wouldn't it be better if it appears with a specific name like PROLOGIX::* ? Like the OP proposed ?

cweickhmann, 2021-03-15:
True that.
I honestly only have a vague idea how to implement that. Maybe someone can give me a hint if I'm on the right track...

It probably needs to be "registered", so to speak, somehow like all the other classes in pyvisa/pyvisa/rname.py. Something along the lines of:
```python
@register_subclass
@dataclass
class PrologixTCPIPInstr(ResourceName):
    """Prologix TCPIP INSTR
    The syntax is:
    PROLOGIXTCPIP::host address[::LAN device name][:Port][::INSTR]
    # Does it need "INSTR" though?
    """

    #: Board to use.
    board: str = "0"

    #: Host address of the device (IPv4 or host name)
    host_address: str = ""

    #: LAN device name of the device
    lan_device_name: str = "inst0"

    interface_type: ClassVar[str] = "PROLOGIXTCPIP"
    resource_class: ClassVar[str] = "INSTR"
    is_rc_optional: ClassVar[bool] = True
```

But where to put the actual handling code: no idea yet.

paulgoulain, 2021-03-15:
Also, the Prologix interface is basically a serial interface. One communicates with a GPIB instrument sending serial commands to the prologix interface, that will interpet it and translate it for the GPIB instrument. Can't we use as a starting point how the serial communication is implemented and build upon that ?

cweickhmann, 2021-03-16:
Sure, that's what I did (for the GPIB-Ethernet adapter). In your case it's a GPIB-USB adapter.
The main issue is with configuring the adapter accordingly. I've got boilerplate code for that. But that would go into whatever kind of extension that is available under this `PROLOGIX` prefix.

paulgoulain, 2021-03-22:
So, what do we need to modify/implement if we want to make pyvisa to handle seamlessly a prologix adapter ?

mbielman, 2021-08-24:
Reading (and now following) this with great interest. Have been using the USB-GPIB Prologix for quite some time, communicating via pySerial. That works fine but wanted to try pyVISA. I managed to get that working with a caveat: things do not work well at all when I configure the adapter with ++auto = 0 and ++eos = 3, the read with read eoi. (I have some old gear in my test rack and some items do NOT like the auto replay 'feature' of the adapter!) This scheme works with pySerial but NOT with pyVISA. The query is returned as normal BUT the next query returns the same string from the previous query. Have tried all kinds of stuff w/o success. 

Code is pretty simple:
```
# open via Resource Manager and configure the Prologix
inst.write ('++auto 0\n')
inst.write ('++eos 3\n')
```

then...
```
# send the address
inst.write (query + '\n')
inst.write ('++read eoi\n')
reply = inst.read()
```

So 1st time, say query = '*idn?', the reply is correct.
But the next time (query = something else), the reply is the ID string again.

As mentioned, all works fine with pySerial but NOT with pyVISA.

Thoughts appreciated!

BehnaamRangraz, 2024-08-19:
I hope this message finds you well after 9 years. Many thanks to @hgrecco, @stadi1987 and others who contributed to the content of this issue.

I intend to use the code developed by Daniel Stadelmann and Tobias Badertscher (Revision: 16330) on Ubuntu. First, I produced a list of my serial ports,

>>> import serial
>>> ports = serial.tools.list_ports.comports();
>>> print(str(ports[0]))
/dev/ttyS4 - n/a
>>> print(str(ports[1]))
/dev/ttyS0 - ttyS0
>>> print(str(ports[2]))
/dev/ttyUSB1 - Prologix GPIB-USB Controller - Prologix GPIB-USB Controller

Then, I modified the second line #13 of the code shared by @stadi1987 from,

`import serial, visa, string`

to,

```
import serial, string
import pyvisa as visa
```

Lastly, I executed a modified version of the example code shared by @stadi1987 ,

```
import stadi1987prologix
myPrologicInst = stadi1987prologix.ScpiInstrumentWrapper('PROLOGIX::/dev/ttyUSB1::GPIB::0')
```

Unsurprisingly, I got an error message,

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/behnaam/Desktop/stadi1987prologix.py", line 116, in __init__
    self._inst = PrologixInstrument(gpibAddr,comport)
  File "/home/behnaam/Desktop/stadi1987prologix.py", line 24, in __init__
    comPort = int(comPort[3:])-1
ValueError: invalid literal for int() with base 10: 'v/ttyUSB1'

I'm a confused about multiple things. For example, in '::GPIB::0' how do people come up with the number 0?

How should I correctly modify the code to be usable on Linux where ports are not listed in COMxx format?

Guys, I'm a newbie to this, please kindly help.

bobmcnamara, 2024-11-30:
It looks like this issue has been festering for several years without any resolution.

I'm thinking about adding support for similar GPIB adapters and I'd like to get some feedback on my approach.  The idea is that you would first open a connection to the adapter and in the process associate it with a specific GPIB "board".  After that a connection to a particular instrument need only specify the GPIB "board" and the instrument's GPIB address.

For example:

    import visa
    
    rm = visa.ResourceManager("@py")
    
    adapter = rm.open_resource("GPIB0::usb-prlgx:ASRL1::INTFC")
    
    inst = rm.open_resource("GPIB0::2")
    inst.query("*IDN?")

For the GPIB-ETHERNET device, just change the resource string for the interface such as:

    adapter = rm.open_resource("GPIB0::tcpip-prlgx:10.0.0.13::INTFC")

It seems to me that this is a cleaner way to support a variety of GPIB adapters, Prologix being just one.  And opening a direct
connection to the adapter provides a mechanism for out-of-band communication, if for no other reason than it might prove useful for debugging.

The way I'm thinking about implementing this, it can coexist with linux-gpib and gpib_ctypes.  If no adapter is specified in the GPIB::INTFC resource string, use the existing class.  Likewise if the board number in the GPIB::INSTR resource string hasn't been previously associated with an adapter, use the existing class.

Comments anyone?

cweickhmann, 2024-11-30:
I like the idea. If I understand it correctly, it would also lend itself to a USB GPIB adapter.
It not depending on Prologix to work is a strong point, imho. I think we will see more GPIB adapters based on an approach like Prologix'.

Would that "GPIB0::2" entry appear automagically? Would it be enumeratable/listed? Good naming of the entries would be really helpful (I find the general naming conventions in VISA are already quite confusing - though that's absolutely not pyVISA's fault).

bobmcnamara, 2024-12-02:
It might be possible to identify USB Prologix controllers by enumerating all USB devices, then inspecting the manufacturer and product strings.

I'm less optimistic about being able to identify a device on the local LAN unless it broadcasts its presence.  I don't know if the Prologix device does that.

And at least with the Prologix controllers, finding which GPIB addresses are being used probably isn't reasonable.  I think you would have to try sending a "*IDN?" query to each of the 31 possible addresses and wait for a response.  As far as I can tell, the only error indication the Prologix devices give is a lack of response.

MatthieuDartiailh, 2024-12-02:
Feel free to pick this topic. I do not own such an adapter which makes it completely unfeasible for me to address this issue.

I have however some comments:
- The format you are considering to access the INTFC is not valid from a VISA point of view and will fail to parse. I am not sure how you can go around that.
- Due to how prologix adapters work, I think you will need a different resource class to handle the part of the message that are addressed to  the adapter. I am not sure you can do it using the mentioned approach. Did you consider making a prologix backend ? In this way it would be fairly easy to plug-in custom resources ? Or defining a custom resource name to access prologix instruments ?

cweickhmann, 2024-12-02:
> It might be possible to identify USB Prologix controllers by enumerating all USB devices, then inspecting the manufacturer and product strings.

Oh, that's not what I was after. Sorry, if that came across wrong.
Usually (as in: always ^^) a Prologix Controller only has one Eth or USB interface. No need for enumeration other than what the system gives you, imho.

But why is it `"GPIB0::usb-prlgx:ASRL1::INTFC"` in adapter and then `"GPIB0::2"` for the interface?

The way I set up communication was to have the "raw" interface first, configure it and then handle it as if it was a proper GPIB interface. So, shouldn't it be something like `"usb-prlgx:ASRL1::INTFC"` and once that succeeded you'll have `"GPIB0"` and thus devices behind it available?

MatthieuDartiailh, 2024-12-05:
Thanks for the update @bobmcnamara .

This approach seems interesting (even though it will be limited to the pyvisa-py backend). We could maybe make it work in a more general if we moved away from relying on the VISA implementation to parse resource name. But lets see first what you achieve before thinking in that direction.

bobmcnamara, 2024-12-09:
Status update: the "Prologix compatible" adapter I have is rather buggy.  I'm not sure how other people have been able to use it.

At first, I was unable to use it with "++auto 0"; in that mode, "++read" would read the response from the instrument but not return anything on either the socket or the serial line.  A new firmware update fixed that problem, but (at least with a TCP/IP connection) the adapter assumes that anything arriving in the same packet is a single command.  So for instance if the packet contains "*idn?\n++read\n", the entire string gets sent to the instrument.  For now I am working around this with a 100ms delay before sending any "++" command.

A problem I can't work around is data corruption.  If the instrument response is longer than 100 bytes, the first 99 bytes are correct, then 150 or more bytes are missing, then there are another 99 correct bytes, followed by more missing bytes, and so on.  For a long transfer, what you get are slivers of good data with 99 bytes in each sliver and lots of missing bytes in between the slivers of good data.

Too bad I no longer have access to an actual Prologix device...

bobmcnamara, 2024-12-14:
Several firmware revisions from the vendor later, it's finally mostly working on both the USB and ethernet interfaces!  I still have to put explicit time.sleep() calls before every write, but I'll work with the vendor next week and hopefully resolve that.

Here's a small example using the USB interface:

```
import time
import pyvisa as visa

rm = visa.ResourceManager()

prlgx = rm.open_resource("PRLGX-ASRL::/dev/cu.usbserial-KISS488_1236::INTFC")
ena = rm.open_resource("GPIB::17::INSTR")
ena.write_termination = "\n"

print(ena.query("syst:err?"))
print(ena.query("*idn?"))
```

I stumbled across a somewhat thorny issue while testing, having to do with read timeouts.  Because there is no out-of-band communication channel, a timeout after sending the "++read" command to the adapter is signaled by the string "Timed Out\r\n" being returned.  I don't know if the Prologix device has exactly the same behavior, but I expect it does something similar.

I'll need to study this some more before I try to implement a solution.

mattjaffe42, 2024-12-18:
@bobmcnamara do you have a branch for this? i have a prologix ethernet device and am very interested in this implementation. I could do some tests if that would be helpful.

greyltc, 2024-12-18:
@mattjaffe42 I think bob's talking about some other 3rd party GPIB adapter that's not from Prologix.

mattjaffe42, 2024-12-18:
yeah, bob mentioned that he didn't know if the Prologix device had exactly the same behavior, so i was offering to test.

bobmcnamara, 2024-12-18:
@mattjaffe42 It would be wonderful if you were willing and able to be an alpha-tester with an actual Prologix device.  The 3rd party device I have available has been incredibly buggy.  And the firmware revisions to address the data corruption still drop data and then eventually crash.

Here are the repositories/branches to test:

git@github.com:bobmcnamara/pyvisa.git (branch prlgx_intfc)
git@github.com:bobmcnamara/pyvisa-py.git (branch read_timeout)

I welcome any feedback you can provide!

mattjaffe42, 2024-12-19:
great! i tried this out on a windows device and it almost, but doesn't quite work. I have two devices on a GPIB bus (a SRS DS335 fxn gen and a Fluke 8840A multimeter) and am just trying to make subsequent reads between them.

if i try a more manual approach based on @cweickhmann's decorator approach above from 2019, it works well:
```
import pyvisa
from time import sleep
import decorator

class Prologix(pyvisa.resources.TCPIPSocket):
    """
    Encapsulating class providing write("++auto 1") before and
    write("++auto 0") after queries. Every TCPIPSocket.query*() method
    is wrapped in a decorator implementing this functionality.
    Furthermore, configure() and a setGpibAddr() methods are added
    to work around limitations of the Prologix controller.
    """
    
    @decorator.decorator
    def AutoFetchResponse(fun, self, *args, **kargs):
        self.write("++auto 1")
        resp = fun(self, *args, **kargs)
        self.write("++auto 0")
        return resp
    
    def __init__(self, *args, **kargs):
        super().__init__(*args, **kargs)
    
    def configure(self, device_gpib_address):
        """Set up the Prologix controller to use:
            * set Prologix into Controller mode (++mode 1)
            * not automatic read after write (++auto 0)
            * read timeout 500 ms (++read_tmo_ms 500)
            * enable EOI assertion (++eoi 1)
            * termination character "\n" (++eos 2)
            * Disable character appending (++eot_enable 0)
            * Set device GPIB address to `vna_gpib_address`
        """
        self.write_termination = "\r\n"
        self.read_termination = "\n"
        self.write("++mode 1")
        self.write(f"++addr {device_gpib_address}")
        self.write("++auto 0") # No automatic read after write!
        self.write("++read_tmo_ms 500")
        self.write("++eos 2")
        self.write("++eoi 1")
        self.write("++eot_enable 0")
    
    def setGpibAddr(self, device_gpib_address):
        """Set GPIB address."""
        self.write(f"++addr {device_gpib_address}")
    
    @AutoFetchResponse
    def query(self, message, delay=None):
        return super().query(message, delay=delay)

rm = pyvisa.ResourceManager("@py")
prlgx = rm.open_resource("TCPIP0::172.24.176.53::1234::SOCKET", resource_pyclass=Prologix)
srsDS335_gpib_addr   = 22
fluke8840a_gpib_addr = 5

nReps = 10
for jj in range(nReps):
    prlgx.configure(device_gpib_address=srsDS335_gpib_addr)
    print(prlgx.query("*IDN?"))
    prlgx.setGpibAddr(device_gpib_address=fluke8840a_gpib_addr)
    print(prlgx.query("?"))
    
prlgx.close()
```

Using your branches, communication with the DS335 seems to work fine, but mostly doesn't work with the 8840A. Occasionally the query does successfully return after a long time (setting the timeout to 10 s), but it mostly doesn't work.

The @cweickhmann-based code block above does work fine with the Fluke 8840, as does the more standard `pyvisa` approach with a NI GPIB-USB-HS converter. I wonder if the difference between the two is the `AutoFetchResponse` decorator setting and unsetting auto-reply around queries? I poked around a little bit in your branch, but didn't understand quite well enough to try implementing that myself; it'd take me more time (and probably also more software skills) to figure out the structure of `Sessions`, VISA, etc., none of which am I super familiar with.

The code block using @bobmcnamara's branches is below:

```
import time
import pyvisa

rm = pyvisa.ResourceManager("@py")

prlgx = rm.open_resource("PRLGX-TCPIP0::172.24.176.53::INTFC", timeout=10000)

srsDS335 = rm.open_resource("GPIB0::22")
Fluke8840A = rm.open_resource("GPIB0::5")

for jj in range(2):
    print(srsDS335.query("*IDN?"))
    print(srsDS335.query("FREQ?"))
    srsDS335.write("FREQ 2000000.00")
    print(srsDS335.query("FREQ?"))
    srsDS335.write("FREQ 1000000.00")
    
    print(Fluke8840A.query("?"))

srsDS335.close()
Fluke8840A.close()
rm.close()
 ```

Even though they don't time out, the SRS responses also seem much slower than with the other two approaches. Eventually though, something (usually, but not always, the Fluke, and on most of its queries) times out:

```
VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

Let me know if this is helpful, if you can think of a way to implement an analog of the `AutoFetchResponse` decorator in your branch, or if you have any other ideas of things you'd like me to try! I do also have things set up on a linux system that i haven't tested your branch on yet, so i can try that too later when i get a bit more time.

mattjaffe42, 2024-12-20:
i see the same behavior on linux after testing there too

bobmcnamara, 2024-12-21:
@mattjaffe42 I've added monitor.py to my fork of the pyvisa-py repository which should help to debug why there's a difference in behavior between my code and the approach using @cweickhmann's decorator.

To use it, do this in a separate Python instance:

```
from monitor import proxy
proxy("<ipaddr of Prologix>")
```

Then in the call to open_resource, use "localhost" instead of the IP address of the Prologix adapter.

I'm hoping that it will be obvious what's going on once we compare the results of the two approaches.

mattjaffe42, 2024-12-23:
both my windows and linux systems hang on line 27 of monitor.py:
```rd, _wr, _ex = select.select([server], [], [])```

what system are you using? any suggestions?

bobmcnamara, 2024-12-23:
The "++read" should have been "++read eoi".  (The "compatible" device I was testing with wouldn't accept "++read eoi".)

I've checked in that change to the repository.  Hopefully that explains what you were seeing.  (Without the "eoi", the adapter will read until timeout.)

mattjaffe42, 2024-12-23:
it works now!!! at least with the 2-device test setup i was using up to here.

[10 more replies]