pyvisa/pyvisa #115: Power Sensor Support (Probably not Issue)

ewal31, 2015-01-23:
I am trying to use a Rohde and Schwarz USB Power Sensor.

 I am just wondering if you have any thoughts on this??
Perhaps it is not possible to use this with PyVisa

import visa
>>> rm = visa.ResourceManager()
>>> a = rm.list_resources()
>>> print a
(u'RSNRP::0x00a8::100906::INSTR',)     <- This is not directly visa supported and is running through
>>> inst = rm.open_resource(a[0])               Visa Passport. Visa Max recognises it by this name
Traceback (most recent call last):
  File "<pyshell#5>", line 1, in <module>
    inst = rm.open_resource(a[0])
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1618, in open_resource
    info = self.resource_info(resource_name)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1585, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1378, in parse_resource_extended
    return ResourceInfo(constants.InterfaceType(interface_type.value),
  File "C:\Python27\lib\site-packages\enum\__init__.py", line 330, in __call__
    return cls.__new__(cls, value)
  File "C:\Python27\lib\site-packages\enum\__init__.py", line 642, in __new__
    raise ValueError("%s is not a valid %s" % (value, cls.__name__))
ValueError: 33024 is not a valid InterfaceType

ewal31, 2015-01-23:
Through VISA Interactive which I can use to manually send queries the following properties are listed as Invalid
I wondered if perhaps one or many of these are causing the invalid interface type?

Is there a way to make this work by bypassing some of the requirements

Invalid Properties
Allow DMA Transfers
Is 488.2 Compliant
I/O Protocol
Model Name
Model Code
Slot
Trigger ID
Window Base Address
Window Size
Window Access
Source Increment Count
Destination Increment Count
GPIB Primary Address
GPIB Secondary Address
Enable Readdressing
Enable Unaddressing
REN Line State
Number of Bytes at Port
Baud Rate
Data Bits
Parity
Stop Bits
Flow Control
End Mode for Reads
End Mode for Writes
XON Character
XOFF Character
Error Replacement Character
Discard NULL Characters
Break Length
and Many More

stadi1987, 2015-01-23:
Hi

I can confirm that the USB power sensors worked with pyvisa 1.4 with visa.instrument('RSNRP::0x000c::108766::INSTR') (depending on the serial number). However with 1.6 I have the same problem as you.

Hope this can be fixed soon.

hgrecco, 2015-01-24:
I do not think is an issue with PyVISA but I do thing we might look into how to make it work. `RSNRP` is not a VISA InterfaceType. As you mentioned before, you are using a Visa Passport Driver to talk to it. 
But  under the hood, the Visa Passport Driver is talking via a USB port using a a proprietary binary protocol for the data transfer. If we can get info about this protocol, we might implement it using as USB RAW device. in a similar way as we are doing it with the prologix adapter (#112)

ewal31, 2015-01-24:
So the Visa Passport Manual can be accessed at this address

http://cdn.rohde-schwarz.com/pws/dl_downloads/dl_software/pdf_2/RsViNrp_Software_Manual_2_4.pdf

Page 6 and 7 details the available visa functions

There is some more detail in the document but not a large amount

hgrecco, 2015-01-24:
@stadi1987 Thanks for your report. It brings another perspective. If it was working with 1.4 then we should make it work again and I have a pretty good idea how. But I will need you and @ewal31 to test it as I do not have the patch.

Still, my previous comment holds. If we are able to get our hands in the  proprietary binary protocol that the Visa Passport Driver is implementing then we will be able to make these instruments work in any OS in which VISA works even if the passport driver is not available.

hgrecco, 2015-01-24:
@ewal31 Yes, there is not so much info. I will work to restore the 1.4 functionality without loosing some of the nice checks that 1.5 introduced.

ewal31, 2015-01-24:
I will see if I can find anything else

hgrecco, 2015-01-24:
Going back to the standard fix, I think that adding `rsnrp = 33024` in the InterfaceType enum within `constants.py` will solve the problem (at least the one posted in the issued, will be great if you can check!). But the question is if this is a sustainable solution:

To go in more details, VISA defines a constant for each interface type:
```
VI_INTF_GPIB                 = 1
VI_INTF_VXI                  = 2
VI_INTF_GPIB_VXI             = 3
VI_INTF_ASRL                 = 4
VI_INTF_PXI                  = 5
VI_INTF_TCPIP                = 6
VI_INTF_USB                  = 7
VI_INTF_RIO                  = 8
VI_INTF_FIREWIRE             = 9
```

As you can see, there is no `VI_INTF_RSNRP` because it is not standard and the functionality is provided by the Rohde and Schwarz library. What is not clear to me is where the RSNRP constant (`33024`) is defined. In other words, is there a list or a VISA function that I can call and get the "prefix to number" relationship for non standard interface types?

ewal31, 2015-01-24:
Thank you I will try this. Though I can't until Tuesday when I return to work.

I suppose that this is very situation dependent. Perhaps nothing more could be done except to allow an override giving the user the option to bypass the type check. Assuming that this doesn't cause more problems.

hgrecco, 2015-01-24:
@ewal31 Allowing values outside the enum is certainly a possibility (that is how 1.4 was working). But using enums helps to debug and enforce correctness. So if there is a more comprehensive solution we should give it a try.

However, if the quick fix solves the problem we can included it in the next minor release until we find a better solution.

ewal31, 2015-01-24:
All I can think is to search through the dll files then as I think the passport driver just maps there location in the registry. I don't believe this would really be plausible though

hgrecco, 2015-01-24:
As far as I know, extensibility through passport is not part of the official VISA API and therefore there is not function in VISA to ask for this. But there has to be a way in NI VISA to do this because NI-MAX is able to list the installed passports. My only worry about hardcoding `rsnrp = 33024` is what if in another OS or bitness the number is different. In any case, lets wait until we now if this works out for you and we see how to proceed.

stadi1987, 2015-01-26:
Why does VISA list the resource as RSNRP::... if this is not a VISA identifier?

stadi1987, 2015-01-26:
By the way there is another way of controlling the NRP-Z power sensors using the VXI-driver and DLL (Without VISA) but I don't think this will help. If someone's interested I can provide the code.

hgrecco, 2015-01-26:
@stadi1987  Let me rephrase. RSNRP is not a **canonical** resource. It is provided by an extension (in this case the Rohde and Schwarz library) through a hook in the VISA library. If the Interface Type value is fixed (i.e. is always 33024 for RSNRP) then we can hard code it.

ewal31, 2015-01-26:
It still seems to be returning the same error.

hgrecco, 2015-01-26:
@ewal31 Are you sure it is picking up the changes. Because the error is related to the creation of an enum object: `ValueError: 33024 is not a valid InterfaceType` tells that there is no Value for InterfaceType with that value.

ewal31, 2015-01-26:
No I am not sure

But I deleted all the compiled files so that it would have to recompile

hgrecco, 2015-01-26:
try to run the following code:
```
>>> from pyvisa import constants
>>> constants.InterfaceType(33024)
```

ewal31, 2015-01-26:
Traceback (most recent call last):
  File "<pyshell#3>", line 1, in <module>
    constants.InterfaceType(33024)
  File "C:\Python27\lib\site-packages\enum\__init__.py", line 330, in __call__
    return cls.__new__(cls, value)
  File "C:\Python27\lib\site-packages\enum\__init__.py", line 642, in __new__
    raise ValueError("%s is not a valid %s" % (value, cls.__name__))
ValueError: 33024 is not a valid InterfaceType

So its not recognizing the change then

hgrecco, 2015-01-26:
The code should look like:
```python
class InterfaceType(enum.IntEnum):
    """The hardware interface
    """

    #: GPIB Interface.
    gpib = VI_INTF_GPIB

    #: VXI (VME eXtensions for Instrumentation), VME, MXI (Multisystem eXtension Interface).
    vxi = VI_INTF_VXI

    #: GPIB VXI (VME eXtensions for Instrumentation).
    gpib_vxi = VI_INTF_GPIB_VXI

    #: Serial devices connected to either an RS-232 or RS-485 controller.
    asrl = VI_INTF_ASRL

    #: PXI device.
    pxi = VI_INTF_PXI

    #: TCPIP device.
    tcpip = VI_INTF_TCPIP

    #: Universal Serial Bus (USB) hardware bus.
    usb = VI_INTF_USB

    #: Rio device.
    rio = VI_INTF_RIO

    #: Firewire device.
    firewire = VI_INTF_FIREWIRE

    rsnrp = 33024
```

You can also do:
```python
>>> print(constants.__file__)
```
to find out where is the import location

ewal31, 2015-01-26:
Yes sorry I just saw that function and had changed it there and it is correctly returning the interface type

Traceback (most recent call last):
  File "C:\Users\Edward\Desktop\Power Sensor.py", line 4, in <module>
    inst = rm.open_resource(a)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1623, in open_resource
    raise ValueError('There is no class defined for %r' % ((info.interface_type, info.resource_class),))
ValueError: There is no class defined for (<InterfaceType.rsnrp: 33024>, u'INSTR')

This is the new error

hgrecco, 2015-01-26:
Ok. This is something that I have to solve. But now that I know that this thing is fixing your (first) problem, I can patch it and we can try it again. I will keep you posted

hgrecco, 2015-01-27:
please update to the latest version in the repo and let me know

ewal31, 2015-01-27:
As far as I can tell it now works perfectly.
You are a genius!
Thanks

hgrecco, 2015-01-28:
I am glad that it works. I have associated the `rsnrp` to a `MessageBasedResource` which is the most generic one suitable for this. If you find that some attribute / function is missing let me know. I am closing the issue now.

closed 2015-01-28 by hgrecco