pyvisa/pyvisa #116: VI_SUCCESS_MAX_CNT warning not suppressed during read_raw()

TonyBeswick, 2015-02-02:
`VisaIOWarning: VI_SUCCESS_MAX_CNT (1073676294): The number of bytes transferred is equal to the requested input count. More data may be available.`

This warning is displayed when reading large amounts of data (requiring multiple reads) from an instrument.

**MessageBasedResource.read_raw()** checks for more data so it handles this status appropriately.  I believe it is not necessary to present this warning to the user when it originates from **read_raw()**.

Could we change this line in read_raw() from:
`with self.ignore_warning(constants.VI_SUCCESS_DEV_NPRESENT):`
to
`with self.ignore_warning(constants.VI_SUCCESS_DEV_NPRESENT, 
                                         constants.VI_SUCCESS_MAX_CNT):`

hgrecco, 2015-02-03:
In principle, I thing that `read_raw` should be as close as possible to the original one. I would be fine to add this in `read`, though.

TonyBeswick, 2015-02-05:
pyvisa 1.4 does ignore VI_SUCCESS_MAX_CNT during `read_raw`.

`read_values` and `query_binary_values` also call `read_raw`, in my opinion ignoring it only in `read` doesn't fully address the issue.

hgrecco, 2015-02-05:
`read_values` is deprecated.
But you are right about `query_binary_values`. We could add the ignore there. My idea is that read_raw should be `viRead` + concatenating. Is there a reason why you do not like this approach?

TonyBeswick, 2015-02-06:
My understanding is that VI_SUCCESS_MAX_CNT is returned by `viRead` when there is likely to be more data to be read so the caller knows to call `viRead` again.
  The concatenating in `read_raw` responds to this status appropriately by calling `viRead` again.  My view is that because `read_raw` handles that status, then there is no need to present it as a warning to the user (its already taken care of and the user need not address it).

In my opinion putting the `ignore_warning` one level higher in all methods that call `read_raw` is not the best approach, it creates more code and doesn't address the problem at its source.

Did I mention that pyvisa 1.4 ignores VI_SUCCESS_MAX_CNT in `read_raw`?  My suggestion keeps the 1.6 functionality in line with pyvisa 1.4.

hgrecco, 2015-02-06:
I will look into it again and decide. Thanks for your thoughts in this matter.

hgrecco, 2015-02-19:
I tested different ways and I agree with your view. Please submit a Pull Request and I will merge it.

commit 2d5ced60 (2015-03-25): Suppress VI_SUCCESS_MAX_CNT warning on read raw

Close #116
closed 2015-03-25 by hgrecco