pyvisa/pyvisa #117: Setting Attributes

joeadam, 2015-02-07:
This may just be me missing the right way to do it.

I want to set the VI_ATTR_TERM_CHAR_EN to True.  However I don't seem to be able to write to it, I always get an error about incorrect type.  I've tried True, 1 and pyvisa.constants.VI_TRUE

The resource I working with is a GPIB instrument, which responds, but doesn't seem to terminate the message correctly (so NI Spy shows the reply bytes in the buffer, with termination character, but the call timesout).

hgrecco, 2015-02-07:
Please notice that the constant is called `VI_ATTR_TERMCHAR_EN`. I guess is a typo here, not in you code so I am answering anyway.

The most general way to set a VISA attribute is:
```python
inst.set_visa_attribute(<property number>, <property state>)
``` 
where `inst` is the result the `open_resource` method, `<property number>` is the constant for the attribute (you can look in `visa.constants`) and `<property state>` is the state you want to set.

Many attributes has a python friendly name, but not this one. The reason is that this value is automatically set to True if you set the `read_termination` value to something different from an empty string.

Please provide code if this does not answer your question.

joeadam, 2015-02-09:
Ah,
My error message was actually the type of property number.  I had assumed that the property number was actually a string.
All working now.
Thanks for the quick reply – much appreciated!

Joseph Adam, Technology Developer

Sentec Ltd              phone:  +44(0) 1223 303800
5 The Westbrook Centre  fax:    +44(0) 1223 303801
Milton Road              
Cambridge               email:  jadam@sentec.co.uk
CB4 1YG, UK             web:    www.sentec.co.uk

This email is confidential. If you have received it in error, please notify Sentec Ltd UK at postmaster@sentec.co.uk immediately,
delete it from your system and note that you may not copy, distribute or use its contents.

Sentec Limited is registered at the above address UK Company Number 3452194.

hgrecco, 2015-02-09:
Then I am closing this. Great that I could help

closed 2015-02-09 by hgrecco