pyvisa/pyvisa #118: Problem transferring binary data over TCPIP

andyTPE, 2015-02-10:
Trying to transfer binary trace data from an instrument to a PC.  Using PyVISA 1.6.1 with a TCPIP0 connection.  Ascii trace transfers work without issue.  Binary traces transfers work for a limited set of use cases. It would seem that reading back is terminated early (insufficient number of bytes read) and then an error is thrown when the bytes cannot be parsed back into floating point (either 'f' or 'd').

I have tried this with both NI and Keysight VISA.

debug_info() returns:
4295920
4295920
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.9
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          Dec 10 2014 12:24:55 (#default)
   Unicode:        UCS2

PyVISA Version: 1.6.1

Backends:
   ni:
      Version: 1.6.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904

Samples Code to illustrate issue:
#-------------------------------------------------------------------------------
# Name:        testVisa
# Purpose:
#
# Author:      astreet
#
# Created:     10/02/2015
# Copyright:   (c) astreet 2015
# Licence:     <your licence>
#-------------------------------------------------------------------------------

#-------------------------------------------------------------------------------
# Sample code to illustrate BINARY READ ISSUE
#-------------------------------------------------------------------------------
print '*** Code snippet to explore BIN trace transfer issue ***'
# ---------------------------------
# Import modules
# ---------------------------------
import visa
import numpy as np

# ---------------------------------
# Connect to instrument
# ---------------------------------
rm = visa.ResourceManager()
# print rm
# Resource Manager of Visa Library at C:\Windows\system32\visa32.dll

myInst = rm.open_resource('TCPIP0::192.168.1.100::5025::SOCKET')

myInst.read_termination  ='\n'
myInst.write_termination = '\n'
myInst.timeout = 5000

# Clear and reset instrument
myInst.clear()
myInst.write('syst:pres;*wai;')
opcBit = myInst.query('*opc?')
myInst.write(':mmem:load "d:/mmm.csa";*wai')
opcBit = myInst.query('*opc?')

# Check initial comms with instrument
idStr = myInst.query('*idn?')
print '*IDN? string...{0}'.format(idStr)

# ---------------------------------
# Do sweep
# ---------------------------------
# Initialize one sweep
opcBit = myInst.query('sens1:swe:mode single;*wai;*opc?')
# Select trace to transfer (TRC-1)
myInst.write('calc1:par:mnum 1')

# ---------------------------------
# ASCII DATA TRANSFER
# fData returns formatted data: in this instance dB magnitude
# sData return complex pairs (2x #trace points)
# ---------------------------------
print ''
print '*** ASCII TRACE TRANSFER ***'
myInst.write('form:data ascii,0')
a_fData = myInst.query_ascii_values('calc1:data? fdata')
a_sData = myInst.query_ascii_values('calc1:data? sdata')
print '#pts Fdata...\t{0}'.format(len(a_fData))
print '#pts Sdata...\t{0}'.format(len(a_sData))

# Check that fData and sData give same results
F0 = a_fData[0]
S0 = 20 * np.log10 (np.abs(complex(a_sData[0], a_sData[1])))

print 'a_fdata[0]={0:+3.3f}\t\ta_sData[0]={1:+3.3f}'.format(F0, S0)
# a_fdata[0]=-10.218		a_sData[0]-10.218

F50 = a_fData[50]
S50 = 20 * np.log10 (np.abs(complex(a_sData[100], a_sData[101])))

print 'a_fdata[50]={0:+3.3f}\t\ta_sData[50]={1:+3.3f}'.format(F50, S50)
# a_fdata[50]=-9.545		a_sData[50]=-9.545

# ==== SUMMARY: CAN SUCCESSFULLY READ FDATA AND SDATA IN ASCII FORMAT ====

# ---------------------------------
# BINARY DATA TRANSFER
# Supports 32 and 64 bit floating point binary transfers
# Format of data is:
# <hdr><Ndigits><Nbytes><data>
# <hdr> = #
# <Ndigits> = Number of digits that follows which represents the byte count
# <Nbytes>  = Number of bytes in the data payload
# <data>    = binary data formatted in either 'f' or 'd'
# Example #3204<data> 204 bytes of data in payload
# ---------------------------------
print ''
print '*** BINARY TRACE TRANSFER ***'
myInst.write('form:data real,64')
myInst.write('form:border norm') # NORM | SWAP for MSB/LSB first control

b_fData = myInst.query_binary_values('calc1:data? fdata', datatype=u'd', is_big_endian=True)
print '#pts b_fData...\t{0}'.format(len(b_fData))
#pts b_fData...	51
print 'b_fData[0]...\t{0:+3.3f}'.format(b_fData[0])
# b_fData[0]...	-10.217
print 'b_fData[50]...\t{0:+3.3f}'.format(b_fData[50])
# b_fData[50]...	-9.545

myInst.write('calc1:data? fdata')
r_fData = myInst.read_raw()
print r_fData[0:5]
# #3408

# PROBLEMS START WHEN EITHER REAL,32 FORMAT IS USED OR SDATA TRANSFER (DOUBLE THE
# NUMBER OF POINTS)

myInst.write('form:data real,32')
myInst.write('form:border norm') # NORM | SWAP for MSB/LSB first control

# Following lines fails:
# b_fData = myInst.query_binary_values('calc1:data? fdata', datatype=u'f', is_big_endian=True)
# Error Message:
##Traceback (most recent call last):
##  File "C:\Users\astreet\Documents\Python\Learn\VISA\testVisa.py", line 122, in <module>
##    b_fData = myInst.query_binary_values('calc1:data? fdata', datatype=u'f', is_big_endian=True)
##  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 468, in query_binary_values
##    raise errors.InvalidBinaryFormat(e.args)
##  File "C:\Python27\lib\site-packages\pyvisa\errors.py", line 425, in __init__
##    description = ": " + description
##TypeError: coercing to Unicode: need string or buffer, tuple found

myInst.write('calc1:data? fdata')
rawChk = myInst.read_raw()
#>>> rawChk
# '#3204\xc1#z\xe8\xc14\xa1)\xc1J\xbe9\xc1g\x89\xf9\xc1\x85E\x9a\xc1\x9ae\x00\xc1\xb0\xf3\xae\xc1\xb3\xdd\xd9\xc1\x9d\xd3\x8f\xc1\x86\xd1d\xc1g5_\xc1H\x03\xf0\xc1/x\x99\xc1\x1f`\xe4\xc1\x11\x10\x19\xc0\xff\xcd,\xc0\xed\x04\xba\xc0\xd2\xf78\xc0\xb7\xcb\xb1\xc0\xa5.E\xc0\x86^,\xc0gv\xa5\xc0`*v\xc0B\xa8\x8a\xc0\x14\xff|\xc0\x154\x8a\xc0\x06\xb7\xde\xc0<\xc3\xa1\xc0I\xe6F\xc0w#/\xc0\x9b\xfa-\xc0\xb7V\xc2\xc0\xdbQ\xe0\xc0\xfe\x05\xb7\xc1\x0c\x1d^\xc1\x1cJD\xc10F\xd3\xc1=\xb5\xee\xc1Jn^\xc1Z\x86|\xc1n4~\xc1\x83H\x82\xc1\x94q\xf2\xc1\xac\xdb"\xc1\xc5\x87O\xc1\xb9\xb8}\xc1\x98\xc98\xc1|z\x87\xc1U\x9b-\xc142J\xc1\x18\xc2z\n'
# >>>
# Appears to have correct header, but unable to parse bytes correctly

# Appears that complete binary payload is not being read completely
# Same issues seen if TCPIP0::192.168.1.100::inst0 is used

hgrecco, 2015-02-10:
I have taken the freedom to add triple quotes around your code to make it easy (actually possible, :-) ) to read.

hgrecco, 2015-02-10:
Please update to latest pyvisa which fixes a number of import issues.

Regarding your question. The binary format that you are describing is an IEEE binary block (which is the default). When I parse it directly, I get:

```python
>>> from pyvisa import util, __version__
>>> print(__version__)
1.6.3
>>> raw = b'#3204\xc1#z\xe8\xc14\xa1)\xc1J\xbe9\xc1g\x89\xf9\xc1\x85E\x9a\xc1\x9ae\x00\xc1\xb0\xf3\xae\xc1\xb3\xdd\xd9\xc1\x9d\xd3\x8f\xc1\x86\xd1d\xc1g5_\xc1H\x03\xf0\xc1/x\x99\xc1\x1f`\xe4\xc1\x11\x10\x19\xc0\xff\xcd,\xc0\xed\x04\xba\xc0\xd2\xf78\xc0\xb7\xcb\xb1\xc0\xa5.E\xc0\x86^,\xc0gv\xa5\xc0`*v\xc0B\xa8\x8a\xc0\x14\xff|\xc0\x154\x8a\xc0\x06\xb7\xde\xc0<\xc3\xa1\xc0I\xe6F\xc0w#/\xc0\x9b\xfa-\xc0\xb7V\xc2\xc0\xdbQ\xe0\xc0\xfe\x05\xb7\xc1\x0c\x1d^\xc1\x1cJD\xc10F\xd3\xc1=\xb5\xee\xc1Jn^\xc1Z\x86|\xc1n4~\xc1\x83H\x82\xc1\x94q\xf2\xc1\xac\xdb"\xc1\xc5\x87O\xc1\xb9\xb8}\xc1\x98\xc98\xc1|z\x87\xc1U\x9b-\xc142J\xc1\x18\xc2z\n'
>>> util.from_ieee_block(raw, datatype='f', is_big_endian=True )
[-10.217506408691406, -11.289345741271973, -12.671441078186035, -14.471184730529785, -16.658985137939453, -19.29931640625, -22.11898422241211, -22.48332405090332, -19.728300094604492, -16.85224151611328, -14.450530052185059, -12.500961303710938, -10.96694278717041, -9.96115493774414, -9.06643009185791, -7.993795394897461, -7.406826972961426, -6.592678070068359, -5.743614673614502, -5.161898136138916, -4.198995590209961, -3.6166164875030518, -3.502591609954834, -3.041536808013916, -2.3280935287475586, -2.331331729888916, -2.1049723625183105, -2.9494402408599854, -3.154679775238037, -3.8615224361419678, -4.874289035797119, -5.729340553283691, -6.8537445068359375, -7.938197612762451, -8.757169723510742, -9.768131256103516, -11.017291069030762, -11.856916427612305, -12.651945114135742, -13.657833099365234, -14.887815475463867, -16.410404205322266, -18.55563735961914, -21.606998443603516, -24.691068649291992, -23.2150821685791, -19.098251342773438, -15.779913902282715, -13.350384712219238, -11.262277603149414, -9.547479629516602]
```
>>> print(len(_))
51

Not sure if the values are right, but the amount of values is correct.

hgrecco, 2015-02-11:
Let me know if after updating works for you.

andyTPE, 2015-02-11:
Hello Hernan,
Upgrade to 1.6.3 made issue worse.  Reading back real,64 ('d') throws an error.  I tried to use read_raw() (to get data back and then parse with util.from_ieee_block() - error is Binary data is malformed.

Sample return is:
'#3408\xc0%\x10\x94@\x00\x00\x00\xc0&\xeb\xfc`\x00\x00\x00\xc0)}\xbe\xc0\x00\x00\x00\xc0,T?\xe0\x00\x00\x00\xc0/\xc4\x02\xc0\x00\x00\x00\xc02[\xa7@\x00\x00\x00\xc05K\x9f`\x00\x00\x00\xc06\xf7\xdb \x00\x00\x00\xc04\xed\x9c \x00\x00\x00\xc01\xb0\xd8\xc0\x00\x00\x00\xc0.!\x0f\xc0\x00\x00\x00\xc0)\xbb\xff\xe0\x00\x00\x00\xc0&Y\xad`\x00\x00\x00\xc0#\xdb\xee \x00\x00\x00\xc0!\xda\xca\x80\x00\x00\x00\xc0\x1f[\xf9`\x00\x00\x00\xc0\x1b\xe9J\xe0\x00\x00\x00\xc0\x19=\xd6`\x00\x00\x00\xc0\x16\xd6e\xc0\x00\x00\x00\xc0\x14#p\x00\x00\x00\x00\xc0\x0f\xf2\x8f \x00\x00\x00\xc0\x0e\xd1\xb1\xa0\x00\x00\x00\xc0\n'

Was obtained with the following code snippet:
print '*** BINARY TRACE TRANSFER ***'
myInst.write('form:data real,64')
myInst.write('form:border norm') # NORM | SWAP for MSB/LSB first control
myInst.write('calc1:data? fdata')
r_fData = myInst.read_raw()

test = util.from_ieee_block(r_fData, datatype='d', is_big_endian=True )

Please advise on the best way to debug/resolve.  I can send you logs/more data returns etc.

Thank you for your help

Andy

hgrecco, 2015-02-12:
But if in the data that you have pasted, the headers says that we should expect 408 bytes in the data block but there are much less. Right, or I am reading something wrong?

hgrecco, 2015-02-12:
Also if you checked what has changed between 1.6.1 and 1.6.3, nothing relates to this:

https://github.com/hgrecco/pyvisa/compare/1.6.1...1.6.3

andyTPE, 2015-02-12:
I believe that the underlying issue is early termination of the data transfer (i.e. the read does not pull all the data from the buffer).  I do not know why my code has stopped working with upgrading to 1.6.3 as, like you say, nothing changed between 1.6.1 -> 1.6.3 in this area.  Thus I wonder if my install is causing the problem.  I did a pip install pyvisa (and have even tried the uninstall/install cycle) to be at 1.6.3.  Can you advise on the simplest way to revert back to 1.6.1 and I can test again at that code rev.

hgrecco, 2015-02-12:
The strange thing is that your raw data contains a `\n`. So it seems that pyvisa is not cutting the message in an arbitrary way. Is it possible that the instrument is chopping the message? What happens if you call `read_raw`  again?

You can just do 
```
pip install -U pyvisa==1.6.1
```

andyTPE, 2015-02-12:
Thanks for the pip syntax!  Yes read_raw() is completing on \n.  I can execute a further four read_raw() commands until the instrument buffer is empty.  Each transfer is of varying length but ends in a \n.  The next read_raw() results in a timeout since there is nothing left to read...  My read_termination is set to '\n' since for queries (e.g. *idn?) the read terminator is a \n.

andyTPE, 2015-02-12:
I do not believe the instrument is chopping the message - the multiple read_raw() suggest that it is in the interface.  Also, I can execute the same process using Agilent VEE (another programming language) and the binary trace transfer is handled without any problems - I believe is uses the same VISA interface under the hood as pyvisa.

hgrecco, 2015-02-12:
>  I do not believe the instrument is chopping the message - the multiple read_raw() suggest that it is in the interface. 

Are you sure that the Agilent VEE is not doing multiple `read_raw` until the number of received bytes (as given in the header) matches the number of expected bytes? I do not think that VISA is inserting `\n`. 

We could add the option to do multiple calls to `read_raw` in `query_binary_values` but it would be nice to know that this is really the problem.

andyTPE, 2015-02-12:
Hi Hernan,

I used NI IO trace to capture the traffic.  Attached are screenshots and .txt export for the Agilent VEE (it works) and PyVisa (problem) cases.

In the case of VEE it looks like it parses out the IEEE Binblock header (lines 20-23).

For PyVisa it assigns 20480 chunk size, but terminates the read at 145 bytesâ€¦.

Could it be that \n is a valid byte in the data payload, but that PyVisa is causing the read to terminate at this point?

Thanks

Andy

â€”

1.  viOpenDefaultRM (0x023FD500)
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:03.887       Call Duration 00:00:00.068
Status: 0 (VI_SUCCESS)

2.  viParseRsrcEx (0x023FD500, "TCPIP0::192.168...0::5025::SOCKET", 6, 0, "SOCKET", "TCPIP0::192.168...0::5025::SOCKET", "")
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:03.996       Call Duration 00:00:00.000
Status: 0 (VI_SUCCESS)

3.  viOpen (0x023FD500, "TCPIP0::192.168...0::5025::SOCKET", 0, 0, 0x03292F78)
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.037       Call Duration 00:00:00.011
Status: 0 (VI_SUCCESS)

4.  viParseRsrcEx (0x023FD500, "TCPIP0::192.168.1.100::inst0", 6, 0, "INSTR", "TCPIP0::192.168...0::inst0::INSTR", "")
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.088       Call Duration 00:00:00.000
Status: 0 (VI_SUCCESS)

5.  viOpen (0x023FD500, "TCPIP0::192.168.1.100::inst0", 0, 0, 0x032955C0)
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.129       Call Duration 00:00:00.005
Status: 0 (VI_SUCCESS)

6.  viClose (TCPIP0::192.168.1.100::5... (0x03292F78))
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.174       Call Duration 00:00:00.001
Status: 0 (VI_SUCCESS)

7.  viSetAttribute (TCPIP0::192.168.1.100::i... (0x032955C0), TERMCHAR, 10)
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.215       Call Duration 00:00:00.000
Status: 0 (VI_SUCCESS)

8.  viSetAttribute (TCPIP0::192.168.1.100::i... (0x032955C0), TERMCHAR_EN, VI_TRUE)
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.255       Call Duration 00:00:00.000
Status: 0 (VI_SUCCESS)

9.  viSetAttribute (TCPIP0::192.168.1.100::i... (0x032955C0), TMO_VALUE, 5000)
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.295       Call Duration 00:00:00.000
Status: 0 (VI_SUCCESS)

10.  viWrite (TCPIP0::192.168.1.100::i... (0x032955C0), "*idn?.", 6, 6)
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.336       Call Duration 00:00:00.001
Status: 0 (VI_SUCCESS)

11.  viRead (TCPIP0::192.168.1.100::i... (0x032955C0), "Agilent Technologies,...", 20480, 50)
Process ID: 0x00002438         Thread ID: 0x00002778
Start Time: 14:58:04.378       Call Duration 00:00:00.001
Status: 0 (VI_SUCCESS)
[... 150 lines cut ...]

23.  viRead (TCPIP0::192.168.1.100::i... (0x0AE5BF28), "À%.+à...À&êÀ...À)}d ...", 408, 408)
Process ID: 0x00003428         Thread ID: 0x00000E68
Start Time: 14:53:46.333       Call Duration 00:00:00.000
Status: 0x3FFF0006 (VI_SUCCESS_MAX_CNT)

24.  viRead (TCPIP0::192.168.1.100::i... (0x0AE5BF28), ".", 1, 1)
Process ID: 0x00003428         Thread ID: 0x00000E68
Start Time: 14:53:46.333       Call Duration 00:00:00.001
Status: 0 (VI_SUCCESS)

25.  viGetAttribute (TCPIP0::192.168.1.100::i... (0x0AE5BF28), TERMCHAR_EN, VI_FALSE)
Process ID: 0x00003428         Thread ID: 0x00000E68
Start Time: 14:53:46.334       Call Duration 00:00:00.000
Status: 0 (VI_SUCCESS)

26.  viUnlock (TCPIP0::192.168.1.100::i... (0x0AE5BF28))
Process ID: 0x00003428         Thread ID: 0x00000E68
Start Time: 14:53:46.334       Call Duration 00:00:00.000
Status: 0 (VI_SUCCESS)

27.  viClose (TCPIP0::192.168.1.100::i... (0x0AE5BF28))
Process ID: 0x00003428         Thread ID: 0x00000E68
Start Time: 14:53:46.339       Call Duration 00:00:00.028
Status: 0 (VI_SUCCESS)

andyTPE, 2015-02-12:
Hi Hernan, Think I have got to the root cause.  \n is probably a valid byte in the data payload.  However read_termination='\n' so the binary read terminates at this point.  Solution is to set read_termination=None prior to binary read.  Then, I have robust data transfer.  Maybe should mention in docs?

hgrecco, 2015-02-12:
Great that you found the problem.

I am afraid that solution might give us some maintenance headaches as depending on the data, people will not realize this and will have intermittent failures.

A better solution will be to continue reading based on the header. I need to see what is the standard solution as it depends if there is something to signal the end of message. For example in some binary protocols, the number of points is unknown.

andyTPE, 2015-02-12:
Here is a typical return.  Note that \n is a valid byte in the payload - and one is also attached to signify the end of the array.  Right now, my solution is to toggle the read_termination between None and LF (\n) - since typically I am querying the instrument prior to asking for binary trace data....

'#3408\xc0%\x12+`\x00\x00\x00\xc0&\xeb\x16\x80\x00\x00\x00\xc0)}\xd4\x80\x00\x00\x00\xc0,Ws@\x00\x00\x00\xc0/\xc3~\x80\x00\x00\x00\xc02]\x1f\xc0\x00\x00\x00\xc05M\xef\xa0\x00\x00\x00\xc06\xf9\xf8@\x00\x00\x00\xc04\xeey`\x00\x00\x00\xc01\xb1\xed\xa0\x00\x00\x00\xc0.#\xea \x00\x00\x00\xc0)\xbc\xdc\xc0\x00\x00\x00\xc0&ZX`\x00\x00\x00\xc0#\xdb\xc5\x00\x00\x00\x00\xc0!\xdbY\xc0\x00\x00\x00\xc0\x1f^\xe0 \x00\x00\x00\xc0\x1b\xe9\xf3`\x00\x00\x00\xc0\x19?6`\x00\x00\x00\xc0\x16\xd6\xd0`\x00\x00\x00\xc0\x14$\x84\xa0\x00\x00\x00\xc0\x0f\xf4\xca`\x00\x00\x00\xc0\x0e\xd2W\xa0\x00\x00\x00\xc0\n\xdc\xcb\x00\x00\x00\x00\xc0\x08\xd4W`\x00\x00\x00\xc0\x01<\xe8\xe0\x00\x00\x00\xc0\x02\x89\xa2\xe0\x00\x00\x00\xc0\x05\x82\x92 \x00\x00\x00\xc0\x04\r\x1f \x00\x00\x00\xc0\t\xe6\xf7\xe0\x00\x00\x00\xc0\x10\xf1\x9c@\x00\x00\x00\xc0\x12\xb2\xd6`\x00\x00\x00\xc0\x17Z~@\x00\x00\x00\xc0\x1a\xd5\x12\x80\x00\x00\x00\xc0\x1f\x11\x9e\x80\x00\x00\x00\xc0"\x0b<`\x00\x00\x00\xc0$\x1bJ\xe0\x00\x00\x00\xc0&^\xc9\xc0\x00\x00\x00\xc0(J\xa8\x00\x00\x00\x00\xc0*\rv`\x00\x00\x00\xc0+\xfaM \x00\x00\x00\xc0.Hc\xa0\x00\x00\x00\xc00\x90S`\x00\x00\x00\xc02x\xc2\x00\x00\x00\x00\xc05\x1d\x91`\x00\x00\x00\xc07\xe5\xd8`\x00\x00\x00\xc07\x10\x8f\x00\x00\x00\x00\xc03fj@\x00\x00\x00\xc001\\@\x00\x00\x00\xc0*\xf7\x1f\x00\x00\x00\x00\xc0&\xe8\x86\xc0\x00\x00\x00\xc0#\xc2x`\x00\x00\x00\n'

hgrecco, 2015-02-18:
@andyTPE I have pushed a change to `query_binary_values`. It continues to read while the data length is not achieved. Can you take a look?

andyTPE, 2015-03-16:
Hi Hernan, Apologies for the silence - I did not have access to my instrument for a couple of weeks. I would like to get back on this - please advise on how I can download your new version with the change to query_binary_values().  Thanks Andy

hgrecco, 2015-03-16:
@andyTPE it would be great if you can check. Try the following:
```
pip install https://github.com/hgrecco/pyvisa/archive/master.zip
```

HKIAL, 2015-04-17:
Hello hgrecco,
ich had the same problem; in my binary data the terminition symbol might occur.
I tired the new package (https://github.com/hgrecco/pyvisa/archive/master.zip) with this settings
"XMC = rm.open_resource(u'ASRL7::INSTR', baud_rate=9600, timeout = 600,write_termination = '\0',read_termination = '\r\n')"
to read a datapackage where all (16 bit) values between 0 and 8191 occur.
The new code seems to work, because the ord('\r\n') = 3338 was correctly transmitted and the transfer was not stoped like before. I forgot to check out the version of my old pyvisa package before I installed this one, but I think it was 1.6.3 or 1.6.4 (not sure!).

Thank you for your work and this great piece of software!

hgrecco, 2015-04-17:
Great! I am closing this issue. Feel free to reopen if needed.

closed 2015-04-17 by hgrecco

cubuff419, 2017-09-16:
Hi guys, I stumbled across this thread because I identified the very same issue still occurring for me with PyVisa 1.8 and a GPIB instrument.  My desired termination character is "CR+LF" but there is a high probability that this occurs as a valid binary sequence in my single-byte unsigned integer data (datatype 'B').  I'm finding that read_raw() doesn't honor my desired data length and query_binary_values() won't take a size argument as suggested in a comment above.  My only solution so far is to remove the termination character, as discussed early on in the thread.  I'm new to this, but it looks like there was a move around PyVisa 1.6 to not automatically strip trailing "CR+LF" from string responses, which is why I found the need to set the termination character... can I not have my cake and eat it too??