pyvisa/pyvisa #119: Query_ascii_values issue reading RAW and ASCII data from DSP over USB

phsdv, 2015-02-13:
When trying to get all data from a channel from the HP ( eh sorry Agilent, eh sorry, Keysight) DSO X 3024A scope. However query_ascii_values fails with a VisaIOError: VI_ERROR_IO (-1073807298)
And often the DSO is hanging too. So not sure if this is an instument issue or a pyvisa. But would like to find out which one is causing the issue.
I tried first with WAVeform:FORMat RAW and reading bytes. Also tries with :WAVeform:FORMat ASCII but not much difference. I did made the RAW version works sometimes, but after a few times reading it failed again. I did change the chunk_size to a rather high value, as there could be a lot of data, but I am not sure this is needed.

The DSO is connected to a Windows 7 PC via (a long) USB cable. I am using pyvisa 1.6.2 and Python 3.4

Help would be appreciated to get this working.

Some output from my program:

DEBUG:1 *IDN?: ['AGILENT TECHNOLOGIES', 'DSO-X 3024A', 'MY53160346', '02.30.2013032600']
DEBUG: *RST
DEBUG: *CLS
DEBUG: DSOX3000 5.484s WRITE=':WAVeform:SOUR CHAN4'
DEBUG: DSOX3000 5.491s WRITE=':WAVeform:POINts:MODE NORMAL'
DEBUG: DSOX3000 5.501s WRITE=':WAVeform:POINts 5000'
DEBUG: DSOX3000 5.506s WRITE=':WAVeform:FORMat ASCII'
DEBUG: DSOX3000 5.511s ASK=':WAVeform?'
DEBUG: DSOX3000 5.674s RES=':WAV:UNS 1;VIEW MAIN;BYT MSBF;FORM ASC;POIN +4807;SOUR CHAN4;SOUR:SUBS SUB0'
:WAV:UNS 1;VIEW MAIN;BYT MSBF;FORM ASC;POIN +4807;SOUR CHAN4;SOUR:SUBS SUB0
Traceback (most recent call last):
  File "...Scripts\scopegetdata.py", line 13, in <module>
    x4, y4 = scope.GetData(chan='4')
  File "...\instr\DSOX3000.py", line 281, in GetData
    data = self.instr.query_ascii_values(":WAVeform:DATA?")
  File "C:\Progs\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 432, in query_ascii_values
    block = self.read()
  File "C:\Progs\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 309, in read
    message = self.read_raw().decode(enco)
  File "C:\Progs\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 283, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Progs\Python34\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1569, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Progs\Python34\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 180, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

thanks
Paul

phsdv, 2015-02-13:
just installed version 1.6.3 still the same error:
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

phsdv, 2015-02-18:
With visa.log_to_screen() I get this result:
2015-02-18 11:00:34,633 - pyvisa - DEBUG - viWrite(104713128, b'*CLS\n', 5, 'c_ulong(5)') -> 0
2015-02-18 11:00:34,633 - pyvisa - DEBUG - viGetAttribute(104713128, 1073676314, 'c_ulong(2000)') -> 0
2015-02-18 11:00:34,634 - pyvisa - DEBUG - viSetAttribute(104713128, 1073676314, 50000) -> 0
2015-02-18 11:00:34,635 - pyvisa - DEBUG - viWrite(104713128, b':WAVeform:SOUR CHAN4\n', 21, 'c_ulong(21)') -> 0
2015-02-18 11:00:34,636 - pyvisa - DEBUG - viWrite(104713128, b':WAVeform:POINts:MODE NORMAL\n', 29, 'c_ulong(29)') -> 0
2015-02-18 11:00:34,637 - pyvisa - DEBUG - viWrite(104713128, b':WAVeform:POINts 5000\n', 22, 'c_ulong(22)') -> 0
2015-02-18 11:00:34,638 - pyvisa - DEBUG - viWrite(104713128, b':WAVeform:FORMat ASCII\n', 23, 'c_ulong(23)') -> 0
2015-02-18 11:00:34,639 - pyvisa - DEBUG - viWrite(104713128, b':WAVeform?\n', 11, 'c_ulong(11)') -> 0
2015-02-18 11:00:34,640 - pyvisa - DEBUG - USB0::0x0957::0x17A6::MY53160346::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
DEBUG: DSOX3000 5.824s RES=':WAV:UNS 1;VIEW MAIN;BYT MSBF;FORM ASC;POIN +4807;SOUR CHAN4;SOUR:SUBS SUB0'
:WAV:UNS 1;VIEW MAIN;BYT MSBF;FORM ASC;POIN +4807;SOUR CHAN4;SOUR:SUBS SUB0
2015-02-18 11:00:34,841 - pyvisa - DEBUG - viRead(104713128, <ctypes.c_char_Array_20480 object at 0x078ED3A0>, 20480, 'c_ulong(76)') -> 0
2015-02-18 11:00:34,842 - pyvisa - DEBUG - viWrite(104713128, b':WAVeform:DATA?\n', 16, 'c_ulong(16)') -> 0
2015-02-18 11:00:34,842 - pyvisa - DEBUG - USB0::0x0957::0x17A6::MY53160346::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-02-18 11:01:27,662 - pyvisa - DEBUG - viRead(104713128, <ctypes.c_char_Array_20480 object at 0x078ED3A0>, 20480, 'c_ulong(7934)') -> -1073807298
2015-02-18 11:01:27,662 - pyvisa - DEBUG - USB0::0x0957::0x17A6::MY53160346::INSTR - exception while reading: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Progs\Python34\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 601, in runfile
    execfile(filename, namespace)
  File "C:\Progs\Python34\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 80, in execfile
    exec(compile(open(filename, 'rb').read(), filename, 'exec'), namespace)
  File "C:/Users/fg3zyt/Documents/Scripts/Instruments/scopegetdata.py", line 13, in <module>
    x4, y4 = scope.GetData(chan='4')
  File "C:\Progs\Python34\lib\instr\DSOX3000.py", line 283, in GetData
    data = self.instr.query_ascii_values(":WAVeform:DATA?")
  File "C:\Progs\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 432, in query_ascii_values
    block = self.read()
  File "C:\Progs\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 309, in read
    message = self.read_raw().decode(enco)
  File "C:\Progs\Python34\lib\site-packages\pyvisa\resources\messagebased.py", line 283, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Progs\Python34\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1569, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Progs\Python34\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 180, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
>>>

hgrecco, 2015-03-25:
Do you still have a problem with the latest version?

hgrecco, 2015-04-11:
I am closing this. Feel free to reopen.

phsdv, 2015-04-28:
I have update the firmware (march 2015) on the scope and now the issues are gone. So seems that it was an issue with the Agilent firmware.

phsdv, 2015-04-28:
I still have the issue with 1.6.4.dev0. I think it is working a bit better now,  but I still have issues transferring large data sets. I also updated the firmware of the scope to the latest one of march this year.

It seems that the bigger the data set the earlier I get fails. Small data sets I can transfer more than 40x before an error. Larger sets maybe 10x or even less. However after a while debuging I get fails sooner, so it might be that smaller datasets seems to be better, but only because I started with those.

Another observation is that I do net get all the requested data,
setting data points to 5000 I get 4807 values back in the list
for 10000 I get 8928 values
for 20000 I get 15625 values
for 50000 I get  31250 values (with  chunk size set to 102400)
And I am not able to get larger sizes than that (100K or more) without an io error after the first try.

The fails are VI_ERROR_IO :

  File "c:\progs\python34\lib\instr\DSOX3000.py", line 305, in GetData
    data = self.ReadBlock(":WAVeform:DATA?", datatype='B', tdelay=0.25)
  File "c:\progs\python34\lib\instr\Instruments.py", line 189, in ReadBlock
    data = self.instr.query_binary_values(cmd, datatype=datatype, is_big_endian=is_big_endian, container=container, delay=tdelay)
  File "c:\progs\python34\lib\site-packages\pyvisa\resources\messagebased.py", line 458, in query_binary_values
    block = self.read_raw()
  File "c:\progs\python34\lib\site-packages\pyvisa\resources\messagebased.py", line 283, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "c:\progs\python34\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1569, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "c:\progs\python34\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 180, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

My machine info:
C:>python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 42 Stepping 7, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\progs\python34\python.exe
   Version:        3.4.0
   Compiler:       MSC v.1600 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 16 2014 19:24:06 (#v3.4.0:04f714765c13)
   Unicode:        UCS4

PyVISA Version: 1.6.4.dev0

Backends:
   ni:
      Version: 1.6.4.dev0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5243904
         Spec. Version: 5243136
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5243904
         Spec. Version: 5243136
   py:
      Version: 0.1
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: N/A
      TCPIP INSTR: Available
      ASRL INSTR: Available via PySerial (N/A)

hgrecco, 2015-04-28:
Can you use the NI Measurement and Automation to check if that works?

phsdv, 2015-04-29:
I close this bug as I can reproduce this with NI Measurement and Automation, It seems thus not related to Pyvisa

closed 2015-04-29 by phsdv

hgrecco, 2015-04-29:
Ok. Let me know if we can help in the future.

phsdv, 2015-05-01:
I do get a huge improvement by switching to the visa driver that is supplied by the KeySight (Agilent) IO library tools. Some day I should try to get the pyvisa-py backend running and see if this works (better).

hgrecco, 2015-05-01:
Maybe they do not implement the standard correctly and they own library has the right fixes. It would be nice to know more about this, it might help for other instruments. Feel free to post any other information.

phsdv, 2015-06-01:
Keysight (Agilent) confirmed this as firmware issue. They will work on a fix.

hgrecco, 2015-06-01:
Thanks for looking into this (and confirming it is not a PyVISA issue).