pyvisa/pyvisa #120: Support '\x00' string terminator

bessman, 2015-02-17:
I have an instrument (Amrel PLA800-60-300) which terminates its strings with the character \x00. For example:
```
>>> pla.write('MEASure:VOLTage?')
(18L, <StatusCode.success: 0>)
>>> pla.read_raw()
'3.8050\x00'
```
query_ascii_values() doesn't like this:
```
>>> pla.query_ascii_values('MEASure:VOLTage?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 434, in query_ascii_values
    return from_ascii_block(block, converter, separator, container)
  File "/usr/lib/python2.6/site-packages/pyvisa/util.py", line 195, in from_ascii_block
    return container(converter(raw_value) for raw_value in data)
  File "/usr/lib/python2.6/site-packages/pyvisa/util.py", line 195, in <genexpr>
    return container(converter(raw_value) for raw_value in data)
UnicodeEncodeError: 'decimal' codec can't encode character u'\x00' in position 6: invalid decimal Unicode string
```

bessman, 2015-02-17:
The problem arises from these two lines:
https://github.com/hgrecco/pyvisa/blob/master/pyvisa/resources/messagebased.py#L137
https://github.com/hgrecco/pyvisa/blob/master/pyvisa/resources/messagebased.py#L314

First, the termination is correctly detected and encoded in read_termination():
```python
@read_termination.setter
def read_termination(self, value):
    if value:
        # termination character, the rest is just used for verification after
        # each read operation.
        last_char = value[-1:]
        # Consequently, it's illogical to have the real termination character
        # twice in the sequence (otherwise reading would stop prematurely).

        if last_char in value[:-1]:
            raise ValueError("ambiguous ending in termination characters")

        self.set_visa_attribute(constants.VI_ATTR_TERMCHAR, ord(last_char))
        self.set_visa_attribute(constants.VI_ATTR_TERMCHAR_EN, constants.VI_TRUE)
    else:
        self.set_visa_attribute(constants.VI_ATTR_TERMCHAR_EN, constants.VI_FALSE)

    self._read_termination = value
´´´
However, as it happens the unicode code point for '\x00' is 0, so the message is then returned before the termination is stripped due to 'if not termination' evalualting to True:
´´´
def read(self, termination=None, encoding=None):
    """Read a string from the device.

    Reading stops when the device stops sending (e.g. by setting
    appropriate bus lines), or the termination characters sequence was
    detected.  Attention: Only the last character of the termination
    characters is really used to stop reading, however, the whole sequence
    is compared to the ending of the read string message.  If they don't
    match, a warning is issued.

    All line-ending characters are stripped from the end of the string.

    :rtype: str
    """
    enco = self._encoding if encoding is None else encoding

    if termination is None:
        termination = self._read_termination
        message = self.read_raw().decode(enco)
    else:
        with self.read_termination_context(termination):
            message = self.read_raw().decode(enco)

    if not termination:
        return message

    if not message.endswith(termination):
        warnings.warn("read string doesn't end with "
                      "termination characters", stacklevel=2)

    return message[:-len(termination)]
´´´

bessman, 2015-02-17:
Hmm, that wasn't quite it. Even with https://github.com/hgrecco/pyvisa/blob/master/pyvisa/resources/messagebased.py#L314 changed to 'if termination is None:', it still returns too early. It would seem the termination isn't detected after all.

hgrecco, 2015-02-17:
Thanks for bringing this up. Please upgrade to the latest pyvisa, and to the minimal script that shows the problem add the following code on top:
```
import visa
visa.log_to_screen()
```
And then paste the output here.

bessman, 2015-02-18:
```
>>> pla.write('MEASure:VOLTage?')
2015-02-18 10:22:05,067 - pyvisa - DEBUG - viWrite(157643384L, 'MEASure:VOLTage?\r\n', 18, 'c_ulong(18L)') -> 0
(18L, <StatusCode.success: 0>)
>>> pla.read_raw()
2015-02-18 10:22:14,170 - pyvisa - DEBUG - GPIB0::5::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-02-18 10:22:14,336 - pyvisa - DEBUG - viRead(157643384L, <ctypes.c_char_Array_20480 object at 0x9556c44>, 20480, 'c_ulong(7L)') -> 0
'3.2883\x00'
>>> pla.query_ascii_values('MEASure:VOLTage?')
2015-02-18 10:22:53,551 - pyvisa - DEBUG - viWrite(157643384L, 'MEASure:VOLTage?\r\n', 18, 'c_ulong(18L)') -> 0
2015-02-18 10:22:53,551 - pyvisa - DEBUG - GPIB0::5::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-02-18 10:22:54,154 - pyvisa - DEBUG - viRead(157643384L, <ctypes.c_char_Array_20480 object at 0x955689c>, 20480, 'c_ulong(7L)') -> 0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 433, in query_ascii_values
    return from_ascii_block(block, converter, separator, container)
  File "/usr/lib/python2.6/site-packages/pyvisa/util.py", line 195, in from_ascii_block
    return container(converter(raw_value) for raw_value in data)
  File "/usr/lib/python2.6/site-packages/pyvisa/util.py", line 195, in <genexpr>
    return container(converter(raw_value) for raw_value in data)
UnicodeEncodeError: 'decimal' codec can't encode character u'\x00' in position 6: invalid decimal Unicode string
```

hgrecco, 2015-02-19:
Using [pyvisa-sim](http://pyvisa-sim.readthedocs.org/), I simulated your instrument. Basically I wrote the following content to a file named `issue120.yaml`
```
spec: "1.0"
devices:
    issue120:
        eom:
          ASRL INSTR:
            q: "\r\n"
            r: "\x00"
        dialogues:
            - q: "MEASure:VOLTage?"
              r: "3.8050"
resources:
    ASRL10::INSTR:
        device: issue120
```
And then the following python script in the same folder to test:
```python
import visa
rm = visa.ResourceManager('issue120.yaml@sim')
print(rm.list_resources())
inst = rm.open_resource('ASRL10::INSTR')
inst.read_termination = '\x00'

print('Without parsing:')
inst.write('MEASure:VOLTage?')
print(repr(inst.read_raw()))
print('Parsing:')
print(inst.query_ascii_values('MEASure:VOLTage?'))
```

And it works in Python 2.7 and Python 3.2+. It might be a Python 2.6 problem. Can you try it with 2.7?

hgrecco, 2015-02-19:
And please try in 2.6 the following code:
```python
>>> pla.write('MEASure:VOLTage?')
(18L, <StatusCode.success: 0>)
>>> pla.read()
'3.8050\x00'
```
(Notice is read not read_raw)

bessman, 2015-02-19:
Same problem with Python 2.7:
```
>>> pla.query_ascii_values('MEASure:VOLTage?')
2015-02-19 09:45:29,686 - pyvisa - DEBUG - viWrite(167125160L, 'MEASure:VOLTage?\r\n', 18, 'c_ulong(18L)') -> 0
2015-02-19 09:45:29,687 - pyvisa - DEBUG - GPIB0::5::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-02-19 09:45:30,290 - pyvisa - DEBUG - viRead(167125160L, <ctypes.c_char_Array_20480 object at 0x9ea3b24>, 20480, 'c_ulong(7L)') -> 0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 434, in query_ascii_values
    return from_ascii_block(block, converter, separator, container)
  File "/usr/local/lib/python2.7/site-packages/pyvisa/util.py", line 195, in from_ascii_block
    return container(converter(raw_value) for raw_value in data)
  File "/usr/local/lib/python2.7/site-packages/pyvisa/util.py", line 195, in <genexpr>
    return container(converter(raw_value) for raw_value in data)
UnicodeEncodeError: 'decimal' codec can't encode character u'\x00' in position 6: invalid decimal Unicode string
```
Python 3.3 has a slightly different error message, but ultimately fails too:
```
>>> pla.query_ascii_values('MEASure:VOLTage?')
2015-02-19 09:58:48,390 - pyvisa - DEBUG - viWrite(155965272, b'MEASure:VOLTage?\r\n', 18, 'c_ulong(18)') -> 0
2015-02-19 09:58:48,391 - pyvisa - DEBUG - GPIB0::5::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-02-19 09:58:48,993 - pyvisa - DEBUG - viRead(155965272, <ctypes.c_char_Array_20480 object at 0xb71018e4>, 20480, 'c_ulong(7)') -> 0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.3/site-packages/pyvisa/resources/messagebased.py", line 434, in query_ascii_values
    return from_ascii_block(block, converter, separator, container)
  File "/usr/local/lib/python3.3/site-packages/pyvisa/util.py", line 195, in from_ascii_block
    return container(converter(raw_value) for raw_value in data)
  File "/usr/local/lib/python3.3/site-packages/pyvisa/util.py", line 195, in <genexpr>
    return container(converter(raw_value) for raw_value in data)
ValueError: could not convert string to float: '3.9922\x00'
```
read() gives the following output (with Python 2.6):
```
>>> pla.read()
2015-02-19 09:21:39,687 - pyvisa - DEBUG - GPIB0::5::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-02-19 09:21:39,800 - pyvisa - DEBUG - viRead(157643384L, <ctypes.c_char_Array_20480 object at 0x9556c44>, 20480, 'c_ulong(7L)') -> 0
u'3.6157\x00'
```
Python 2.7:
```
>>> pla.read()
2015-02-19 09:45:51,522 - pyvisa - DEBUG - GPIB0::5::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-02-19 09:45:51,523 - pyvisa - DEBUG - viRead(167125160L, <ctypes.c_char_Array_20480 object at 0x9ea34ac>, 20480, 'c_ulong(7L)') -> 0
u'3.8080\x00'
```
Python 3.3:
```
>>> pla.read()
2015-02-19 10:00:15,543 - pyvisa - DEBUG - GPIB0::5::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-02-19 10:00:15,613 - pyvisa - DEBUG - viRead(155965272, <ctypes.c_char_Array_20480 object at 0xb74f83d4>, 20480, 'c_ulong(7)') -> 0
'4.0103\x00'
```

bessman, 2015-02-19:
Waitwaitwait, hold on. I just noticed you did this:
```
inst.read_termination = '\x00'
```
If I do that, it works, with all versions of Python. I was under the impression that query_ascii_values() should autodetect the termination and Just Work™. Is that not the case?

hgrecco, 2015-02-19:
How should autodetection works? It depends on the instrument and can be ANY character. There is no universal value so you need to provide it as I showed, or during opening:
```python
inst = rm.open_resource('ASRL10::INSTR', read_termination = '\x00')
```

bessman, 2015-02-19:
OK, then I'll close this. Thanks for the help!

closed 2015-02-19 by bessman

hgrecco, 2015-02-19:
You are welcome. I am happy because it allowed me to use PyVISA-sim to test something with an equipment that I do not have. So thank you!