pyvisa/pyvisa #121: With cx_Freeze - TypeError: unexpected keyword argument

Chickenmarkus, 2015-02-20:
Sorry but I have a new problem. It shouldn't be a bug of pyVISA, only a strange interaction with cx_Freeze. Over python console everything works fine but frozen I get a TypeError.

debug.py:
```
import visa

def main():
    visa.log_to_screen()
    rm = visa.ResourceManager()
    # works
    instr = rm.open_resource("GPIB0::12::INSTR")
    # fails in frozen executable
    instr = rm.open_resource("GPIB0::12::INSTR", write_termination="\r\n")
if __name__=="__main__":
    main()
```

setup.py:
```
from cx_Freeze import setup, Executable
import os, site

[python_dir, site_dir] = site.getsitepackages()
include_files = []
missing_dll = ["msvcr100.dll"]
for dll in missing_dll:
    include_files.append((os.path.join(python_dir, dll), dll))

buildOptions_exe = dict(
    compressed = False,
    # includes = ["pyvisa"],       # doesn't solve this issue
    include_files = include_files
    )

setup(
    name = "Debugger",
    author = "Name",
    version = "0.0.0",
    options = dict(build_exe = buildOptions_exe),
    executables = [Executable("debug.py"),]
```

Execution of frozen script:
```
O:\>build\exe.win32-3.3\debug.exe
2015-02-06 09:51:38,270 - pyvisa - DEBUG - Created library wrapper for C:\Windows\system32\visa32.dll
2015-02-06 09:51:38,290 - pyvisa - DEBUG - viOpenDefaultRM(<cparam 'P' (02BCDA58)>,) -> 0
2015-02-06 09:51:38,290 - pyvisa - DEBUG - Created ResourceManager with session8240592
2015-02-06 09:51:38,360 - pyvisa - DEBUG - viOpen(8240592, 'GPIB0::12::INSTR', 0, 0, <cparam 'P' (02BCDA58)>) -> 1073676413
O:\build\exe.win32-3.3\library.zip\pyvisa\ctwrapper\functions.py:1059: VisaIOWarning: VI_SUCCESS_DEV_NPRESENT (1073676413): Session opened successfully, but the device at the specified address is not responding.
Traceback (most recent call last):
  File "C:\Python\WinPython-32bit-3.3.2.3\python-3.3.2\lib\site-packages\cx_Freeze\initscripts\Console3.py", line 27, in <module>
  File "debug.py", line 14, in <module>
    main()
  File "debug.py", line 12, in main
    instr = rm.open_resource("GPIB0::12::INSTR", write_termination="\r\n")
TypeError: open_resource() got an unexpected keyword argument 'write_termination'
```

```write_termination``` is only an example. The same bevavior is with other keyword arguments.
Maybe you have an idea where this redefinition come from?

hgrecco, 2015-03-05:
Sorry for the late response. This is very strange. I have no experience with cx_Freeze but `open_resource` has a `**kwargs` argument. So it should accept everything.  (If a certain argument is not valid is rejected later by the resource) raising a ValueError

Chickenmarkus, 2015-03-06:
Could be that you have an idea. I will leave a comment if I unexpectedly found a solution.

hgrecco, 2015-03-06:
No problem. PyVISA does not have any magic inside, it is mostly standard python. The only "dynamic" code (if you allow me that word) is the registration of the Resource sublcasses (which is also done in a very simple way). Leave it open, and if you think is useful ask in the cx_Freeze forums linking to this issue. I will be happy to help to make this work.

hgrecco, 2015-05-19:
I am closing this. Feel free to reopen if needed.

closed 2015-05-19 by hgrecco