pyvisa/pyvisa #122: No bitness for NI-VISA backend in Mac OS

dalonsoa, 2015-03-04:
Hello, 

I am trying to use PyVISA with the National Instruments VISA backend and it appears that the witness of Python and PyVISA don't agree. I've run **python3.4 -m visa info** to check what is going on, I've seen in other posts and this is the output:

2015-03-04 18:27:54.031 Python[800:63868] Error loading /Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport:  dlopen(/Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport, 265): no suitable image found.  Did find:
        /Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport: no matching architecture in universal wrapper
Machine Details:
   Platform ID:    Darwin-14.1.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Library/Frameworks/Python.framework/Versions/3.4/bin/python3.4
   Version:        3.4.2
   Compiler:       GCC 4.2.1 (Apple Inc. build 5666) (dot 3)
   Bits:           64bit
   Build:          Oct  5 2014 20:42:22 (#v3.4.2:ab2c023a9432)
   Unicode:        UCS4

PyVISA Version: 1.6.3

Backends:
   ni:
      Version: 1.6.3 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: n/a
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136

----

It appears that the backend has not defined bitness. NI only offer one single packages for Mac and it does not give me any choice during the installation, so I don't know how to proceed now. The installation of National Instruments tools is OK, I can communicate with the instruments with no problem. It is simply it does not talk to Python, or the other way round. 

Any ideas on how to solve this?

Best wishes, 
Diego

hgrecco, 2015-03-04:
It does work for me:
```
Machine Details:
   Platform ID:    Darwin-14.1.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Users/username/anaconda/bin/python
   Version:        3.4.2
   Compiler:       GCC 4.2.1 (Apple Inc. build 5577)
   Bits:           64bit
   Build:          Oct 21 2014 17:42:20 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.3

Backends:
   ni:
      Version: 1.6.3 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: user
         bitness: 32
         Could not get more info: Interpreter and library have different bitness.
```

And with a 32 bit Python
```
Machine Details:
   Platform ID:    Darwin-14.1.0-x86_64-i386-32bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Users/grecco/envs/lantz/bin/python
   Version:        3.2.3
   Compiler:       GCC 4.2.1 (Apple Inc. build 5666) (dot 3)
   Bits:           32bit
   Build:          Apr 10 2012 11:25:50 (#v3.2.3:3d0686d90f55)
   Unicode:        UCS2

PyVISA Version: 1.6.3

Backends:
   ni:
      Version: 1.6.3 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: user
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5243392
         Spec. Version: 5243136
```
I guess those warnings are interfering with the ability to infer the bitness. Please run the following command in the command line:
```
file /Library/Frameworks/visa.framework/visa
````

But what I can tell you is that in general you should use a 32 bit Python to talk to VISA

dalonsoa, 2015-03-04:
Hi, 

The output is not very enlightening:

/Library/Frameworks/visa.framework/visa: symbolic link to `Versions/Current/VISA'

If I run the command on the file the symbolic link is point to, I get:

VISA: 2 [i386: Mach-O i386 dynamically linked shared library] [x86_64]

I hope this tells you more than it tells me!

hgrecco, 2015-03-04:
It seems to me that your `file` command is not standard. Usually it should follow symbolic links. Try:
```
file -L /Library/Frameworks/visa.framework/visa
```

dalonsoa, 2015-03-04:
Now it follows the symbolic link:

/Library/Frameworks/visa.framework/visa: 2 [i386: Mach-O i386 dynamically linked shared library] [x86_64]

hgrecco, 2015-03-04:
The temporary solution is to create a `.pyvisarc` file as described here but pointing to the real file.

dalonsoa, 2015-03-04:
OK, thanks... but where is "here"?

hgrecco, 2015-03-04:
ups!
http://pyvisa.readthedocs.org/en/master/configuring.html#configuring-the-ni-backend

dalonsoa, 2015-03-04:
OK, I've just seen the instructions. I'll check it.

Thanks!

dalonsoa, 2015-03-04:
Apparently, it does not work. I got two backends but none of them with a bitness value:

----

Backends:
   ni:
      Version: 1.6.3 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/Versions/Current/VISA:
         found by: user
         bitness: n/a
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
      #2: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: n/a
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136

----

Any other suggestion?

hgrecco, 2015-03-04:
What happens when you try to use pyvisa? (not just get the debug info, but actually use it)

hgrecco, 2015-03-04:
By the way, just to get all the info. Can you paste the output of `file --version`?

dalonsoa, 2015-03-05:
About the first question, here you have the output for just getting the list of resources:

----

>>> import visa
>>> rm = visa.ResourceManager()
2015-03-05 06:20:43.144 Python[3245:15590] Error loading /Library/Frameworks/VISA.framework/Frameworks/RemotePassport.framework/RemotePassport:  dlopen(/Library/Frameworks/VISA.framework/Frameworks/RemotePassport.framework/RemotePassport, 265): no suitable image found.  Did find:
	/Library/Frameworks/VISA.framework/Frameworks/RemotePassport.framework/RemotePassport: no matching architecture in universal wrapper
2015-03-05 06:20:43.147 Python[3245:15590] Error loading /Library/Frameworks/VISA.framework/Frameworks/SerialPassport.framework/SerialPassport:  dlopen(/Library/Frameworks/VISA.framework/Frameworks/SerialPassport.framework/SerialPassport, 265): no suitable image found.  Did find:
	/Library/Frameworks/VISA.framework/Frameworks/SerialPassport.framework/SerialPassport: no matching architecture in universal wrapper
2015-03-05 06:20:43.151 Python[3245:15590] Error loading /Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport:  dlopen(/Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport, 265): no suitable image found.  Did find:
	/Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport: no matching architecture in universal wrapper
2015-03-05 06:20:43.152 Python[3245:15590] Error loading /Library/Frameworks/VISA.framework/Frameworks/RemotePassport.framework/RemotePassport:  dlopen(/Library/Frameworks/VISA.framework/Frameworks/RemotePassport.framework/RemotePassport, 265): no suitable image found.  Did find:
	/Library/Frameworks/VISA.framework/Frameworks/RemotePassport.framework/RemotePassport: no matching architecture in universal wrapper
2015-03-05 06:20:43.152 Python[3245:15590] Error loading /Library/Frameworks/VISA.framework/Frameworks/SerialPassport.framework/SerialPassport:  dlopen(/Library/Frameworks/VISA.framework/Frameworks/SerialPassport.framework/SerialPassport, 265): no suitable image found.  Did find:
	/Library/Frameworks/VISA.framework/Frameworks/SerialPassport.framework/SerialPassport: no matching architecture in universal wrapper
2015-03-05 06:20:43.152 Python[3245:15590] Error loading /Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport:  dlopen(/Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport, 265): no suitable image found.  Did find:
	/Library/Frameworks/VISA.framework/Frameworks/PxiPassport.framework/PxiPassport: no matching architecture in universal wrapper

----

About the second question: file --version

file-5.17
magic file from /opt/local/share/misc/magic

hgrecco, 2015-03-05:
Then PyVISA is working. But if, of course, only able to talk to instruments over the interfaces supported by your current VISA installation.

dalonsoa, 2015-03-05:
I don't get it. What do you mean by "is working"? If there is no devices it can talk to, shouldn't the above  command work and then produce an empty list when calling list_resources, and not an error?

hgrecco, 2015-03-05:
I think then you forgot to paste something. There is not call to `list_resources`

But I what I wanted to say is that PyVISA is opening correctly the library. Those warnings (issued by NI-VISA, not PyVISA) are telling you is that certain parts (called passports) of NI-VISA are not available either because you have not installed or because the architecture is wrong.

Have you checked what is supported by NI-VISA in your architecture and OS?

dalonsoa, 2015-03-05:
OK, I think I understand. list_resources shows, indeed, an empty tuple, as expected (since I don't have anything connected, at the moment). 

What I don't understand, now is why, having pretty much the same configuration, for what I can see above, yours works, giving the proper bitness, and mine does not.

hgrecco, 2015-03-05:
There are two issues here. 
1.- PyVISA is not able to find the bitness of your library. This is probably a difference in `file` and I can fix it easily by adding [this package](https://pypi.python.org/pypi/macholib/) to PyVISA. I was trying to avoid it because `file` was working fine till now
2.- Why I can use NI-VISA in my OS X without any problem. I am using a 32 bit NI-VISA installation (which contains all passports) with a 32 bit Python.

dalonsoa, 2015-03-05:
Thanks for this instructive explanation. I will try to figure out how to use a 32 bit NI-VISA and python in a 64 bit computer without messing anything else.

dalonsoa, 2015-03-05:
Done! 

Running python in 32 bit mode ($ arch -32 python3.4), visa works perfectly fine with no errors:

>>> import visa
>>> rm = visa.ResourceManager()
>>> rm.list_resources()
('ASRL1::INSTR', 'ASRL2::INSTR', 'ASRL3::INSTR')

Thanks for the indications. Cheers!

hgrecco, 2015-03-05:
Happy to help.

closed 2015-03-05 by hgrecco