pyvisa/pyvisa #123: Pyvisa to find devices in Usbserial on Mac OS X

thoo, 2015-03-05:
Hi , 
I cannot find my device by simply running this. 
```
import visa
rm = visa.ResourceManager()
rm.list_resources()
```
I can use **pyserial** with address ` /dev/cu.usbserial-FTDGQUTZ` . Can I use the same address like pyserial ?  Can I force pyvisa to search in specific locations? Thanks,

hgrecco, 2015-03-05:
The NI backend only sees the devices seen by NI-VISA. So if it appears in NI-MAX automation and explorer it will appear in PyVISA with the NI-backend. And in any case, if it does not appear in the list you can always try to open the resource directly by providing the correct resource name. Maybe `ASRL/dev/cu.usbserial-FTDGQUTZ` works, but I do not know.

thoo, 2015-03-06:
@hgrecco Thanks. I found out it was the issue between my python (Anaconda 64bit) and pyvisa. Now I install canopy 32bit and everything is working.
Cheers,

hgrecco, 2015-03-06:
Great.

closed 2015-03-06 by hgrecco