pyvisa/pyvisa #124: could not make pyvisa work for Anaconda 64 in windows 7

V33RA, 2015-03-13:
Hi I'm new to python, I'm trying to communicate with the instruments. I installed PyVISA from https://github.com/hgrecco/pyvisa/zipball/master and ran the following in Anaconda command prompt

$ python setup.py install

Then I typed "import visa" in the spyder console and it seemed to go through, then I typed rm = visa.ResourceManager(), and I see the following error.:

In [3]: rm = visa.ResourceManager()
Traceback (most recent call last):

  File "<ipython-input-3-36edb213cd9c>", line 1, in <module>
    rm = visa.ResourceManager()

  File "C:\Anaconda\lib\site-packages\pyvisa-1.6.4.dev0-py2.7.egg\pyvisa\highlevel.py", line 1477, in __new__
    visa_library = open_visa_library(visa_library)

  File "C:\Anaconda\lib\site-packages\pyvisa-1.6.4.dev0-py2.7.egg\pyvisa\highlevel.py", line 1452, in open_visa_library
    raise e

OSError: Could not open VISA library:

Other things I did: I checked the versions, Anaconda version I'm using is 64 bit and NI-VISA 14.0 documentation says it can support both 32 and 64 bit in windows 7. And I ran [python -c "from pyvisa import util; util.get_debug_info()"] too (took this from another issue posted) and the output was:

C:\Anaconda>python -c "from pyvisa import util; util.get_debug_info()"
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda\python.exe
   Version:        2.7.7
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Jun 11 2014 10:40:02 (#default)
   Unicode:        UCS2

PyVISA Version: 1.6.4.dev0

Backends:
   ni:
      Version: 1.6.4.dev0 (bundled with PyVISA)
      Binary library: Not found

I don't know what i'm doing wrong. Any solution?

Thanks in advance.

V33RA, 2015-03-13:
Never mind: 

"PyVISA expects a file called visa32.dll in the PATH . For example, on my system you find this file in C:\WINNT\system32\ . Either copy it there or expand your PATH. Alternatively, you can create an INI file. You must do this anyway if the file is not called visa32.dll on your system."

In my computer the path is C:\Windows\system32\ and I added that to the 'environment variables' in my computer>properties>Advanced system settings> Advanced tab

Felt like an idiot.

\\I wish there is a way to detele the issues posted,tried to add images but i posted it as new issue.\\

hgrecco, 2015-03-25:
No problem. I am closing this Feel free to reopen.

closed 2015-03-25 by hgrecco