pyvisa/pyvisa #127: pyvisa 1.6.1 can not set stopbits

StachowP, 2015-03-26:
Hi,
i have a problem with pyvisa 1.6.1 and the RS232 setting.
my device requires two stop bits. But the syntax in phyton "device.stop_bits=2" returns an error "ValueError: 2 is an invalid value for attribute VI_ATTR_ASRL_STOP_BITS, should be a <enum 'StopBits'>".

hgrecco, 2015-03-27:
You should use the constants In the current dev version,
```
from pint import constants
device.stop_bits = constants.VI_ASRL_STOP_TWO
````
or 
```
from pint import constants
device.stop_bits = constants.StopBits.two
```

In any case, we should improve the error message.

StachowP, 2015-03-27:
I installed the pint 0.6 version. Now the Syntax "from pint Import constants" returns an error "ImportError: cannot import name constants" Did i build-in the wrong Pint version?

hgrecco, 2015-03-27:
Sorry. it should have said 
```
from visa import constants
```
or (in older pyvisa versions)
```
from pyvisa import constants
```

StachowP, 2015-03-27:
Still does not work.
File "C:/Workspaces/spyder/test/helloagilent.py", line 2, in <module>
    from visa import constants
ImportError: cannot import name constants

hgrecco, 2015-03-28:
As mentioned before, `from visa` only works in more recent versions of pyvisa

hgrecco, 2015-04-11:
I am closing this. Feel free to reopen.

closed 2015-04-11 by hgrecco