pyvisa/pyvisa #129: Not very helpful error when resource is not found.

inuyasha129, 2015-03-30:
Hello,

I was trying to connect to a serial device with address "ASRL4::INSTR", but I ended up getting the error below. I ended up printing ret from functions.py:1365 and found out the actual problem from the error code which was VI_ERROR_RSRC_NFOUND. The device couldn't be found, so interface_type ended up being 0. Perhaps there should be a check before doing constants.InterfaceType(interface_type.value).

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-22-4de3db13ed0b> in <module>()
----> 1instr = rm.open_resource("ASRL4::INSTR", baud_rate=9600, data_bits=8, stop_bits=pyvisa.constants.StopBits.one, parity=pyvisa.constants.Parity.none)

/Users/test/anaconda/lib/python2.7/site-packages/pyvisa/highlevel.pyc in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1616         :rtype: :class:`pyvisa.resources.Resource`
   1617         """
-> 1618         info = self.resource_info(resource_name)
   1619
   1620         try:

/Users/test/anaconda/lib/python2.7/site-packages/pyvisa/highlevel.pyc in resource_info(self, resource_name)
   1583         :rtype: :class:`pyvisa.highlevel.ResourceInfo`
   1584         """
-> 1585         ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
   1586         if err == constants.StatusCode.success:
   1587             return ret

/Users/test/anaconda/lib/python2.7/site-packages/pyvisa/ctwrapper/functions.pyc in parse_resource_extended(library, session, resource_name)
   1376         res[-1] = None
   1377
-> 1378     return ResourceInfo(constants.InterfaceType(interface_type.value),
   1379                         interface_board_number.value, *res), ret
   1380

/Users/test/anaconda/lib/python2.7/site-packages/enum/__init__.pyc in __call__(cls, value, names, module, type)
    328         """
    329         if names is None:  # simple value lookup
--> 330             return cls.__new__(cls, value)
    331         # otherwise, functional API: we're creating a new Enum type
    332         return cls._create_(value, names, module=module, type=type)

/Users/test/anaconda/lib/python2.7/site-packages/enum/__init__.pyc in __new__(cls, value)
    640             if member.value == value:
    641                 return member
--> 642     raise ValueError("%s is not a valid %s" % (value, cls.__name__))
    643 temp_enum_dict['__new__'] = __new__
    644del __new__

ValueError: 0 is not a valid InterfaceType

hgrecco, 2015-03-31:
agreed.

hgrecco, 2015-03-31:
can you try now with the latest dev.

inuyasha129, 2015-04-01:
Working great! Thanks for fixing ;)

closed 2015-04-02 by inuyasha129