pyvisa/pyvisa #130: using PXI davices with pyvisa

PKTcad, 2015-03-31:
Hello hgrecco,
I hope this is the right way to ask for help with pyvisa.

I am trying to operate a pxi device using pyvisa. i can see the devices in the 
visa.ResourceManager()
an can connect to them using 
rm.open_resource("PXI3::15")
but i am unable to communicate with the device.
I wasent able to find any working example for pyvisa using a pxi device.

any help would be appreciated
PKTcad

hgrecco, 2015-03-31:
I haven't used a PXI with PyVISA but I will be happy to help you if I can. Can you describe what do you mean by "i am unable to communicate with the device." What is the error message that you get?

PKTcad, 2015-03-31:
Thank you for the fast reply.

i am not getting any error message.
i do not understand the command structure to use a PXI card with PyVISA.
i have 4 different cards is a pxi chassis they are connected via a PXI to PCIe card.
with a different two wrapper i was able to communicate with 3 of them unsing python.
but now i am trying to use PyVISA to communicate with all of them.

therefore searching for an example to switch a relays on the matrix card or measure a voltage with the multimeter. or any other basic command. 

used Cards:
NI PXI-4071
NI PXI-6528
Pickering 40-199-002
Pickering 40-587A-001

closed 2015-04-22 by PKTcad

hgrecco, 2015-04-22:
The structure is the same as talking to PXI using C. From PyVISA you can access any of the `vi*` functions of the VISA library, and additionally you have a Python friendly wrapper. I would say that if you get a C example and you follow through you shoudl be able to make it work.