pyvisa/pyvisa #131: library problem with Cygwin_x64 on Win7_x64

fishersa, 2015-03-31:
I am having trouble loading the binary library on a cygwin installation of python.  When I include a path to visa32.dll (in either the .pyvisarc file or with ResourceManager()), I get no output besides a "Segmentation fault (core dumped)" message.

Is pyvisa assuming a linux platform?

I have VISA 5.4.1 installed

$ python -m visa info
Machine Details:
   Platform ID:    CYGWIN_NT-6.1-1.8.6-2-0.280-5-3-x86_64-64bit
   Processor:

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.8
   Compiler:       GCC 4.8.3
   Bits:           64bit
   Build:          Jul 25 2014 14:04:36 (#default)
   Unicode:        UCS2

PyVISA Version: 1.6.3

Backends:
   ni:
      Version: 1.6.3 (bundled with PyVISA)
      Binary library: Not found

hgrecco, 2015-03-31:
> Is pyvisa assuming a linux platform?

Not at all. pyvisa works on windows, linux and mac. I have used it on Windows XP, 7 and 8 with Python downloaded from python.org and the Anaconda Python distribution. I have never used it within cygwin.

fishersa, 2015-04-01:
I meant does it interpret a cygwin installation as linux, and then look for a .so library?

Can you give me any hints on how to debug?

I am using redhat's Cygwin installed with ftp://ftp.ges.redhat.com/cygwin/x86_64/rhsetup-x86_64.exe

I have used pyvisa 1.4 with cygwin_x32 a few years ago, but back then we loaded the library by hand.

hgrecco, 2015-04-02:
Instantiating a ResourceManager with the NI backend (default) is a two stage process:
1. **the library file is located**: It can be done with a user provided argument, a configuration file (pyvisarc) or via the OS. Only in this last way, the OS matters. In other words if you run `ResourceManager('whatever')` it will try to open `whatever` in spite of the OS
2. **opening the library**: this depends on the OS. Basically the important part of the code is:
```
if _os.name == 'nt':
    FUNCTYPE, Library = _ctypes.WINFUNCTYPE, _ctypes.WinDLL
else:
    FUNCTYPE, Library = _ctypes.CFUNCTYPE, _ctypes.CDLL
```
So in cygwin it should take the `False` path. So I do not really get why it does not work. Is it possible that the bitness of the library and python interpreter do not match?

Ps.- Why are you using cygwin?

fishersa, 2015-04-02:
The use of cygwin is historical.

The segfault is occurring in functions.py on this line (inside method set_signatures):
if not hasattr(library, '_functions'):

library is:
<CDLL 'C:/Windows/System32/visa32.dll', handle 7fef7c40000 at 6ffff864d10>

Does this give you any ideas to debug further?

Thanks for your help.

fishersa, 2015-04-02:
library is:
<CDLL 'C:/Windows/System32/visa32.dll', handle 7fef7c40000 at 6ffff864d10>

fishersa, 2015-04-02:
library is (not including quotes):

"<CDLL 'C:/Windows/System32/visa32.dll', handle 7fef7c40000 at 6ffff864d10>"

hgrecco, 2015-04-02:
It is really strange because that line should not cause any problem. Just to check, do the following:
```python
>>> import ctypes
>>> path = '/Library/Frameworks/visa.framework/visa' # Here put your path
>>> ctypes.cdll.LoadLibrary(path)
<CDLL '/Library/Frameworks/visa.framework/visa', handle 3192d0 at 2844a2c>
>>> lib = ctypes.cdll.LoadLibrary(path)
>>> hasattr(lib, '_functions')
False
>>> lib._functions = []
>>> hasattr(lib, '_functions')
True
```

Can you add the following line inside `set_signature` (Notice it is not `set_signatures`):
```python
print(library, function_name, argtypes, restype, errcheck)
```

fishersa, 2015-04-03:
$ python
Python 2.7.8 (default, Jul 25 2014, 14:04:36)
[GCC 4.8.3] on cygwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes
>>> path = 'C:/Windows/System32/visa32.dll'
>>> ctypes.cdll.LoadLibrary(path)
\<CDLL 'C:/Windows/System32/visa32.dll', handle 7fefb480000 at 6ffffe61b90\>
>>> lib = ctypes.cdll.LoadLibrary(path)
>>> print(lib)
\<CDLL 'C:/Windows/System32/visa32.dll', handle 7fefb480000 at 6ffffe61850\>
>>> print(dir(lib))
['_FuncPtr', '__class__', '__delattr__', '__dict__', '__doc__', '__format__', '__getattr__', '__getattribute__', '__getitem__', '__hash__', '__init__', '__module__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_func_flags_', '_func_restype_', '_handle', '_name']
>>> hasattr(lib, '_functions')
Segmentation fault (core dumped)

The first call to set_signature is after the segfault.  I added print(library) and print(dir(library)) before the hasattr in set_signatures:
\<CDLL 'C:/Windows/System32/visa32.dll', handle 7fefb480000 at 6ffff822d10\>
['_FuncPtr', '__class__', '__delattr__', '__dict__', '__doc__', '__format__', '__getattr__', '__getattribute__', '__getitem__', '__hash__', '__init__', '__module__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_func_flags_', '_func_restype_', '_handle', '_name']

hgrecco, 2015-04-03:
The `_functions` attribute is used to store the name of the functions handled by the library without resorting to global variables. I wonder if cygwin has a problem with this (it works in windows, linux and mac). We could rewrite it to store it someplace else.

fishersa, 2015-04-03:
Here is a workaround for me that appears to work:
#if not hasattr(library, '_functions'):
if not '_functions' in dir(library):

What do you think?

I can't fully test this yet.  Maybe not until Monday.

Someone else reported segfaults with ctypes and cygwin as well:
https://www.cygwin.com/ml/cygwin/2014-11/msg00558.html

Thanks.

fishersa, 2015-04-06:
Everything work great in the lab with the workaround for a segfault from hasattr (noted in the comment above - the change is in file ctwrapper/functions.py).
This is required for RedHat Cygwin x64 on a Win7 machine.

Thanks for the help.

closed 2015-04-06 by fishersa

hgrecco, 2015-04-06:
would you mind submitting a PR. Please also include a comment on the code (very) briefly commenting the problem as if somebody sees `if not '_functions' in dir(library)` for sure will flag it as ugly code! :-D