pyvisa/pyvisa #132: GPIB/Serial/TCPIP Instrument Drivers for pyvisa 1.6

sc-splitter, 2015-03-31:
I'm having trouble with my old drivers that I used when on pyvisa 1.4.

Before, each instrument was it's own class (e.g. - E3631A(pyvisa.visa.GpibInstrument)), allowing me to write specific functions for each different instrument. Now, I can't quite figure out how to accomplish the same with pyvisa 1.6.

I've searched around and found precious little info on how to do this. Everything I've seen simply says to throw everything I've got away and use the rm.open_resource() line. That would be fine if I was starting from scratch, but I have about 15 different drivers that I've written for various instruments, each of which have many dedicated functions. Not to mention all of the scripts that I've written using said functions.

Is there a way for me to modify my current drivers for 1.6, or should I simply revert back to 1.4? I'd like to use some of the capabilities of 1.6, but it's probably not worth it if I'll have to re-write every program I've ever written.

sc-splitter, 2015-04-02:
@hgrecco, do you have any suggestions on how I can do this? I'd really appreciate your help.

hgrecco, 2015-04-02:
If you do not want to change anything in your code, then the only way is to keep using 1.4. 

If you want to upgrade to 1.6, the best route is to subclass. It is not that hard.

Basically you do:
```python
class  MyClass(object):

    def __init__(self, resource):
        self.resource = resource

    def write(self, *args, **kwargs):
        return self.resource.write(*args, **kwargs)

    def read(self, *args, **kwargs):
        return self.resource.read(*args, **kwargs)

    def query(self, *args, **kwargs):
        return self.resource.query(*args, **kwargs)

    # here goes all the methods that were in your old class.

```

and then you use it as:

```python
import visa
rm = visa.ResourceManager()
inst = MyClass(rm.open_resource('GPIB::8'))
```

and that is all.

(you might need to change a few other things, for example `ask` is now `query`. Read the migration guide in the docs)

Feel free to write again if you have any problem.

hgrecco, 2015-04-11:
I am closing this. Feel free to reopen.

closed 2015-04-11 by hgrecco