pyvisa/pyvisa #134: No error thrown when resource doesn't exist

timrae, 2015-04-14:
I just updated to v1.6.1.2 from something much older (not sure which version sorry), and while I used to get an exception when I opened an instrument that wasn't switched on, but I now get a `GPIBInstrument` object back without any failures:

```
rm=pyvisa.ResourceManager()
rm.open_resource("GPIB::15")
<'GPIBInstrument'(u'GPIB0::15::INSTR')>
```

```
>>> rm.list_resources()
(u'ASRL1::INSTR', u'ASRL10::INSTR', u'ASRL15::INSTR', u'GPIB0::20::INSTR', u'GPIB0::25::INSTR')
```

Querying the instrument leads to the following error:

```
>>> i.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Tim-Lab\AppData\Local\Enthought\Canopy\User\lib\site-packages\pyvisa\resources\messagebased.py", line 378, in query
    self.write(message)
  File "C:\Users\Tim-Lab\AppData\Local\Enthought\Canopy\User\lib\site-packages\pyvisa\resources\messagebased.py", line 184, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Users\Tim-Lab\AppData\Local\Enthought\Canopy\User\lib\site-packages\pyvisa\resources\messagebased.py", line 162, in write_raw
    return self.visalib.write(self.session, message)
  File "C:\Users\Tim-Lab\AppData\Local\Enthought\Canopy\User\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1854, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\Users\Tim-Lab\AppData\Local\Enthought\Canopy\User\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 180, in _return_handler
    raise errors.VisaIOError(ret_value)
VisaIOError: VI_ERROR_NLISTENERS (-1073807265): No listeners condition is detected (both NRFD and NDAC are deasserted).
```

Is there a way to re-enable the old behavior, or a better way to test if the instrument is available other than catching an exception to `query`? I've tried updating to the latest development version (56515d3d00543b37b44f35f41fbc53a2e7b7b34c) with no luck

hgrecco, 2015-05-10:
Sorry for the delay. PyVISA < 1.5 was raising an error because it was setting some parameters on the instrument when open was called. I prefer the new design because is cleaner and closer all other VISA APIs around. But you can still specify something to do after opening but in the same line. For example:
```
rm.open_resource("GPIB::15", read_termination='\n')
```

hgrecco, 2015-05-19:
I am closing this. Feel free to reopen if needed.

closed 2015-05-19 by hgrecco

timrae, 2015-08-12:
> `rm.open_resource("GPIB::15", read_termination='\n')`

This doesn't result in an error when resource doesn't exist either

hgrecco, 2015-08-12:
I have just found the same error after updating to the latest visa library and using it in 64 bits. There was a 32/64 bit problem (argg). I have just pushed a fix