pyvisa/pyvisa #136: strange error when failing to load a library

lupien, 2015-04-17:
When I was testing loading the resource manager with different visa library (using the agilent one instead of the NI one) I tried using a bad filename and got a strange error message (a UnicodeDecodeError exception)

This was on a windows 7 machine, running in french. I am using python 2.7

I tracked down the problem to LibraryError.from_exception (errors.py) which was called by NIVisaLibrary._init (ctwrapper/highlevel.py). In that class method, the unicode string ': image not found' is compared with the error message coming from windows that contains accents (codes >128). This comparison implies a conversion of my string using the 'ascii' codec and it fails.

I don't know what the fix should be.

I don't know where the message you compare too are coming from, but under windows and linux many error message can depend on the operating system language. On linux, I get french error messages doing:
    import locale, os
    locale.setlocale(locale.LC_ALL, 'fr_FR.utf8')
    f=open('/tmp/testtest')
    os.strerror(1)

hgrecco, 2015-04-17:
There are two aspects. First is how to diagnose an error from a library (to provide a better error message). As far as I know, there is no good way (such as an error code) and that is why we are parsing the error message. I agree that this is not a good idea in general because messages can change, they depend on the language, etc. But it seems to be the only way and therefore we have settled for this (and for using the english messages).

Second, the UnicodeDecodeError that you are pointing. That we should be able to fix. When we fix it, the comparisons will fall through (your messages are in french and the internal versions are in english) and therefore you will get a general error message: `Error while accessing <filename>: <here is the error message given by ctypes>`

Please paste the output of: `python -m visa info`  and the error message that you are getting (full with the stack)

lupien, 2015-04-19:
Here is the requested info:
```
C:\>python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 44 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.9
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          Dec 10 2014 12:24:55 (#default)
   Unicode:        UCS2

PyVISA Version: 1.6.3

Backends:
   ni:
      Version: 1.6.3 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5243137
         Spec. Version: 4194304
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5243137
         Spec. Version: 4194304
```

and here is the error:
```
>>> import visa
>>> rm = visa.ResourceManager('/testfile')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1477, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1452, in open_visa_library
    raise e
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe9 in position 24: ordinal not in range(128)
```
if I run it again I get a different error (the previous bad loading has been cached in the _registry)
```
>>> rm = visa.ResourceManager('/testfile')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1486, in __new__
    obj.session, err = visa_library.open_default_resource_manager()
  File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1212, in open_default_resource_manager
    ret = library.viOpenDefaultRM(byref(session))
AttributeError: 'NIVisaLibrary' object has no attribute 'viOpenDefaultRM'
>>> import pyvisa
>>> pyvisa.highlevel.VisaLibraryBase._registry
{(<class 'pyvisa.ctwrapper.highlevel.NIVisaLibrary'>, '/testfile'): <VisaLibrary('/testfile')>}
```

The first error comes from loading the library and is:
```
>>> try:
...     pyvisa.ctwrapper.highlevel.Library('/testfile')
... except Exception as exc:
...     pass
...
>>> exc
WindowsError(126, 'Le module sp\xe9cifi\xe9 est introuvable')
```

hgrecco, 2015-04-19:
I cannot reproduce. I wonder if this is a problem with your console. Can you try replacing:
```python
cls = pyvisa.ctwrapper.highlevel.Library
argument = '/testfile'
try:
    cls(argument)
except Exception as e:
    print('Could not open VISA wrapper %s: %s\n%s' % (cls, str(argument), e))
```

if this works try replacing the print statement by:
```python
logging.debug('Could not open VISA wrapper %s: %s\n%s' % (cls, str(argument), e))
```
and finally by
```python
logging.debug('Could not open VISA wrapper %s: %s\n%s', cls, str(argument), e)
```
(Remember to `import logging`)

lupien, 2015-04-20:
I got the same error message (126) and the same text in all the cases:
```
>>> import logging
>>> l=logging.getLogger()
>>> l.setLevel(logging.DEBUG)
>>> cls = pyvisa.ctwrapper.highlevel.Library
>>> argument = '/testfile'
>>> try:
...     cls(argument)
... except Exception as e:
...     print('Could not open VISA wrapper %s: %s\n%s' % (cls, str(argument), e))
...     logging.debug('Could not open VISA wrapper %s: %s\n%s' % (cls, str(argument), e))
...     logging.debug('Could not open VISA wrapper %s: %s\n%s', cls, str(argument), e)
...
Could not open VISA wrapper <class 'ctypes.WinDLL'>: /testfile
[Error 126] Le module spÚcifiÚ est introuvable
DEBUG:root:Could not open VISA wrapper <class 'ctypes.WinDLL'>: /testfile
[Error 126] Le module spÚcifiÚ est introuvable
DEBUG:root:Could not open VISA wrapper <class 'ctypes.WinDLL'>: /testfile
[Error 126] Le module spÚcifiÚ est introuvable
```

You will not be able to reproduce this message if your windows is not running in FRENCH.
Some versions of windows 7 (Ultimate, enterprise) can change the display language. Others cannot.

hgrecco, 2015-04-20:
I understand the issue about the language, but I was trying to figure out why you are getting an ugly error when you call `rm = visa.ResourceManager('/testfile')`  instead of failing graciously. You should get
```
Could not open VISA wrapper <class 'ctypes.WinDLL'>: /testfile
[Error 126] Le module spÚcifiÚ est introuvable
```
but you are getting
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1477, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1452, in open_visa_library
    raise e
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe9 in position 24: ordinal not in range(128)
```

lupien, 2015-04-20:
I think I understand your problem.
The UnicodeDecodeError is raised in the handling of the exception (in LibraryError.fromException of errors.py). The first if statement of the method does a comparison of a unicode string (because of `__future__` unicode_literals) with the string containing non ascii values.
I can reproduce the error from that first if statement with:
```
>>> u'sp' in 'sp\xe9cifi'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe9 in position 2: ordinal not in range(128)
```
Here is a way to create the error with the above method (I did this on linux but I believe it should also work on windows):
```
>>> import pyvisa
>>> pyvisa.errors.LibraryError.from_exception(OSError(126, 'sp\xe9'), '/testfile')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/pyvisa/errors.py", line 444, in from_exception
    if ': image not found' in msg:
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe9 in position 14: ordinal not in range(128)
```

And this produces the same error I showed (and is the equivalent of the wrapping of open_visa_library)
```
>>> try:
...     try:
...         raise OSError(126, 'sp\xe9')
...     except OSError as exc:
...         pyvisa.errors.LibraryError.from_exception(exc, '/testfile')
... except Exception as e:
...         raise e
... 
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe9 in position 14: ordinal not in range(128)
```
Note that if the last line was just `raise` instead of `raise e`, it would be easier to track down the problem (better traceback).

I have been thinking about a fix and I can suggest to either catch the UnicodeDecodeError in LibraryError.fromException or to detect a WindowsError instance (or OSError) and then use the errno instead of looking at the text.

Also, in case of failure to load the library, it should probably skip putting an entry in the _registry cache of VisaLibraryBase.

commit 4f428b24 (2015-08-24): Fix error reporting with Unicode filenames

Close #136
closed 2015-08-24 by hgrecco