pyvisa/pyvisa #140: bug in binary data writing

burnpanck, 2015-04-27:
In the function `pyvisa.util.to_ieee_block`, the header of an IEEE binary data block is computed as
```
array_length = len(iterable)
header = '%d' % array_length
data_length = array_length * element_length
return '#%d%d' % (len(header), data_length)
```
which is obviously wrong: the format should `#[number of digits][number of bytes]`, where `[number of digits]` is the number of digits it takes to write the `[number of bytes]` field, but this code obviously calculates how many digits it needs to write the number of elements. If the elements are multi-bytes, the number of digits may be larger.

hgrecco, 2015-04-27:
Is the `number of digits` allowed to be a multibyte number? I do not have the spec around to check.

burnpanck, 2015-04-27:
actually, the spec does not specify the interpretation of the data at all, it's just binary as a stream of bytes for the spec. it's up to the device to interpret this as a multibyte array. however, since there are devices that do so, it's cool that pyvisa supports this, i.e. we specify the data-dtype and pyvisa encodes this then for transfer. however it does so with a bug:
the header is all ascii. the proper encoding of `np.array([1,2,3],'u4')`(assuming little-endian) would be `#212\x01\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00` i.e. the header says:
 - '#': Binary data!
 - '2': I'm about to send you the length in bytes, that will require two decimals
 - '12': Here they are, it's 12 bytes.
The current code incorrectly concludes that it takes one decimal to announce '3' array elements, even though it then correctly announces 12 bytes. So the current code would output `#112...`, which will be interpreted as '1' decimal of announcement, '1' bytes of data of ascii value '2', and then the device chokes on the 12 bytes 'garbage' that follows.

hgrecco, 2015-04-27:
I see two problems with the current implementation.

The first is the one you mentioned. Basically, the header should be built with `data_length`, not `array_length`. That is a bug that we can (should) fix. (It would be great if you submit a PR) 

The second problem is how `[number of digits]` field is encoded. We could add an argument to specify this but I think is hard to get the API right without an actual example.

hgrecco, 2015-04-28:
Maybe it was not clear my second point. Can the header take non decimal number for `[number of digits]` (or for `[number of bytes]`). This is what I do not remember from the spec and I do not get from your comment. The first thing is definitely a bug and should be fixed. But what about the second?

burnpanck, 2015-04-28:
as far as I understand, the only possibly encoding for the two header fields is as an ascii string representing a decimal number. i'll do a PR as soon as I'm back on my development machine

hgrecco, 2015-04-28:
Great!. Thanks for taking care of this.

hgrecco, 2015-05-10:
Merged

closed 2015-05-10 by hgrecco