pyvisa/pyvisa #144: attributes handling missing entries and creating an empty entry

lupien, 2015-04-29:
Many of the attributes in pyvisa/attributes.py have an empty ('') py_name.

Is there a plan to give a name to all of them?

If not, then I would give one to some more important ones like at least AttrVI_ATTR_ASRL_FLOW_CNTRL.

But also, the presence of empty py_name creates an empty entry in all resources:
```
>>> getattr(pyvisa.resources.serial.SerialInstrument, '')
<pyvisa.attributes.AttrVI_ATTR_MAX_QUEUE_LENGTH object at 0x03123990>
```
This is a bug in Resource.register of pyvisa/resources/resource.py
I can fix that, but what should be done with missing py_name?
-Drop them (renders the attribute entry useless, use will instead use set/get_visa_attribute)
-use the visa_name or the __name__.
-find a way to put all attributes in an 'attributes' object inside of the resource and make them work.

Finally, the visa_attributes_classes variable of a resource contains a list of all the classes. What is the intended use for that list? I does not seem very useful to me.

hgrecco, 2015-04-29:
The standard way (low level and old way) to get an attribute value from a resource is to use `get_visa_attribute` which returns a value close to what the library provides. For example, given a defined `resource_name`

```python
import visa
rm = visa.ResourceManager()
inst = rm.open_resource(resource_name)
inst.get_visa_attribute(visa.constants.VI_ATTR_4882_COMPLIANT)
```
returns `1` (*VI_TRUE*) or `0` (*VI_FALSE*). 

The new (and more pythonic way) is to use the `py_name`:
```python
print(inst.is_4882_compliant)
```
which not only is nicer to call but it also returns `True` or `False`.

(By the way, you should use this in an instance not in a class.)

Now going to your questions/issues:

* The real bug is in the `register` method in `resource.py` which should not `setattr` those attributes with empty `py_name`.
* The plan is to provide a `py_name` for the most common attributes. `VI_ATTR_ASRL_FLOW_CNTRL` is definitely one of those.
* The plan is NOT to provide a `py_name` for all attributes in one go. The plan is to grow these attributes as we get more experience with the real needs.
* `visa_attributes_classes` is used for introspection. Is what enables, for example, the [shell](http://pyvisa.readthedocs.org/en/latest/shell.html)

lupien, 2015-04-30:
Thanks for the details.
I will make a push request for the flow_control and the bug only.

hgrecco, 2015-05-06:
Thanks for all work. Merged

closed 2015-05-06 by hgrecco