pyvisa/pyvisa #149: VISA installation problem

enzo96, 2015-05-07:
Hi, 

I'm working on a project in which i have to automate an atenna rotator in order to make radiance diagrams measurments. 
My system has to rotate my rotator and when the rotator reaches a wanted position, the rotator stops and during this little
break, my Vector Network Analyser Rohde&Schwarze zvl6 has to save in a file markers datas. My rotator is driven by a Raspberry Pi 2
in which i installed Raspbian. I linked the Raspberry Pi 2 and the VNA with an Ethernet cable.

I installed the pyVisa library on my Raspberry Pi 2(Raspbian).
I installed the NI VISA 5.4.1.iso
I use Python 2.7.3 32 bits 

But when I do the test procedure : 

>>> import visa
>>> rm = visa.ResourceManager()
>>> print(rm.list_resources())

I can import visa but when I do the second line, I have this :

Traceback (most recent call last) :
   File "<pyshell#1>", line 1 in <module>
	rm = visa.ResourceManager()
   File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1477,
 in __new__
	visa_library = open_visa_library(visa_library)
   File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1452,
 in open_visa_library
	raise e 
OSError : Could not open VISA library:

What do I have to do ?

hgrecco, 2015-05-10:
Please post the output (between triple quotes) of the following command:
```python
python -m visa info
```

enzo96, 2015-05-11:
python -m visa info :

Machine Details:
   Platform ID:  Linux-3.18.7-v7+-armv7l-with-debian-7.8
   Processor:

Python:
   Implementation: CPython 
   Executable:   /usr/bin/python
   Version:   2.7.3
   Compiler:  GCC 4.6.3
   Bits:    Mar 18 201405:13:23 (#default)
   Unicode:   UCS4

PyVISA Version 1.6.3

Backends:
   ni:
	Version: 1.6.3 (bundled with PyVISA)
	Binary Library: Not found

Even so, I got my NI-VISA-5.4.1.iso in my Downloads folder.

hgrecco, 2015-05-11:
as far as I know, NI-VISA is not compatible with linux on ARM. You might want to look at pyvisa-py, a pure python backend for the ni backend

enzo96, 2015-05-12:
Thanks for your quickness.

I installed pyvisa-py using pip install pyvisa-py. But I still have an error after : 
>>> import visa 
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()

Traceback (most recent call last):
  File "<pyshell#2>", line 1, in <module>
     rm.list_resources()
  File "/usr/local/lib/python2.7/dist_packages/pyvisa/highlevel.py", line 1554,
in list_resources
       find_list, return_counter, instrument_description, err = lib.find_resources(
self.session, query)
   File "/usr/local/lib/python2.7/dist_packages/pyvisa-py/highlevel.py", line 236
, in find_resources
      for key, st in sessions.Session.iter_valid_session_classes () ], [ ] )
   File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/serial.py", line 54, in
list_resources
      return ['ASRL%s: :INSTR' %port [0] for port in comports () ]
   File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/serial.py", line 29, in 
comports
      raise ValueError('Please install PySerial to use this ressource type')
ValueError : Please install PySerial to use this resource type

But I don't need PySerial because i don't want to use the serial port. I want to communicate via TCPIP. Furthermore PySerial is already installed : 

sudo pip install pyserial returns : 

Requirement already satisfied (use --upgrade to update): pyserial in usr/lib/python2.7/dist-packages
Cleaning up ...

If it can help, here's what python -m visa info returns : 

Machine Details:
   Platform ID : Linux-3.18.7-v7+-armv7l-with-debian-7.8
   Processor : 

Python : 
     Implementation : CPython
     Executable : /usr/bin/python
     Version : 2.7.3
     Compiler : GCC 4.6.3
     Bits : 32bit
     Build : Mar 18 2014 05:13:23 (#default)
     Unicode : UCS4

PyVISA Version : 1.6.3

Backend : 
    ni:
       Version : 1.6.3 (bundled with PyVISA)
       Binary library : Not found
   py : 
        Version : 0.1
        ASRL INSTR : Available via PySerial (N/A)
        TCPIP INSTR : Available
        USB INSTR : 
             Please install PyUSB to use this resource type.
             No module named usb
  
What do I have to do ?

hgrecco, 2015-05-12:
(Please always paste the output between triple backticks to facilitate reading, see code blocks here: https://guides.github.com/features/mastering-markdown/ ).

Do you know if pyserial works on arm? Can you try this?

```
python -m serial.tools.list_ports
```

enzo96, 2015-05-12:
~~ python -m serial.tools.list_ports~~

~~ /usr/bin/python: No module named tools ~~

enzo96, 2015-05-12:
``` python -m serial.tools.list_ports

/usr/bin/python: No module named tools ```

hgrecco, 2015-05-12:
This is taken directly from the pyserial documentation (see http://pyserial.sourceforge.net/pyserial_api.html#tools )
If this does not work, then maybe your pyserial installation has problems or you have an older version.

enzo96, 2015-05-12:
"""/usr/bin/python: No module named tools"""

What does that mean ?

enzo96, 2015-05-12:
I've just upgraded PySerial and now I have a different error when I use python -m serial.tools.list_ports.
It returns : """ No ports found """.
What can I do now ? 

PS : Why does PySerial matter because I don't use any ASRL communication in my project ? I want to send Ethernet control lines with my Raspi 2 on my Vector Network Analyzer.

hgrecco, 2015-05-12:
You are right. It should not be necessary. I will fix that in pyvisa-py. But now you should be able to list resources, right?

enzo96, 2015-05-12:
Yes, Thank you. It's better. I no longer have the error message. But when I do 
>>> rm.list_resources()

I have : ''' ( ) '''

What does that mean ?

hgrecco, 2015-05-12:
It means that nothing detectable is connected. Now try open the resource using the IP address of the device. for example
`TCPIP0::192.168.0.10::inst0::INSTR``

FusionLi, 2015-05-12:
Hi everyone,

I have a different error appeared on installing pyvisa-py pn on raspberry pi2, which is "ImportError: cannot import name rname" after I run command "rm = visa.ResourceManager('@py')". Here is the message given by interrupter:
"Traceback (most recent call last):
 File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1477, in __new__
    visa_library = open_visa_library(visa_library)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1446, in open_visa_library
    cls = get_wrapper_class(wrapper)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1426, in get_wrapper_class
    pkg = __import__(pkgname)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/__init__.py", line 26, in <module>
    from .highlevel import PyVisaLibrary
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 20, in <module>
    from pyvisa import constants, errors, highlevel, rname
ImportError: cannot import name rname
"
Could anyone help me solve it? Thanks a lot

hgrecco, 2015-05-12:
> File "", line 1, in

you have a dev versión of pyvisa-py but a stable pyvisa. please install
pyvisa from github

FusionLi, 2015-05-12:
Hi hgrecco,

I tried the github pyvisa-py but it still does not work, and I got same error and message from 'python -m visa info'

hgrecco, 2015-05-12:
you need to update pyvisa from github

FusionLi, 2015-05-12:
Thanks for quick reply!
What do you mean by 'update pyvisa from github', does it mean the command: "sudo pip install --update git+https://github.com/hgrecco/pyvisa-py.git". If that is the case, then it also does not solve the problem. I am new to the linux. Thanks for your patience.

enzo96, 2015-05-13:
Hi hgrecco, I tried to open the resource using the IP adress of the device but I got an error. 
Here it is : 

>>> import socket
>>> import visa 
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
>>> inst = rm.open_resource('TCPIP0: :192.168.0.139: :inst0: :INSTR')

``` Traceback (most recent call last) : 
       File "<pyshell#41>", line 1, in<module>
          rm.open_resource('TCPIP0::192.168.0.139::inst0::INSTR')
       File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1638, in open_resource
          res.open(access mode, open_timeout)
       File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/resource.py", line 165, in open
          self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
       File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line1602, in open_bare_resource
       return.self.visalib.open(self.session, resource_name, access_mode, open_timeout)
      File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 180, in open
           sess = cls(session, resource_name, parsed)
      File"/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 183, in __init__
            self.after_parsing()
      File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/tcpip.py", line 45, in after_parsing
            self.interface = vxill.Coreclient (self.parsed['host_adress'])
      File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/vxill.py", line 199, in __init__
           super (Coreclient, self).__init__(host, DEVICE_CORE_PROG, DEVICE_CORE_VERS)
TypeError : Struct () argument 1 must be string, not unicode ``` 

Where does it come from ?

hgrecco, 2015-05-13:
Before I look into it, is it possible that your connection string is wrong? There seems to be spaces between the colons. It should be `TCPIP0::192.168.0.139::inst0::INSTR` And please wrap you pasted text in triple backquotes to maintain the format

enzo96, 2015-05-18:
No, I added spaces in my comment but not in my python instructions.

hgrecco, 2015-05-19:
I am moving this to pyvisa-py

closed 2015-05-19 by hgrecco