pyvisa/pyvisa #152: Cannot see my instrument using rm.list_resources()

nktsaini, 2015-05-08:
I am trying to use PyVISA on the beagleboard black which runs a ARM version of debian 7.8.

python -m visa info fetched this:

Machine Details:
   Platform ID:    Linux-3.8.13-bone70-armv7l-with-debian-7.8
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.3
   Compiler:       GCC 4.6.3
   Bits:           32bit
   Build:          Mar 14 2014 17:55:54 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.4.dev1

Backends:
   ni:
      Version: 1.6.4.dev1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2.dev0
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named tools.list_ports
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP INSTR: Available 
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.0rc1). Backend: libusb1

When I run the following code:
import visa
rm = visa.ResourceManager('@py')
rm.list_resources()                         #2 

#2 gives me an error: VisaIOError: Insufficient location information or the requested device or resource is not present in the system.

Now, I have a Keithley 2701 DMM plugged in via ethernet.

Help is welcome and appreciated

PS. I am relatively new to PyVISA, using it for the first time.

dtnorton, 2015-05-08:
Try explicitly entering the IP address of your instrument:

    import visa
    DMM = visa.ResourceManager().get_instrument('TCPIP0::XXX.XXX.XXX.XXX::inst0::INSTR')
    DMM.ask('*IDN?')

hgrecco, 2015-05-10:
I agree with @dtnorton try to explicitly. But the suggested syntax is:
```python
import visa
rm = visa.ResourceManager('@py')
dmm = rm.open_resource('TCPIP0::XXX.XXX.XXX.XXX::inst0::INSTR')
dmm.query('*IDN?')
```

nktsaini, 2015-05-11:
I ran these commands. I get the following error:

>AttributeError: 'module' object has no attribute 'parse_resource_name'

PS. Interesting thing is that I can ping to the instrument using the os module in python like this:
>import os
>report=os.system("ping -c 4 169.254.198.115")

I get returned all the packets with 0% loss.

What am I doing wrong here?

hgrecco, 2015-05-11:
pyvisa-py is experimental. Please update to the latest pyvisa and pyvisa-py and try again.

nktsaini, 2015-05-11:
I have the lastest version I believe. PyVisa 1.6.4.dev1 and PyVisa-py ver 0.2.dev0

hgrecco, 2015-05-11:
can you give me more context about the error. (line number, etc)

hgrecco, 2015-05-11:
By the way, I pushed an update a few hours ago

nktsaini, 2015-05-11:
Context: I am trying to interface a Keithley 2701 DMM with a Beaglebone Black (BBB). It has a Debian ARM image (Debian wheezy 7.8). Since the visa library is not compatible with debian, I am trying to use the pyvisa-py as an alternative and need pyvisa as a wrapper to acquire data using codes written in python. I can ping the Keithley DMM and get replies as well. I have updated as you had asked.
 
>python -m visa info

gives me this:

Machine Details:
   Platform ID:    Linux-3.8.13-bone70-armv7l-with-debian-7.8
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.3
   Compiler:       GCC 4.6.3
   Bits:           32bit
   Build:          Mar 14 2014 17:55:54 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.4.dev1

Backends:
   ni:
      Version: 1.6.4.dev1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2.dev0
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named tools.list_ports
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP INSTR: Available 
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.0rc1). Backend: libusb1

hgrecco, 2015-05-11:
yes, but can you provide the full stack trace where you got `AttributeError: 'module' object has no attribute 'parse_resource_name'`

nktsaini, 2015-05-11:
Here it is: 

>dmm = rm.open_resource('TCPIP0::169.254.198.115::inst0::INSTR')
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-881fd17430d6> in <module>()
----> 1 dmm = rm.open_resource('TCPIP0::169.254.198.115::inst0::INSTR')

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.pyc in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1626         :rtype: :class:`pyvisa.resources.Resource`
   1627         """
-> 1628         info = self.resource_info(resource_name)
   1629 
   1630         try:

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.pyc in resource_info(self, resource_name)
   1593         :rtype: :class:`pyvisa.highlevel.ResourceInfo`
   1594         """
-> 1595         ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
   1596         if err == constants.StatusCode.success:
   1597             return ret

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/highlevel.pyc in parse_resource_extended(self, session, resource_name)
    273         """
    274         try:
--> 275             parsed = common.parse_resource_name(resource_name)
    276 
    277             return (highlevel.ResourceInfo(parsed.interface_type_const,

AttributeError: 'module' object has no attribute 'parse_resource_name'

hgrecco, 2015-05-11:
This was fixed in the latest version of pyvisa-py please update.

hgrecco, 2015-05-11:
Sorry. There was another. This is due to a change in PyVISA. Please update now

nktsaini, 2015-05-11:
Okay. I'll try updating and get back to you.

hgrecco, 2015-05-11:
Ps.- By the way your usage of pyvisa-py is exactly one of the reasons why pyvisa-py was created. I am happy to see that!

nktsaini, 2015-05-11:
PyVisa-py is a God-send for me, if only I can get it to work. [:

Anyway, I performed the update and the positive news is that the error I am getting is different now.

>import visa
>rm = visa.ResourceManager('@py')
>rm.open_resource('TCPIP::169.254.198.115::1394::INSTR')

error                                     Traceback (most recent call last)
<ipython-input-16-aad4e6946656> in <module>()
----> 1 rm.open_resource("TCPIP:169.254.198.115::1394::INSTR")

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.pyc in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1646                 raise ValueError('%r is not a valid attribute for type %s' % (key, res.__class__.__name__))
   1647 
-> 1648         res.open(access_mode, open_timeout)
   1649 
   1650         for key, value in kwargs.items():

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/resources/resource.pyc in open(self, access_mode, open_timeout)
    182         logger.debug('%s - opening ...', self._resource_name, extra=self._logging_extra)
    183         with self._resource_manager.ignore_warning(constants.VI_SUCCESS_DEV_NPRESENT):
--> 184             self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
    185 
    186             if status == constants.VI_SUCCESS_DEV_NPRESENT:

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.pyc in open_bare_resource(self, resource_name, access_mode, open_timeout)
   1610         :return: Unique logical identifier reference to a session.
   1611         """
-> 1612         return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
   1613 
   1614     def open_resource(self, resource_name,

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/highlevel.pyc in open(self, session, resource_name, access_mode, open_timeout)
    196         cls = sessions.Session.get_session_class(parsed.interface_type_const, parsed.resource_class)
    197 
--> 198         sess = cls(session, resource_name, parsed)
    199 
    200         return self._register(sess), constants.StatusCode.success

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/sessions.pyc in __init__(self, resource_manager_session, resource_name, parsed)
    182                       constants.VI_ATTR_RSRC_CLASS: parsed.resource_class,
    183                       constants.VI_ATTR_INTF_TYPE: parsed.interface_type}
--> 184         self.after_parsing()
    185 
    186     def after_parsing(self):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/tcpip.pyc in after_parsing(self)
     43         # TODO: board_number not handled
     44         # TODO: lan_device_name not handled
---> 45         self.interface = vxi11.CoreClient(self.parsed.host_address)
     46 
     47         self.lock_timeout = 10000

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/vxi11.pyc in __init__(self, host)
    195         self.packer = Vxi11Packer()
    196         self.unpacker = Vxi11Unpacker('')
--> 197         super(CoreClient, self).__init__(host, DEVICE_CORE_PROG, DEVICE_CORE_VERS)
    198 
    199     def create_link(self, id, lock_device, lock_timeout, name):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host, prog, vers)
    580     """
    581     def __init__(self, host, prog, vers):
--> 582         pmap = TCPPortMapperClient(host)
    583         port = pmap.get_port((prog, vers, IPPROTO_TCP, 0))
    584         pmap.close()

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host)
    558 
    559     def __init__(self, host):
--> 560         RawTCPClient.__init__(self, host, PMAP_PROG, PMAP_VERS, PMAP_PORT)
    561         PartialPortMapperClient.__init__(self)
    562 

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host, prog, vers, port)
    323     def __init__(self, host, prog, vers, port):
    324         Client.__init__(self, host, prog, vers, port)
--> 325         self.connect()
    326 
    327     def connect(self):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/rpc.pyc in connect(self)
    327     def connect(self):
    328         self.sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
--> 329         self.sock.connect((self.host, self.port))
    330 
    331     def close(self):

/usr/lib/python2.7/socket.pyc in meth(name, self, *args)
    222 
    223 def meth(name,self,*args):
--> 224     return getattr(self._sock,name)(*args)
    225 
    226 for _m in _socketmethods:

error: [Errno 101] Network is unreachable

hgrecco, 2015-05-11:
Are you sure that is the right connection string? What is 1394? Is that the port? For an INSTR device the port is not used. Only for a SOCKET. See http://digital.ni.com/public.nsf/allkb/80C28CC795611E4A86256AB100209521

nktsaini, 2015-05-11:
I stand corrected. 1394 is a port but it is not needed with INSTR. So I tried this now:

>rm.open_resource('TCPIP::169.254.198.115::INSTR')

I get the following error:
ValueError                                Traceback (most recent call last)
<ipython-input-32-fdc103a44b27> in <module>()
----> 1 rm.open_resource('TCPIP:169.254.198.115::INSTR')

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.pyc in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1626         :rtype: :class:`pyvisa.resources.Resource`
   1627         """
-> 1628         info = self.resource_info(resource_name)
   1629 
   1630         try:

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.pyc in resource_info(self, resource_name)
   1596         if err == constants.StatusCode.success:
   1597             return ret
-> 1598         raise ValueError('Could not parse resource: %s (error code %s)' % (resource_name, ret))
   1599 
   1600     def open_bare_resource(self, resource_name,

ValueError: Could not parse resource: TCPIP:169.254.198.115::INSTR (error code 0)

hgrecco, 2015-05-11:
This is a bug in pyvisa which is not using the default value for lan device name. You can bypass it using:
`'TCPIP:169.254.198.115::inst0::INSTR'`

nktsaini, 2015-05-11:
So I ran this command

>dmm=rm.open_resource('TCPIP::169.254.198.115::inst0::INSTR')

gaierror                                  Traceback (most recent call last)
<ipython-input-33-dc98a9a93731> in <module>()
----> 1 rm.open_resource('TCPIP:169.254.198.115::inst0::INSTR')

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.pyc in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1646                 raise ValueError('%r is not a valid attribute for type %s' % (key, res.__class__.__name__))
   1647 
-> 1648         res.open(access_mode, open_timeout)
   1649 
   1650         for key, value in kwargs.items():

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/resources/resource.pyc in open(self, access_mode, open_timeout)
    182         logger.debug('%s - opening ...', self._resource_name, extra=self._logging_extra)
    183         with self._resource_manager.ignore_warning(constants.VI_SUCCESS_DEV_NPRESENT):
--> 184             self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
    185 
    186             if status == constants.VI_SUCCESS_DEV_NPRESENT:

/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.pyc in open_bare_resource(self, resource_name, access_mode, open_timeout)
   1610         :return: Unique logical identifier reference to a session.
   1611         """
-> 1612         return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
   1613 
   1614     def open_resource(self, resource_name,

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/highlevel.pyc in open(self, session, resource_name, access_mode, open_timeout)
    196         cls = sessions.Session.get_session_class(parsed.interface_type_const, parsed.resource_class)
    197 
--> 198         sess = cls(session, resource_name, parsed)
    199 
    200         return self._register(sess), constants.StatusCode.success

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/sessions.pyc in __init__(self, resource_manager_session, resource_name, parsed)
    182                       constants.VI_ATTR_RSRC_CLASS: parsed.resource_class,
    183                       constants.VI_ATTR_INTF_TYPE: parsed.interface_type}
--> 184         self.after_parsing()
    185 
    186     def after_parsing(self):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/tcpip.pyc in after_parsing(self)
     43         # TODO: board_number not handled
     44         # TODO: lan_device_name not handled
---> 45         self.interface = vxi11.CoreClient(self.parsed.host_address)
     46 
     47         self.lock_timeout = 10000

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/vxi11.pyc in __init__(self, host)
    195         self.packer = Vxi11Packer()
    196         self.unpacker = Vxi11Unpacker('')
--> 197         super(CoreClient, self).__init__(host, DEVICE_CORE_PROG, DEVICE_CORE_VERS)
    198 
    199     def create_link(self, id, lock_device, lock_timeout, name):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host, prog, vers)
    580     """
    581     def __init__(self, host, prog, vers):
--> 582         pmap = TCPPortMapperClient(host)
    583         port = pmap.get_port((prog, vers, IPPROTO_TCP, 0))
    584         pmap.close()

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host)
    558 
    559     def __init__(self, host):
--> 560         RawTCPClient.__init__(self, host, PMAP_PROG, PMAP_VERS, PMAP_PORT)
    561         PartialPortMapperClient.__init__(self)
    562 

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host, prog, vers, port)
    323     def __init__(self, host, prog, vers, port):
    324         Client.__init__(self, host, prog, vers, port)
--> 325         self.connect()
    326 
    327     def connect(self):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/protocols/rpc.pyc in connect(self)
    327     def connect(self):
    328         self.sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
--> 329         self.sock.connect((self.host, self.port))
    330 
    331     def close(self):

/usr/lib/python2.7/socket.pyc in meth(name, self, *args)
    222 
    223 def meth(name,self,*args):
--> 224     return getattr(self._sock,name)(*args)
    225 
    226 for _m in _socketmethods:

gaierror: [Errno -5] No address associated with hostname

hgrecco, 2015-05-11:
Do you have by chance a computer in which NI-VISA works? It would be nice to test this using the default library just to be sure that everything is properly configured. Once we know that, we could then debug PyVISA-py.

nktsaini, 2015-05-11:
That'll take me some time but okay

I'll download NI VISA on a windows machine along with the pyvisa and try it there, I'll get back to you when I am done.

Thanks a ton for all your help.

hgrecco, 2015-05-11:
You are welcome! pyvisa-py is still very young but as I said, your use case is the exact reason why pyvisa-py was created. So thanks for help us debugging and if it is a problem in our side, we should fix it.

nktsaini, 2015-05-11:
I installed NI VISA on a windows machine along with the latest pyvisa. I ran the following command:

>python -m visa info

This yielded:
```
Machine Details:
   Platform ID:    Windows-8-6.2.9200
   Processor:      Intel64 Family 6 Model 69 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Miniconda3\python.exe
   Version:        3.4.2
   Compiler:       MSC v.1600 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan  9 2015 10:32:40 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.4.dev1

Backends:
   ni:
      Version: 1.6.4.dev1 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243648
         Spec. Version: 5243136
      #2: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243648
         Spec. Version: 5243136
   py:
      Version: 0.2.dev0
      TCPIP INSTR: Available
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named 'serial'
```

Then I proceed to ipython where I wrote the following commands:
>import visa
>rm=visa.ResourceManager()
>dmm=rm.open_resource("TCPIP::169.254.198.115::inst0::INSTR")

This yielded the following error:
```
VisaIOError                               Traceback (most recent call last)
<ipython-input-3-577f2eeff240> in <module>()
----> 1 dmm = rm.open_resource("TCPIP::169.254.198.115::inst0:INSTR")

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\highlevel.py
in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1626         :rtype: :class:`pyvisa.resources.Resource`
   1627         """
-> 1628         info = self.resource_info(resource_name)
   1629
   1630         try:

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\highlevel.py
in resource_info(self, resource_name)
   1593         :rtype: :class:`pyvisa.highlevel.ResourceInfo`
   1594         """
-> 1595         ret, err = self.visalib.parse_resource_extended(self.session, re
source_name)
   1596         if err == constants.StatusCode.success:
   1597             return ret

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\ctwrapper\fun
ctions.py in parse_resource_extended(library, session, resource_name)
   1376                                 byref(interface_board_number), resource_
class,
   1377                                 unaliased_expanded_resource_name,
-> 1378                                 alias_if_exists)
   1379
   1380     if ret != constants.VI_SUCCESS:

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\ctwrapper\hig
hlevel.py in _return_handler(self, ret_value, func, arguments)
    186
    187         if ret_value < 0:
--> 188             raise errors.VisaIOError(ret_value)
    189
    190         if ret_value in self.issue_warning_on:

VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location informati
on or the requested device or resource is not present in the system.
```

hgrecco, 2015-05-11:
I think that you are missing a colon before INSTR. It should be double colon `::` like `"TCPIP::169.254.198.115::inst0::INSTR"`

nktsaini, 2015-05-12:
__Sorry about that. I ran the command again and the reply is still the same:__

```
VisaIOError                               Traceback (most recent call last)
<ipython-input-3-8546aff2eb01> in <module>()
----> 1 dmm = rm.open_resource("TCPIP::169.254.198.115::inst0::INSTR")

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\highlevel.py
in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1646                 raise ValueError('%r is not a valid attribute for type %
s' % (key, res.__class__.__name__))
   1647
-> 1648         res.open(access_mode, open_timeout)
   1649
   1650         for key, value in kwargs.items():

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\resources\res
ource.py in open(self, access_mode, open_timeout)
    182         logger.debug('%s - opening ...', self._resource_name, extra=self
._logging_extra)
    183         with self._resource_manager.ignore_warning(constants.VI_SUCCESS_
DEV_NPRESENT):
--> 184             self.session, status = self._resource_manager.open_bare_reso
urce(self._resource_name, access_mode, open_timeout)
    185
    186             if status == constants.VI_SUCCESS_DEV_NPRESENT:

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\highlevel.py
in open_bare_resource(self, resource_name, access_mode, open_timeout)
   1610         :return: Unique logical identifier reference to a session.
   1611         """
-> 1612         return self.visalib.open(self.session, resource_name, access_mod
e, open_timeout)
   1613
   1614     def open_resource(self, resource_name,

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\ctwrapper\fun
ctions.py in open(library, session, resource_name, access_mode, open_timeout)
   1206     # [ViSession, ViRsrc, ViAccessMode, ViUInt32, ViPSession]
   1207     # ViRsrc converts from (str, unicode, bytes) to bytes
-> 1208     ret = library.viOpen(session, resource_name, access_mode, open_timeo
ut, byref(out_session))
   1209     return out_session.value, ret
   1210

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\ctwrapper\hig
hlevel.py in _return_handler(self, ret_value, func, arguments)
    186
    187         if ret_value < 0:
--> 188             raise errors.VisaIOError(ret_value)
    189
    190         if ret_value in self.issue_warning_on:

VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location informati
on or the requested device or resource is not present in the system.
```

__Interestingly, when I used the py library:__
>rm = visa.ResourceManager("@py")
>dmm = rm.open_resource("TCPIP::169.254.198.115::inst0::INSTR")

__This happened:__
```
VisaIOError                               Traceback (most recent call last)
<ipython-input-3-8546aff2eb01> in <module>()
----> 1 dmm = rm.open_resource("TCPIP::169.254.198.115::inst0::INSTR")

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\highlevel.py
in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1646                 raise ValueError('%r is not a valid attribute for type %
s' % (key, res.__class__.__name__))
   1647
-> 1648         res.open(access_mode, open_timeout)
   1649
   1650         for key, value in kwargs.items():

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\resources\res
ource.py in open(self, access_mode, open_timeout)
    182         logger.debug('%s - opening ...', self._resource_name, extra=self
._logging_extra)
    183         with self._resource_manager.ignore_warning(constants.VI_SUCCESS_
DEV_NPRESENT):
--> 184             self.session, status = self._resource_manager.open_bare_reso
urce(self._resource_name, access_mode, open_timeout)
    185
    186             if status == constants.VI_SUCCESS_DEV_NPRESENT:

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\highlevel.py
in open_bare_resource(self, resource_name, access_mode, open_timeout)
   1610         :return: Unique logical identifier reference to a session.
   1611         """
-> 1612         return self.visalib.open(self.session, resource_name, access_mod
e, open_timeout)
   1613
   1614     def open_resource(self, resource_name,

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\ctwrapper\fun
ctions.py in open(library, session, resource_name, access_mode, open_timeout)
   1206     # [ViSession, ViRsrc, ViAccessMode, ViUInt32, ViPSession]
   1207     # ViRsrc converts from (str, unicode, bytes) to bytes
-> 1208     ret = library.viOpen(session, resource_name, access_mode, open_timeo
ut, byref(out_session))
   1209     return out_session.value, ret
   1210

C:\Miniconda3\lib\site-packages\pyvisa-1.6.4.dev1-py3.4.egg\pyvisa\ctwrapper\hig
hlevel.py in _return_handler(self, ret_value, func, arguments)
    186
    187         if ret_value < 0:
--> 188             raise errors.VisaIOError(ret_value)
    189
    190         if ret_value in self.issue_warning_on:

VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location informati
on or the requested device or resource is not present in the system.
```

dtnorton, 2015-05-12:
For me, at least, I need a zero at the end of the TCPIP in the resource name. Try TCPIP0.
<div class="highlight highlight-python"><pre><span class="pl-k">import</span> visa
rm <span class="pl-k">=</span> visa.ResourceManager(<span class="pl-s"><span class="pl-pds">'</span>@py<span class="pl-pds">'</span></span>)
dmm <span class="pl-k">=</span> rm.open_resource(<span class="pl-s"><span class="pl-pds">'</span>TCPIP0::169.254.198.115::inst0::INSTR<span class="pl-pds">'</span></span>)
dmm.query(<span class="pl-s"><span class="pl-pds">'</span>*IDN?<span class="pl-pds">'</span></span>)</pre></div>

hgrecco, 2015-05-12:
If I remember correctly `0` is the default value for board. I will check it and check that is implemented in pyvisa. But in any case, it would be nice to see that both backends behave the same: i.e. they succeed and fail in the same way.

dtnorton, 2015-05-12:
I think you are correct.  I tested omitting the <code>0</code> at the end of <code>TCPIP</code> and it worked just fine.  Must have been my bad memory...

hgrecco, 2015-05-19:
I am closing this. Feel free to reopen if needed.

closed 2015-05-19 by hgrecco

nktsaini, 2015-05-21:
Can we reopen this? I'm still trying to get PyVISA-py to work as a backend to the PyVISA?

hgrecco, 2015-05-21:
If this is pypvisa-py problem, please open an issue in the pyvisa-py tracker https://github.com/hgrecco/pyvisa-py/issues

Please update pyvisa to the latest version (in PyPI) and pyvisa-py

nktsaini, 2015-05-31:
__I've updated pyvisa and pyvisa-py to the latest updates on my windows machines and am still unable to connect to my device over the ethernet.__

>python -m visa info

__yields:__
```
C:\windows\system32>python -m visa info
Machine Details:
   Platform ID:    Windows-8-6.2.9200
   Processor:      Intel64 Family 6 Model 69 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda\python.exe
   Version:        2.7.9
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 18 2014 16:57:52 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8.dev0

Backends:
   ni:
      Version: 1.8.dev0 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
      #2: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
   py:
      Version: 0.2.dev2
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named serial
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP INSTR: Available
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
```

__When I try to connect to the instrument by:__
>import visa
>rm = visa.ResourceManager()
>dmm = rm.open_resource('TCPIP::10.66.22.150::inst0:INSTR')

__I get:__
```
In [8]: dmm = rm.open_resource('TCPIP::10.66.22.150::inst0::INSTR')
---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
<ipython-input-8-6eb781494e4f> in <module>()
----> 1 dmm = rm.open_resource('TCPIP::10.66.22.150::inst0::INSTR')

C:\Anaconda\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\highlevel.pyc in
open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1632                 raise ValueError('%r is not a valid attribute for type %
s' % (key, res.__class__.__name__))
   1633
-> 1634         res.open(access_mode, open_timeout)
   1635
   1636         for key, value in kwargs.items():

C:\Anaconda\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\resources\resourc
e.pyc in open(self, access_mode, open_timeout)
    182         logger.debug('%s - opening ...', self._resource_name, extra=self
._logging_extra)
    183         with self._resource_manager.ignore_warning(constants.VI_SUCCESS_
DEV_NPRESENT):
--> 184             self.session, status = self._resource_manager.open_bare_reso
urce(self._resource_name, access_mode, open_timeout)
    185
    186             if status == constants.VI_SUCCESS_DEV_NPRESENT:

C:\Anaconda\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\highlevel.pyc in
open_bare_resource(self, resource_name, access_mode, open_timeout)
   1596         :return: Unique logical identifier reference to a session.
   1597         """
-> 1598         return self.visalib.open(self.session, resource_name, access_mod
e, open_timeout)
   1599
   1600     def open_resource(self, resource_name,

C:\Anaconda\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\ctwrapper\functio
ns.pyc in open(library, session, resource_name, access_mode, open_timeout)
   1209     # [ViSession, ViRsrc, ViAccessMode, ViUInt32, ViPSession]
   1210     # ViRsrc converts from (str, unicode, bytes) to bytes
-> 1211     ret = library.viOpen(session, resource_name, access_mode, open_timeo
ut, byref(out_session))
   1212     return out_session.value, ret
   1213

C:\Anaconda\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\ctwrapper\highlev
el.pyc in _return_handler(self, ret_value, func, arguments)
    186
    187         if ret_value < 0:
--> 188             raise errors.VisaIOError(ret_value)
    189
    190         if ret_value in self.issue_warning_on:

VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location informati
on or the requested device or resource is not present in the system.
```

__This seems to me like a pyvisa issue. I went through other issues for pyvisa and seems like only one other person had problem connecting to instruments over the ethernet. Interestingly, someone before me about a year ago used PyVISA 1.4 or lower with NI VISA and could connect to the instrument.__

hgrecco, 2015-05-31:
A few questionst:
- Can you see the device using NI-MAX? 
- Does it have the same the resource name that you are using?
- Can you connect to it?

nktsaini, 2015-06-01:
While trying to communicate with the device (a Keithley DMM 2701) using MAX, I came across this article on the NI website:

>Once you are able to browse it in IE, you can configure it in LabVIEW. There is no need to configure it in MAX (Measurement and Automation Explorer) since this is not a VXI 11 compliant instrument. You would have to configure it as a raw socket instrument. The problem is that it uses port 1394 for the gpib communication and MAX will not allow opening a visa session with that port.

So, I tried running the following script on my windows machine which has VISA on it:
```
import visa
rm = visa.ResourceManager()
dmm = rm.open_resource('TCPIP::10.66.22.150::1394::SOCKET')
```

and it connected!

I am sorry for missing this before but I think my instrument can only be communicated over a raw socket connection. Now how do I make this work with the pyvisa-py backend since it doesn't support socket.

nktsaini, 2015-06-01:
So I tried connecting to the Keithley over a raw socket connection using the pyvisa-py backend by running the following script:

```
import visa
import socket
rm = visa.ResourceManager('@py')
sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sock.connect((10.66.22.150, 1394))
```

but it returns:

```
File "<ipython-input-19-b7266b5e0716>", line 1
    sock.connect((10.66.22.150, 1394))
                         ^
SyntaxError: invalid syntax
```

MatthieuDartiailh, 2015-06-01:
You need to pass the address as a string.

nktsaini, 2015-06-01:
Thanks. That worked. So I think I've established a connection with my instrument!

MatthieuDartiailh, 2015-06-01:
Great !

hgrecco, 2015-06-01:
We can make pyvisa-py support `SOCKET` but I do not have any instrument to test. So if you are fine with a bumpy road, then we try.

nktsaini, 2015-06-01:
I am prepared to contribute as well as I can. Currently, I am using the socket module to communicate with my instrument which is a pain because now I have to learn SCPI language.

So let me know how I can help.