pyvisa/pyvisa #153: Errors with visa command: instrument.read_values()

dtnorton, 2015-05-08:
I updated from pyvisa1.5 to pyvisa1.6.3 and my code for talking to a Keithley 2601B SMU no long works. The code previously worked fine when reading data from the buffer of the Keithley.  When the buffer was empty, the program kicked out of the while loop and was completed.  Now I have errors thrown.  This occurs with a visa read command "instrument.read_values()".  

If I issued the command as: instrument.read_values()
I get an error message:
>/anaconda/lib/python2.7/site-packages/pyvisa/resources/messagebased.pyc in read_values(self, fmt, container)
    345                 except IndexError:
    346                     msg = ''
--> 347                 raise errors.InvalidBinaryFormat(msg)
    348 
    349         if fmt & 0x01 == 0: # ascii<

>InvalidBinaryFormat: Unrecognized binary data format: Could not find valid hash position<

If I issue the command as: instrument.read_values(2)
I get an error message:
>/anaconda/lib/python2.7/site-packages/pyvisa/util.pyc in <genexpr>((raw_value,))
    193         data = separator(ascii_data)
    194 
--> 195     return container(converter(raw_value) for raw_value in data)
    196 
    197 <

>ValueError: could not convert string to float: <

Here is an example code:

	import visa
	IP = 'TCPIP0::169.254.000.001::inst0::INSTR'

	def DataBufferEx(IP=IP,startV=0.0,stepV=0.01,endV=1.0):

		smu = visa.ResourceManager().get_instrument(IP)
		current = []
		bias = []
		
		smu.clear()
		smu.write("loadandrunscript RunIV" )
		smu.write("reset()")
		smu.write("display.screen = display.SMUA")
		smu.write("display.smua.measure.func = display.MEASURE_DCAMPS")
		smu.write("smua.measure.autorangei = smua.AUTORANGE_ON")
		smu.write("format.data = format.ASCII")
		smu.write("smua.nvbuffer1.clear()")
		smu.write("smua.nvbuffer1.appendmode = 1")
		smu.write("smua.nvbuffer1.collectsourcevalues = 1")
		smu.write("smua.measure.count = 1")
		smu.write("smua.source.func = smua.OUTPUT_DCVOLTS")
		smu.write("smua.source.levelv = 0.0")
		smu.write("smua.source.output = smua.OUTPUT_ON")
		smu.write("local v")
		smu.write("for v = 1, 100 do ")
		smu.write("    smua.source.levelv = v * 0.01 ")
		smu.write("    smua.measure.i(smua.nvbuffer1) ")
		smu.write("    waitcomplete() ")
		smu.write("    printbuffer(smua.nvbuffer1.n, smua.nvbuffer1.n, smua.nvbuffer1.sourcevalues, smua.nvbuffer1)")
		smu.write("end ")
		smu.write("smua.source.levelv = 0.0")
		smu.write("smua.source.output = smua.OUTPUT_OFF")
		smu.write("endscript")
	
		# smu.write("printbuffer(1, smua.nvbuffer1.n, smua.nvbuffer1.readings)")
		# current = smu.read_values(2)
		# smu.write("printbuffer(1, smua.nvbuffer1.n, smua.nvbuffer1.sourcevalues)")
		# voltage = smu.read_values(2)
		# smu.write("printbuffer(1, smua.nvbuffer1.n, smua.nvbuffer1.sourcevalues, smua.nvbuffer1)")
		# voltage,current = smu.read_values(2)
	
		response = smu.read_values(2)
		ct = 0
		while len(response) != 0:
			print len(response), ct, response
			(v,i) = response
			ct += 1
			current.append(i)
			bias.append(v)
			response = smu.read_values(2)
		smu.clear()
		smu.close()
	
		return bias,current

hgrecco, 2015-05-10:
You will find information about the changes here: http://pyvisa.readthedocs.org/en/master/migrating.html 
A few things:
```
smu = visa.ResourceManager().get_instrument(IP)
```
should be:
```
rm = visa.ResourceManager()
smu = rm.open_resource(IP)
```

The `read_values` was broken so we replaced by better constructs. Tell me more about your data

dtnorton, 2015-05-10:
Thanks.  I made the changes you recommended.  The errors still occur with the <code>read_values</code>.  The data from the instrument, smu, is formatted as ASCII (see line 16 in code above).  If I do a <code>read_raw()</code> call, the while loop hangs up for the last data point, but ultimately kicks out of the while loop and returns the data.  Thus, working as it did before, only with a small hang up.  The hang up is minor, maybe a second.  The raw data from the smu is a string of comma separated values.  Is there more specific information you need to know?

hgrecco, 2015-05-11:
You need to specify the format as described here: http://pyvisa.readthedocs.org/en/master/rvalues.html#preconfiguring-the-transfer-format

In your case, something like this will work:
```python 
inst.values_format.use_ascii('f', ',')
```

The first argument is the datatype (in this case float), the  second is the separator (in this case comma). There is an optional third parameter which specifies the output container (default is `list`, but you can you anything like a `tuple` or a `numpy.array`)

It would be nice to track down the reason of the delay.

dtnorton, 2015-05-11:
I made the change you recommended, however this did not solve the issues.  I have tried a number of different read commands, each with their own issues.
<code>response = smu.read_values(2)</code>, <code>smu.query_values('CURV?')</code>, or <code>smu.query_ascii_values('CURV?')</code> all fail (return an error) at the last data point read or the smu reaching its compliance value due to an empty string not being converted to a float.

<code>response = smu.read_raw()</code> and <code>smu.read()</code> handle the while loop correctly when the last data point or the compliance is met, however, there is a strange pause when either of these occur.

dtnorton, 2015-05-12:
Changed the logic a bit to my code.  I now have the smu print a line <code>'\n'</code> when it either reaches compliance or it's last voltage value, and then use that as the conditional to my while loop.  The pause no longer occurs.  In short, the pause was due to my coding <b>NOT</b> due to pyvisa.  Thanks for looking into this a bit with me and thanks a ton for your hard work in maintaining and improving pyvisa.

hgrecco, 2015-05-19:
I am happy to hear. I am closing this

closed 2015-05-19 by hgrecco