pyvisa/pyvisa #157: VI_ERROR_TMO (-1073807339) Timeout

dillharr, 2015-05-15:
I am having a timeout issue that will occur at the same point in a python script every time. It is very confusing, because it is giving me this error even though the Satellite Simulator that I am sending SCPI commands to acts like it received the command. I tried extending the timeout to 5 seconds, but this didn't help. Any ideas of what could be causing this. Below is the error output PowerShell gives me.

Traceback (most recent call last):
  File ".\GNSS_FUNC_TEST.py", line 235, in <module>
    box.write('sour:scen:cont start')
  File "C:\Python34\lib\site-packages\pyvisa-1.6.3-py3.4.egg\pyvisa\resources\messagebased.py", line 184, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Python34\lib\site-packages\pyvisa-1.6.3-py3.4.egg\pyvisa\resources\messagebased.py", line 162, in write_raw
    return self.visalib.write(self.session, message)
  File "C:\Python34\lib\site-packages\pyvisa-1.6.3-py3.4.egg\pyvisa\ctwrapper\functions.py", line 1849, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\Python34\lib\site-packages\pyvisa-1.6.3-py3.4.egg\pyvisa\ctwrapper\highlevel.py", line 180, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
PS C:\python34\Lib\Framework\Resource\TestDevice>

MatthieuDartiailh, 2015-05-15:
Have you correctly set the read_termination character ? Visa looks for this character(s) at the end of the instrument answers. If it does not find it a timeout will occur as it will go on waiting for it.

dillharr, 2015-05-15:
Yes I have. I tried changing the read_termination from '\n' to '\r' but it didn't help. I still experience the timeout. The really strange thing is that it has the timeout on a command that works without any issues earlier in the python script that I am running.

hgrecco, 2015-05-15:
But it seems that you are getting a write timeout, right? Can you give more information about the device

dillharr, 2015-05-15:
Yes it appears I am getting a write timeout even though the device appears to take the command and starts the scenario I had it load. The device is a Spectracom GSG-62 Multi-GNSS Simulator. I am using pyvisa to issue SCPI commands to the device to load, start, and stop satellite simulations. I am communicating with the device through USB.

hgrecco, 2015-05-16:
try to put that command first to find out if a previous command is affecting the ability of your device to receive further instructions.

dillharr, 2015-05-18:
I have tried moving it farther up the script. When I do this the next command that talks to the GSG-62 experiences the same timeout. I will try contacting Spectracom to see if they have heard of this as well. I'm almost starting to think it might be something with GSG-62 and not pyvisa.

hgrecco, 2015-05-18:
try the same sequence win NI-MAX and let us know.

dillharr, 2015-05-20:
I finally found what was causing my issue. There were a couple of issues actually. One of them was a setting issue on the satellite simulator. The other issue was a timing problem in my code. Thanks for all your help.

closed 2015-05-20 by dillharr

csagar, 2016-05-12:
@dillharr 

Can you confirm how were you able to resolve this issue?

I am having the exact same issue with this snippet

```
import visa
rm = visa.ResourceManager()
rm.list_resources()
inst = rm.open_resource('TCPIP0::192.168.1.1::56001::SOCKET')
print inst.query('*IDN?\r\n')
```

Thanks.

rakansha, 2016-06-29:
what was the timing problem and how did you rectify it?
I have an operation timed out error too
https://github.com/hgrecco/pyvisa/issues/221#issuecomment-229260596

dillharr, 2016-06-29:
@rakansha sorry it took so long to get back to you. I reviewed my notes on this issue and it appears I was issuing SCPI command to change the plan and then start the plan without a small delay between them. These satellite simulators we have are quite old and don't seem to like a bunch of commands just thrown at them.

rakansha, 2016-06-29:
@dillharr Thanks my issue then seems to be different than you. the first command itself shows Operation timed out. And plus as of now I didn't even execute a set of commands, only tried interacting with instrument.

csagar, 2016-06-29:
@rakansha Can you share you code snippet to better understand the issue? The termination character is key.

rakansha, 2016-06-29:
https://github.com/hgrecco/pyvisa/issues/221#issuecomment-229260596

Here..I have posted that problem in the above link..it contains the code snippet

csagar, 2016-06-29:
Hmm... I assume you have also added the termination character in that snippet. 

Also checking this [Timeout issue](https://github.com/hgrecco/pyvisa-py/issues/3) it seems there is an open issue on Linux machines using PyVisa for USB. The author posted his code snippet which is similar to yours but is on MacOS. You might want to give a try with a different OS may be, if that is even an option.

rakansha, 2016-06-30:
@csagar I will look into it.thanks

muditbatra, 2019-06-27:
Hi, I am facing the same issue, I am using Keysight 33600A waveform generator,

MatthieuDartiailh, 2019-06-27:
I suspect that given the amount of data to transfer it takes actually longer than the configured timeout. A simple fix is to increase the timeout around that operation. You can experiment setting the timeout to None and once you get an idea of the time it actually takes you can go back to using a finite value (to avoid getting stuck forever). 
```python
old_timeout = AFG.timeout
AFG.timeout = 120000  # Set the timeout to 2 minutes
try:
    AFG.write('DATA:ARB Test, '+ frame_data)
finally:
    AFG.timeout = old_timeout
```

muditbatra, 2019-06-28:
Hi,

I did try this too but with no effect I even deleted the timeout to make it infinity but that too didn't work and now I am starting to feel it might be instrument's issue itself. Will raise a query with keysight too regarding it.

Thanks for your suggestions @MatthieuDartiailh