pyvisa/pyvisa #158: error: unpack_from requires a buffer of at least 4 bytes

cidadao, 2015-05-26:
I can open a resource (RIGOL DM3058E, digital multimeter) using USB but I'm unable to get it's IDN.
This is the log and the error I get:

<code>
  File "test_usbtmc.py", line 11, in <module>
    print(inst.ask("*IDN?"))
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 407, in query
    return self.read()
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Python27\lib\site-packages\pyvisa-py\highlevel.py", line 263, in read
    return self.sessions[session].read(count)
  File "C:\Python27\lib\site-packages\pyvisa-py\usb.py", line 102, in read
    usb.USBError)
  File "C:\Python27\lib\site-packages\pyvisa-py\sessions.py", line 313, in _read
    current = reader()
  File "C:\Python27\lib\site-packages\pyvisa-py\usb.py", line 96, in <lambda>
    return self._read(lambda: self.interface.read(1),
  File "C:\Python27\lib\site-packages\pyvisa-py\protocols\usbtmc.py", line 311, in read
    response = BulkInMessage.from_bytes(resp)
  File "C:\Python27\lib\site-packages\pyvisa-py\protocols\usbtmc.py", line 94, in from_bytes
    msgid, btag, btaginverse = struct.unpack_from('BBBx', data)
struct.error: unpack_from requires a buffer of at least 4 bytes
</code>

PyVisa info:
<code>
Machine Details:
   Platform ID:    Windows-8-6.2.9200
   Processor:      Intel64 Family 6 Model 69 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.9
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 10 2014 12:28:03 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8.dev0

Backends:
   ni:
      Version: 1.8.dev0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
   py:
      Version: 0.2.dev2
      ASRL INSTR: Available via PySerial (2.7)
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP INSTR: Available
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.0rc1). Backend: libusb1
</code>

I also tried runing it with libusb-0.1 but the resouce don't even open and withg Python 3.4 I get an error stating that "ConfigParser" doesn't exist (it was renamed to configparser, I think).

Any ideas on how it can be solved?

hgrecco, 2015-05-26:
Can you print the content of `data`? Regarding `ConfigParser`, The only usage is in util.py and shoud work:
```
    try:
        from ConfigParser import SafeConfigParser as ConfigParser, NoSectionError
    except ImportError:
        from configparser import ConfigParser, NoSectionError
```

cidadao, 2015-05-26:
Data:
<pre><code>
C:\Users\mribeiro\Copy\vm_shared_cloud\python>python test_usbtmc.py
(u'USB0::6833::1416::DM3R164851896::0::INSTR',)
data l94= ☻☻² ;   ☺   Rigol Technologies,DM3058E,DM3R164851896,01.01.00.02.01.00

data l97= ☻☻² ;   ☺   Rigol Technologies,DM3058E,DM3R164851896,01.01.00.02.01.00

data l94=
Traceback (most recent call last):
  File "test_usbtmc.py", line 11, in <module>
    print(inst.ask("*IDN?"))
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 407, in query
    return self.read()
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "C:\Python27\lib\site-packages\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Python27\lib\site-packages\pyvisa_py-0.2.dev2-py2.7.egg\pyvisa-py\highlevel.py", line 263, in read
    return self.sessions[session].read(count)
  File "C:\Python27\lib\site-packages\pyvisa_py-0.2.dev2-py2.7.egg\pyvisa-py\usb.py", line 102, in read
    usb.USBError)
  File "C:\Python27\lib\site-packages\pyvisa_py-0.2.dev2-py2.7.egg\pyvisa-py\sessions.py", line 313, in _read
    current = reader()
  File "C:\Python27\lib\site-packages\pyvisa_py-0.2.dev2-py2.7.egg\pyvisa-py\usb.py", line 96, in <lambda>
    return self._read(lambda: self.interface.read(1),
  File "C:\Python27\lib\site-packages\pyvisa_py-0.2.dev2-py2.7.egg\pyvisa-py\protocols\usbtmc.py", line 312, in read
    response = BulkInMessage.from_bytes(resp)
  File "C:\Python27\lib\site-packages\pyvisa_py-0.2.dev2-py2.7.egg\pyvisa-py\protocols\usbtmc.py", line 95, in from_bytes
    msgid, btag, btaginverse = struct.unpack_from('BBBx', data)
struct.error: unpack_from requires a buffer of at least 4 bytes
</code></pre>
Looks good the first two prints (the first time the function "from_bytes" is called), but then data is empty.

Regarding ConfigParser:
<pre><code>
C:\Users\mribeiro\Copy\vm_shared_cloud\python>C:\Python34\python.exe test_usbtmc.py
Traceback (most recent call last):
  File "test_usbtmc.py", line 4, in <module>
    import visa
  File "C:\Python34\lib\site-packages\visa.py", line 1, in <module>
    from pyvisa.visa import *
  File "C:\Python34\lib\site-packages\pyvisa\__init__.py", line 34, in <module>
    import ConfigParser, os, sys
ImportError: No module named 'ConfigParser'
</code></pre>

hgrecco, 2015-05-27:
There is something very wrong with your PyVISA installation. Somehow you have really new stuff (like `python -m visa info`) but really old stuff (that particular line that is failing has not been there since May 2012, see the current version [here](
https://github.com/hgrecco/pyvisa/blob/master/pyvisa/__init__.py))

My guess is that something did not working during the uninstall process. Please uninstall pyvisa `pip uninstall pyvisa` and check that the following files and folders are removed.
- C:\Python34\lib\site-packages\visa.py
- C:\Python34\lib\site-packages\pyvisa

Then install again.

Please fix this first so we are sure that this is a real bug.

cidadao, 2015-05-29:
I'm using Python 2.7.9, does it make any difference?
I uninstalled both pyvisa and pyvisa-py, checked the folder and files were removed and then I installed them again.
Now this is the error I'm getting:
<pre><code>
C:\Users\mribeiro\Copy\vm_shared_cloud\python>python test_pyvisa.py
Traceback (most recent call last):
  File "test_pyvisa.py", line 4, in <module>
    print(rm.list_resources())
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 669, in list_resources
    raise NotImplementedError
NotImplementedError
</code></pre>~

<pre><code>
import visa
rm = visa.ResourceManager("@py")
print(rm.list_resources())

print("Open resource")
inst = rm.open_resource("USB0::6833::1416::DM3R164851896::0::INSTR")
print("Read")
print("Freq: %s" % inst.ask(":MEAS:FREQ?"))
</code></pre>

I'm considering that the problem may be related with underlying modules -> USB. Do you agree?
I'm using libusb1-0 (downloaded from libusb.info) and pyusb downloaded from github page.

hgrecco, 2015-08-24:
Is this still a problem? Feel free to reopen if needed.

closed 2015-08-24 by hgrecco

0xDEADBEEF-ASAP, 2024-03-31:
I got the same error on one machine but not on other machine_ the rest (freash arch install + other) are exactly the same ! Both machines are little endian.

JaminMartin, 2024-06-11:
@0xDEADBEEF-ASAP @hgrecco I too am having this issue on a fresh install (Ubuntu 22.04 LTS) with a siglent oscilloscope. Switching the backend to ni-visa seems to resolve the issue. But obviously NI doesn't target other Linux distributions so it isn't a general fix. 

I am using libusb-dev for PyUSB.

MatthieuDartiailh, 2024-06-11:
@JaminMartin Could you open a fresh issue following the template ? Several things have changed since the original report and having a complete picture may help.