pyvisa/pyvisa #160: Keysight IO Library 17.1

intecele, 2015-06-03:
I tried to get pyvisa up an running with Keysight IO Library 17.1 - there i alway have empty answers from "visa.ResourceManager()". With IO Library 16.3. it works. (Keysight doesn´t offer previous versions, but i had one in 82357B package).
What have i dont wrong ?

sorry i closed the issue by wrong click.

hgrecco, 2015-06-03:
I am not familiar with that library. Is there any reason why you cannot use NI-VISA? Even if you cannot list resources, can you open them?

phsdv, 2015-06-04:
Works for me. I have 17.1.19313.5 installed

>import visa
>rm = visa.ResourceManager(visa_library="C:\\Windows\\SysWOW64\\visa32.Agilent Technologies.dll")
>print(rm)

results in:
>Resource Manager of Visa Library at C:\Windows\SysWOW64\visa32.Agilent Technologies.dll

Maybe it helps if you put in the exact path to dll like I did in above example.

intecele, 2015-06-04:
Thanks a lot for your supply. I now tried another computer, where V16.3 
of the lib was installed, which works. I updated the io Lib to V17.1 and 
it still works. It´s a great work, what is done with the pyvisa. Thank 
you very much.

import visa
rm = visa.ResourceManager()
print(rm.list_resources())

gmclam, 2015-06-08:
I have a similar problem.  I just switched my machines from 32 bit Python to 64 bit python which required that I use pyVisa later than v1.5, I am using 1.6.1.  I am using Agilent (Keysight) IO libraries v16.x because I use the Agilent USB->GPIB controller which requires it (won't work with NI-Visa).  Upon making this switch I find that responses are typically "one behind".  I'll ask a scope *IDN? and get a varied response (could simply be '0').  Then I'll ask the same scope *OPT? and I'll get the response to the *IDN? query and so on.  Code has been working just fine for 4 years on dozens of machines under pyVisa <1.5.  Thoughts?

hgrecco, 2015-06-09:
It might be a difference in the end of line. PyVISA 1.4 was changing VISA defaults on creation. You might need to check the manual and set the right read_termination and write_termination. My guess is that it is '\r\n` and is currently set to `\r` or something similar.

gmclam, 2015-06-09:
Thank you for the reply.  I see in the docs where you can set terminations for each device.  I think it needs to be global (all instruments on a given interface to solve my issues).  I don't see an easy way to implement this without modifying each and every one of my instrument drivers that utilize pyVisa (there are many), which then runs the risk that they would not be backward compatible with those using v1.4 or earlier (under 32 bit).  Any further direction or examples would be appreciated.

hgrecco, 2015-06-09:
You should not really go back to PyVISA 1.4. Current PyVISA supports Python 2.6+ and 32/64 bit libraries. You can see some of the differences here:

http://pyvisa.readthedocs.org/en/master/migrating.html

PyVISA 1.4 respect the VISA default for each interface type which I think is a more reasonable thing to do, requiring the user to change it explicitly.  We will not add in the trunk a way to change these things globally, but you can do it easily in you own code if that is really what you need.

gmclam, 2015-06-09:
I write instrument drivers that are used throughout the company (Intel).  I don't control what code versions others are using.  I know many are using Python as early as 2.5 and 32 bit.  This is the only issue keeping my code from working under 64 bit (which requires PyVisa 1.5+).  Thanks

gmclam, 2015-06-09:
I am not even getting the single instrument change to solve my problem.  I tried setting the read_termination to '\r', '\n', '\r\n' and '' and each produces the same result.   Here are the queries and replies (with one setting).
Scope1.read_termination = '\r\n'
Scope1.fnAskRawString(1,'*IDN?')
TEKTRONIX,DSA70804B,B010297,CF:91.1CT FV:5.3.5 Build 22
Scope1.fnAskRawString(1,'*OPT?')
TEKTRONIX,DSA70804B,B010297,CF:91.1CT FV:5.3.5 Build 22
Scope1.fnAskRawString(1,'*IDN?')
10XL,MTH,PTH1,ASM,RTE,JA3,TDSPTD,DJA

hgrecco, 2015-06-09:
what is `fnAskRawString`?

gmclam, 2015-06-09:
def fnAskRawString(self,fWait,sCommand):
        """
        Perform ASK Command with Raw String
        Parameters: string/int/float: time to wait after write before read
                    string to be written
        Returns:    string from instrument
                     OR bool: False if not connected
        """
        if (self.hDevice == None):
            return False
        if type(fWait) == str:
            try:
                fWait = eval(fWait)
            except:
                fWait = 0
        self.hDevice.delay = fWait
        Result = self.hDevice.ask(sCommand)
        return Result

hgrecco, 2015-06-09:
`delay` was renamed to `query_delay`. Changing that should do the trick.

Optionally `ask` was renamed to `query` (but you can still use `ask` for the time being). Optionally also, you can provide to query the delay:
```python
self.hDevice.query(sCommand, delay=fWait)
```

gmclam, 2015-06-09:
Thanks for the reply.  Yeah I read about all the changes, and am in the process of not using the ask method at all.  I will recode my drivers to use write and read and use a sleep command for the delay in between so that the units don't change (some places use seconds, and I see some use mS) and keep compatibility with all present versions of PyVisa. Changing the .delay to .query_delay did solve this specific issue.  I need to update all the functions to see if it resolves all of my issues.  Thanks again.

hgrecco, 2015-06-09:
You are welcome. You might look at lantz, which is a project that aims to have higher level drivers. The current docs: http://lantz.readthedocs.org/

And we are not undertaking a Great refactoring https://github.com/LabPy with the opening of the project to many more people. It helps a lot when building large scale apps.

closed 2015-08-24 by hgrecco