pyvisa/pyvisa #161: Timeout expired before operation completed.

fromage88, 2015-06-14:
When trying to conntect to mt DG1022A I get a timeout error.
Other devices work perfectly.

> import visa
> visa.log_to_screen()
> rm = visa.ResourceManager() print
> rm.list_resources()
> print rm.list_resources()[0]
> my_instrument = rm.open_resource(rm.list_resources()[0])
> print  my_instrument.query('*IDN?')

I have posted the output here
[output](http://pastebin.com/byyRj8eG)

I would be glad if someone could help me.

jayich, 2015-07-12:
I am getting a similar error at the moment with a Rigol DG1022.  The errors in the post above look the same as what I am seeing.  Help here would be greatly appreciated.  If more information is needed, please let me know.

jayich, 2015-07-12:
Update, I can do the following (rigol is the ope_resource).

```python
>>> rigol.write('*IDN?')
(7L, <StatusCode.success: 0>)
>>> rigol.read()
u'RIGOL TECHNOLOGIES,DG1022 ,DG1D150900538,,00.03.00.09.00.02.11\n'
```

fromage88, 2015-08-19:
I had wrong syntax somewhere else in the code. after fixing that in worked

closed 2015-08-19 by fromage88

hgrecco, 2015-08-19:
Thanks for writing back.