pyvisa/pyvisa #162: Allow setting up padding in IEEE block header

wurbanski, 2015-06-17:
When using write_binary_data(), IEEE header is being created properly, but there are some deviced (eg. Keysight U253x family) which required the header to be of specified length, in this case left-passed with zeros to 8 digits, eg. with data of 123456 bytes, the header should look like this:
```
#800123456
```
and instead it is formed (aligning with standards though) as:
```
#6123456
```
Adding a way of specifying the padding length would be really helpful when working with these devices.

hgrecco, 2015-08-24:
Sorry for the delay but I completely missed this.

We could do with `write_binary_values` some similar to what is done in `query_binary_values`. There is an argument to control the header format (called `header_fmt`). You could add an option (`keysight`) that maps to the right format and change the code accordingly. I like that more than the padding as it allows for other formats in the future. You could add a padding option `to_ieee_block` which is an internal function

hgrecco, 2015-10-06:
any interest in following this?

wurbanski, 2015-10-07:
This time I must have missed it :)

I'd gladly submit a PR with this fix this next week if that's fine for you.

closed 2018-01-25 by MatthieuDartiailh