pyvisa/pyvisa #163: Two or more resource manager sessions have the same session handler

zhgary, 2015-06-24:
Python 3.4.3

Two or more resource manager sessions created within the same script, within different modules imported into the same script, or within the same Python shell will have the same session handler. Is this supposed to be intentional or would it possibly be an issue with my PyVISA installation?

This becomes an issue where two different modules are being imported; if one module closes its resource manager session, the other module's resources become invalid.

Example 1:
```
>>> rm2 = visa.ResourceManager()
>>> rm = visa.ResourceManager()
>>> rm.session
53711736
>>> rm2.session
53711736
```

Example 2:
```
import testA
import testB

A = testA.A()
B = testB.B()

A.begin()
B.begin()

A.end()
B.end()
```

where testA.py contains:
```
import visa

class A:
    rm = None
    inst = None

    def begin(self):
        self.rm = visa.ResourceManager()
        self.inst = self.rm.open_resource("ASRL1")

    def end(self):
        self.inst.close()
        self.rm.close()
```

testB.py contains:
```
import visa

class B:
    rm = None
    inst = None

    def begin(self):
        self.rm = visa.ResourceManager()
        self.inst = self.rm.open_resource("ASRL2")

    def end(self):
        self.inst.close()
        self.rm.close()
```

Example 2 fails at `self.inst.close()` in module testB with the following trace:
```
  File "C:\Programs\Python34\lib\site-packages\pyvisa-1.8.dev0-y3.4.egg\pyvisa\resources\resource.py", line 218, in clo
se
    self.before_close()
  File "C:\Programs\Python34\lib\site-packages\pyvisa-1.8.dev0-py3.4.egg\pyvisa\resources\resource.py", line 210, in before_close
    self.__switch_events_off()
  File "C:\Programs\Python34\lib\site-packages\pyvisa-1.8.dev0-py3.4.egg\pyvisa\resources\resource.py", line 227, in __switch_events_off
    self.disable_event(constants.VI_ALL_ENABLED_EVENTS, constants.VI_ALL_MECH)
  File "C:\Programs\Python34\lib\site-packages\pyvisa-1.8.dev0-py3.4.egg\pyvisa\resources\resource.py", line 282, in disable_event
    self.visalib.disable_event(self.session, event_type, mechanism)
  File "C:\Programs\Python34\lib\site-packages\pyvisa-1.8.dev0-py3.4.egg\pyvisa\ctwrapper\functions.py", line 408, in disable_event
    return library.viDisableEvent(session, event_type, mechanism)
  File "C:\Programs\Python34\lib\site-packages\pyvisa-1.8.dev0-py3.4.egg\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
```

hgrecco, 2015-06-24:
This is the intended behaviour as it calls viOpenDefaultRM. From the VISA manual.

> The viOpenDefaultRM() function must be called before any VISA operations can be invoked. 
> The first call to this function initializes the VISA system, including the Default Resource Manager 
> resource, and also returns a session to that resource. Subsequent calls to this function return unique 
> sessions to the same Default Resource Manager resource.

You should instantiate the ResourceManager only once in you application.

hgrecco, 2015-08-24:
I am closing this. Please feel free to open another or reopen if you need further assistance.

closed 2015-08-24 by hgrecco

dougthor42, 2023-02-08:
> You should instantiate the ResourceManager only once in you application.

Question: is this still the recommendation, _provided I don't close any of the `ResourceManager`s_? Or is it OK to instantiate many because the underlying `session` is the same?

As an example, I have classes that look like the following and things appear to work just fine. /shrug.

```python
class SwitchMatrix:
    def __init__(self):
        self.rm = pyvisa.ResourceManager("@py")
        self.instr = self.rm.open_resource("some VISA addr")

class SMU:
    def __init__(self):
        self.rm = pyvisa.ResourceManager("@py")
        self.instr = self.rm.open_resource("a different VISA addr")

def test():
    switch_matrix = SwitchMatrix()
    smu = SMU()
    assert smu.instr.query("*IDN?").startswith("Keysight Technologies,B2901A)
    assert switch_matrix.instr.query("*IDN?").startswith("Agilent Technologies,34980A")
```

dougthor42, 2023-02-08:
Oh actually it looks like every instantiation of RM _for a given backend_ ends up being the same object:

```python
>>> import pyvisa
>>> rm1 = pyvisa.ResourceManager("@py")
>>> rm2 = pyvisa.ResourceManager("@sim")
>>> id(rm1)
140712192195792
>>> id(rm2)
140712130166944
>>> id(pyvisa.ResourceManager("@sim"))
140712130166944
>>> id(pyvisa.ResourceManager("@sim"))
140712130166944
>>> id(pyvisa.ResourceManager("@py"))
140712192195792
```

So I think I answered my own question as "You can call `ResourceManager()` as many times as you want".