pyvisa/pyvisa #166: time out error ONLY when doing read... write/query works fine

davidlogan96, 2015-07-20:
Hi, 
I'm using pyvisa to control a lab instrument, keysight e5071c, via USB.
Using NI-VISA 5.4.1 library for linux.

Currently i can use write and query to set up my VNA, run it, and get back a formatted data array. However, now, i'm trying to transfer a file from the E5071C to my PC. The scpi command according to programmer manual is MMEM:TRAN? D:\davidsSweepResult.s1p. I do this using .write, then try reading the data, however... on the read, i get a timeout error. I tried using the query command instead, and still a read timeout error, as expected. I also have tried using read_raw, and also have set the termination character manually just in case.

This whole issue is strange to me since i've been using query commands and getting back data perfectly fine, but ONLY when i try the scpi command, mmem:tran, is when it poops out.

here is the error output.

Traceback (most recent call last):
  File "/nfs/site/home/davidmlo/PMO/oPMO/davidmlo/PyCharm/automate.py", line 64, in <module>
    main()
  File "/nfs/site/home/davidmlo/PMO/oPMO/davidmlo/PyCharm/automate.py", line 52, in main
    data = vna.read('\n')
  File "/usr/local/anaconda/lib/python2.7/site-packages/PyVISA-1.7-py2.7.egg/pyvisa/resources/messagebased.py", line 315, in read
    message = self.read_raw().decode(enco)
  File "/usr/local/anaconda/lib/python2.7/site-packages/PyVISA-1.7-py2.7.egg/pyvisa/resources/messagebased.py", line 286, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/anaconda/lib/python2.7/site-packages/PyVISA-1.7-py2.7.egg/pyvisa/ctwrapper/functions.py", line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "/usr/local/anaconda/lib/python2.7/site-packages/PyVISA-1.7-py2.7.egg/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
---------------------------------------------------------------------------------------------------------

Also, my visa is set up right i believe, I ran:
python -c "from pyvisa import util; util.get_debug_info()"
and got:

Machine Details:
   Platform ID:    Linux-2.6.18-308.el5-x86_64-with-redhat-5.8-Tikanga
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/local/anaconda/bin/python2
   Version:        2.7.9
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-1)
   Bits:           32bit
   Build:          Mar  9 2015 16:21:41 (#default)
   Unicode:        UCS4

PyVISA Version: 1.7

Backends:
   ni:
      Version: 1.7 (bundled with PyVISA)
      #1: /usr/local/vxipnp/linux/bin/libvisa.so.7:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
   py:
      Version: 0.2.dev2
      ASRL INSTR: Available via PySerial (2.7)
      TCPIP INSTR: Available
      USB RAW: Available via PyUSB (1.0.0b1). Backend: libusb0
      USB INSTR: Available via PyUSB (1.0.0b1). Backend: libusb0
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available
---------------------------------------------------------------------------------------------------------------

MORE INFO on my instrument per "instr.resource_info"

(ResourceInfo(interface_type=<InterfaceType.usb: 7>, interface_board_number=0, resource_class=u'INSTR', resource_name=u'USB0::0x0957::0x0D09::MY46521455::0::INSTR', alias=None), <StatusCode.success: 0>)
-----------------------------------------------------------------------------------------------------------------

If anyone could help me understand why i get a timeout error only on the mmem:tran scpi command, it would be greatly appreciated!

Thanks,

David

davidlogan96, 2015-07-20:
here is a relevant snippet of the debug code

2015-07-20 15:22:49,619 - pyvisa - DEBUG - viWrite(166781520L, "MMEM:TRAN? 'd:\touchstonefile.s1p'\r\n", 35, 'c_ulong(35L)') -> 0
2015-07-20 15:22:49,619 - pyvisa - DEBUG - USB0::0x0957::0x0D09::MY46521455::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2015-07-20 15:22:51,621 - pyvisa - DEBUG - viRead(166781520L, <ctypes.c_char_Array_20480 object at 0x55d86dac>, 20480, 'c_ulong(0L)') -> -1073807339
2015-07-20 15:22:51,621 - pyvisa - DEBUG - USB0::0x0957::0x0D09::MY46521455::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

davidlogan96, 2015-07-20:
Wow.. I feel very dumb. Apparently, all i had wrong in my code was the filename... it's case sensitive. I initially didn't think it was because i was able to query the file system with d:\ or D:\ but the file names themselves are I guess. Oh well.

hgrecco, 2015-07-20:
I was just writing an answer. I am glad that it worked!

davidlogan96, 2015-07-21:
also, timeout errors occurred because i didn't have a null terminator set.

closed 2015-07-21 by davidlogan96