pyvisa/pyvisa #168: Cannot connect via LeCroy VICP

mikekohlhoff, 2015-07-23:
I'm trying to connect to a LeCroy Wavesurfer 400 series via their VICP VISA passport (TCP/IP) with PyVISA 1.7 under Windows7/32bit and NI-VISA 5.4.1:

    import visa
    rm = visa.ResourceManager()
    scope = rm.open_resource("VICP::169.254.201.2::INSTR")
    print(scope.query("*IDN?"))

I get the following error:

    Warning (from warnings module): File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1378 alias_if_exists)

    VisaIOWarning: VI_WARN_EXT_FUNC_NIMPL (1073676457): The operation succeeded, but a lower level driver did not implement the extended functionality.

    Traceback (most recent call last): File "C:\path\scopeTest.py", line 4, in scope = rm.open_resource("VICP::169.254.201.2::INSTR")

    File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1614, in >open_resource info = self.resource_info(resource_name)

    File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1584, in >resource_info

    raise ValueError('Could not parse resource: %s (error code %s)' % >(resource_name, ret))

    ValueError: Could not parse resource: VICP::169.254.201.2::INSTR (error code None)

Under LabVIEW connection to and communication with the device under this VICP address works. Also, when using` TCPIP::` instead of `VICP::` in open_resource() connection is established without warnings and errors and `print(scope.ask("*IDN?"))` works, but other device commands as `scope.write("C1:VDIV .02")` do not work

    print(scope.query("C1:VDIV .02"))
    WARNING : CURRENT REMOTE CONTROL INTERFACE IS TCPI

concluding that for correct device control the VICP passport has to be used. When using PyVISA 1.5 but otherwise same configuration and trying to connect via:

    scope = visa.instrument("VICP::169.254.201.2::INSTR")
   
gives:

    Warning (from warnings module): File "C:\Python27\lib\site-packages\pyvisa-1.5-py2.7.egg\pyvisa\ctwrapper\functions.py", line 1208 alias_if_exists)

    VisaIOWarning: VI_WARN_EXT_FUNC_NIMPL (1073676457): The operation succeeded, but a lower level driver did not implement the extended functionality.

    Warning (from warnings module):
    File "C:\Python27\lib\site-packages\pyvisa-1.5-py2.7.egg\pyvisa\highlevel.py", line 315 return Instrument(resource_name, resource_manager=self, **kwargs)

    UserWarning: given resource was not an INSTR but Unknown

For earlier versions of PyVISA this was discussed for an empty string return of the instrument:

http://osdir.com/ml/python.pyvisa.devel/2007-07/msg00003.html
and
http://sourceforge.net/p/pyvisa/bugs/5/

I suspect the problem to be the **InterfaceType** in *C:\Python27\lib\site-packages\pyvisa\constants.py* not to exist for the VICP connection protocol in PyVISA. When `resource_info()`, the line that throws the error, in *C:\Python27\lib\site-packages\pyvisa\highlevel.py* is called

    def resource_info(self, resource_name):  
           """Get the extended information of a particular resource

          :param resource_name: Unique symbolic name of a resource.

          :rtype: :class:`pyvisa.highlevel.ResourceInfo`
          """
          ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
          if err == constants.StatusCode.success:
              return ret
          raise ValueError('Could not parse resource: %s (error code %s)' %(resource_name, ret))

seems to result in a mismatch with the *interface_type*,  or might it be a problem with the *resource_class* and a return issue as with the older versions of PyVISA?

hgrecco, 2015-07-23:
VICP is not a standard VISA interface type but is supported via a passport. We can add it as we did in #115 for `RSNRP`. You need
1. find out the constant that relates to this interface type
2. Add it to InterfaceType Enum as done [here](https://github.com/hgrecco/pyvisa/blob/22a9228a888faaeea8292621a9a4995c4a66c506/pyvisa/constants.py#L685)
3. Create a Resource Class for this. If you are unsure which one it should be, just use the most basic one which is MessageBasedResource as done [here](https://github.com/hgrecco/pyvisa/blob/22a9228a888faaeea8292621a9a4995c4a66c506/pyvisa/resources/messagebased.py#L545)

mikekohlhoff, 2015-07-23:
From the thread for the RSNRP it is not clear where this constant comes from other than trying to initiate connection with `inst = rm.open_resource(rm.list_resources())` in the initial code snippet. 
The scope is not listed when `rm.list_resources()`, which is default for LAN connections?

hgrecco, 2015-07-23:
`list_resources` returns a list, so you should not pass the result to `open_resource`. You need a single element from that list.

The constants should be documented in the VICP documentation or header file of the example code. Maybe also you can find it in NI-MAX when opening the resource and then querying the interface type.

mikekohlhoff, 2015-07-23:
Referred to the part of the code from #115 for `RSNRP` passport: 

    rm = visa.ResourceManager()
    a = rm.list_resources()
    print a
    (u'RSNRP::0x00a8::100906::INSTR',)

    inst = rm.open_resource(a[0])
    
    ...
    
    ValueError: 33024 is not a valid InterfaceType

Here, it is not clear where this constant derives from.
Otherwise I was not able to find this constant anywhere (READMEs, documentations, example code etc.).

hgrecco, 2015-07-23:
The constant is defined inside the VICP Passport and therefore should be documented in the VICP documentation or header file of the example code. This is not standard in VISA and therefore it is not in the manuals that we have available.

Additionally, the error that you are getting is before this constant is used by PyVISA. You said that it works with LabVIEW but are you using the standard VISA vi?

mikekohlhoff, 2015-07-23:
There seems to be no detailed documentation out, I contacted the manufacturer for further information.
It does work in LabVIEW with the standard VISA vi's using the constant `VICP::169.254.201.2::INSTR`.

hgrecco, 2015-07-23:
Do you have an example in C/C++? That is really close to PyVISA and therefore it will be easy to compare.

mikekohlhoff, 2015-07-24:
The following code works under W7 with the LeCroy scope via VICP:

    #include <C:\VXIPNP\WinNT\Include\visa.h>
    #include "stdio.h"

    int main()
    {
    ViSession defaultRM, instr;
    char buf [256] = {0};
    
    viOpenDefaultRM(&defaultRM);
    viOpen(defaultRM, "VICP::169.254.201.2::INSTR", VI_NULL, VI_NULL, &instr);
    
    viPrintf(instr, "*IDN?\n");
    viScanf(instr, "%t", &buf);
    printf("Instrument: %s\n", buf);
    
    viClose(instr);
    viClose(defaultRM);
    }

hgrecco, 2015-07-24:
Then we should be able to make it work. Below I am copying an (almost) equivalent python program but with the logging enabled.
```
import visa
visa.log_to_screen()
rm = visa.ResourceManager()
my_instrument = rm.open_resource('VICP::169.254.201.2::INSTR')
print(my_instrument.query('*IDN?'))
```
Please run it and paste the output here. You will see the visa commands being called.

Note: here might be a typo as I am writing this on a rush. I said almost because pyvisa uses the lower level viRead and viRead instead of viPrintf and viScanf but should not be a problem. In any case you can rewrite your c program.

mikekohlhoff, 2015-07-24:
Python output:

    2015-07-24 16:51:08,589 - pyvisa - DEBUG - No user defined library files
    2015-07-24 16:51:08,641 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll', 
    'C:\\WINDOWS\\system32\\visa32.dll']
    2015-07-24 16:51:08,868 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
    2015-07-24 16:51:08,869 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
    2015-07-24 16:51:09,775 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x01F8DB70>',) -> 0
    2015-07-24 16:51:09,778 - pyvisa - DEBUG - Created ResourceManager with session 21029936
    2015-07-24 16:51:09,783 - pyvisa - DEBUG - viParseRsrcEx(21029936L, 'VICP::169.254.201.2::INSTR', 'c_ushort(6)', 
    'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x028BEAD0>, <ctypes.c_char_Array_256 object at 0x028BEB70>, 
    <ctypes.c_char_Array_256   object at 0x028BEC10>) -> 1073676457

    Warning (from warnings module):
        File "C:\Python27\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1378
    alias_if_exists)
    VisaIOWarning: VI_WARN_EXT_FUNC_NIMPL (1073676457): The operation succeeded, but a lower level driver did not implement the extended functionality.

    Traceback (most recent call last):
      File "C:\Users\tpsadmin\Desktop\scopeTestVisa\scopeTest.py", line 5, in <module>
        my_instrument = rm.open_resource('VICP::169.254.201.2::INSTR')
      File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1614, in open_resource
    info = self.resource_info(resource_name)
      File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1584, in resource_info
    raise ValueError('Could not parse resource: %s (error code %s)' % (resource_name, ret))
    ValueError: Could not parse resource: VICP::169.254.201.2::INSTR (error code None)

Confirming that the .cpp code also works with `viWrite()` and `viRead()`

hgrecco, 2015-07-25:
It is interesting because it actually fails before PyVISA is doing anything interesting. It  says that the VICP Passport does not handle properly viParseRsrcEx function, which is actually a very basic function. 

Try the following which is a little bit lowerlevel:
```python
import visa
visa.log_to_screen()
rm = visa.ResourceManager()
print(rm.visalib.parse_resource(rm.session, 'VICP::169.254.201.2::INSTR'))
```

mikekohlhoff, 2015-07-26:
2015-07-26 17:32:31,986 - pyvisa - DEBUG - No user defined library files
    2015-07-26 17:32:32,019 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll']
    2015-07-26 17:32:32,279 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
    2015-07-26 17:32:32,282 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
    2015-07-26 17:32:33,523 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x01E7DB70>',) -> 0
    2015-07-26 17:32:33,526 - pyvisa - DEBUG - Created ResourceManager with session 20702256
    2015-07-26 17:32:33,529 - pyvisa - DEBUG - viParseRsrc(20702256L, 'VICP::169.254.201.2::INSTR', 'c_ushort(6)', 'c_ushort(0)') -> 0
    (ResourceInfo(interface_type=<InterfaceType.tcpip: 6>, interface_board_number=0, resource_class=None, resource_name=None, alias=None), <StatusCode.success: 0>)

hgrecco, 2015-07-26:
A few interesting things. First is that `viParseRsrcEx` is not supported by the VICP passport but `viParseRsrc` is. I will implement a fallback in the code. The second is that it reports `6` as interface_type which means TCPIP as noted. Therefore a new Resource and constant migh not be needed from the pyvisa side.

hgrecco, 2015-07-27:
I pushed a few changes. I am not sure that it will work but it should take your further. Please install from github and try again the test script.

mikekohlhoff, 2015-07-27:
2015-07-27 10:15:29,323 - pyvisa - DEBUG - No user defined library files
    2015-07-27 10:15:29,329 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll']
    2015-07-27 10:15:29,594 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
    2015-07-27 10:15:29,596 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
    2015-07-27 10:15:30,634 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x01E47EE0>',) -> 0
    2015-07-27 10:15:30,637 - pyvisa - DEBUG - Created ResourceManager with session 20571184
    2015-07-27 10:15:30,638 - pyvisa - DEBUG - viParseRsrc(20571184L, 'VICP::169.254.201.2::INSTR', 'c_ushort(6)', 'c_ushort(0)') -> 0
    (ResourceInfo(interface_type=<InterfaceType.tcpip: 6>, interface_board_number=0, resource_class=None, resource_name=None, alias=None), <StatusCode.success: 0>)

hgrecco, 2015-07-27:
Seems to work, I would say that you try the IDN first and if it works, also your other scripts.

mikekohlhoff, 2015-07-27:
`scope = rm.open_resource('VICP::169.254.201.2::INSTR')` gives the error:

    Traceback (most recent call last):
      File "C:\Users\tpsadmin\Desktop\scopeTestVisa\scopeTest.py", line 5, in <module>
        scope = rm.open_resource('VICP::169.254.201.2::INSTR')
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\highlevel.py", line 1622, in open_resource
        raise ValueError('There is no class defined for %r' % ((info.interface_type, info.resource_class),))
    ValueError: There is no class defined for (<InterfaceType.tcpip: 6>, None)

hgrecco, 2015-07-27:
The problem seem to be that the VICP passport is not properly  implement viParseRsrc (or viParseRsrcEx) which used by PyVISA to decide which Class to instantiate (see that resource_class is None and should be INSTR). You could try to make it work by calling lower level functions.

```python
import visa
visa.log_to_screen()
rm = visa.ResourceManager()
sess = rm.open_bare_resource('VICP::169.254.201.2::INSTR')
print(rm.visalib.write(sess, '*IDN?\n'))
print(rm.visalib.read(sess, 128))
```

If this works, and also work for other commands like `C1:VDIV`, we could try to put it more conveniently into PyVISA. Until now, this was not needed because Passports should properly implement this.

mikekohlhoff, 2015-07-27:
Seems connection to the instrument worked but some issue with the ctype porting of `visalib.write()`

    2015-07-27 17:15:49,944 - pyvisa - DEBUG - No user defined library files
    2015-07-27 17:15:49,948 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll']
    2015-07-27 17:15:49,980 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
    2015-07-27 17:15:49,982 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
    2015-07-27 17:15:50,006 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x01EC7EE0>',) -> 0
    2015-07-27 17:15:50,009 - pyvisa - DEBUG - Created ResourceManager with session 21423080
    2015-07-27 17:15:50,013 - pyvisa - DEBUG - viOpen(21423080L, 'VICP::169.254.201.2::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x01EC7EE0>') -> 0

    Traceback (most recent call last):
      File "C:\Users\tpsadmin\Desktop\scopeTestVisa\scopeTest.py", line 11, in <module>
        print(rm.visalib.write(sess, '*IDN?\n'))
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\ctwrapper\functions.py", line 1864, in write
        ret = library.viWrite(session, data, len(data), byref(return_count))
    ArgumentError: argument 1: <type 'exceptions.TypeError'>: wrong type

hgrecco, 2015-07-27:
Sorry, the line should be:
```python
sess, ret = rm.open_bare_resource('VICP::169.254.201.2::INSTR')
```

mikekohlhoff, 2015-07-27:
Great, that works, output:

    2015-07-27 18:48:01,789 - pyvisa - DEBUG - No user defined library files
    2015-07-27 18:48:01,796 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll', 'C:\\WINDOWS\\system32\\visa32.dll']
    2015-07-27 18:48:01,848 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
    2015-07-27 18:48:01,910 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
    2015-07-27 18:48:01,948 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x01F97EE0>',) -> 0
    2015-07-27 18:48:01,951 - pyvisa - DEBUG - Created ResourceManager with session 20177896
    2015-07-27 18:48:01,954 - pyvisa - DEBUG - viOpen(20177896L, 'VICP::169.254.201.2::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x01F97EE0>') -> 0
    2015-07-27 18:48:01,957 - pyvisa - DEBUG - viWrite(41889648L, '*IDN?\n', 6, 'c_ulong(6L)') -> 0
    (6L, <StatusCode.success: 0>)
    2015-07-27 18:48:01,971 - pyvisa - DEBUG - viRead(41889648L, <ctypes.c_char_Array_128 object at 0x01F97EE0>, 128, 'c_ulong(34L)') -> 0
    ('LECROY,WS452,LCRY0301J17474,6.8.1\n', <StatusCode.success: 0>)

hgrecco, 2015-07-27:
I would say that you try other more sophisticated commands with the same structure. If they work, we do it in a more convenient way.

mikekohlhoff, 2015-07-27:
Confirming, all other commands work with output in format as above for `*IDN?\n`

    (6L, <StatusCode.success: 0>)
    ('LECROY,WS452,LCRY0301J17474,6.8.1\n', <StatusCode.success: 0>)

hgrecco, 2015-07-27:
I have implemented a way to force the resource python class to use.
Pull the latest version and try:
```python
import visa
from pyvisa.resources import MessageBasedResource

visa.log_to_screen()
rm = visa.ResourceManager()
my_instrument = rm.open_resource('VICP::169.254.201.2::INSTR', resource_pyclass=MessageBasedResource)
print(my_instrument.query('*IDN?'))
```

mikekohlhoff, 2015-07-28:
Great, all good now for VICP. Thanks a lot! I will open another thread for an issue that might be related to this.

hgrecco, 2015-07-28:
Test it a lit bit more and close this issue when you are satisfied.

closed 2015-07-28 by mikekohlhoff

gtyhughes, 2023-10-04:
It seems that using MessageBasedResource for this issue no longer works. The following error occurs:
`The following error occurred: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.`

open_bare_resource still does work, though

MatthieuDartiailh, 2023-10-05:
There is also native support for VICP in pyvisa-py. It may be a more robust option.

nine-V, 2024-12-12:
Hello, when I use tcpip to communicate between LabVIEW and the Lecroy oscilloscope, this warning also appears: WARNING : CURRENT REMOTE CONTROL INTERFACE IS TCPIP;
 and an error occurs: The format of the CMR? response is not as expected. Make sure that the Command Header is not being reprogrammed from the value configured in the Default Instrument Setup VI, where it is configured to OFF with the command CHDR OFF. (Default Instrument Setup.VI is called from Initialize.vi.) 
Do you know how to solve this problem? Thank you very much!
[image]

jmmelko, 2025-01-17:
> I have implemented a way to force the resource python class to use. Pull the latest version and try:
> 
> import visa
> from pyvisa.resources import MessageBasedResource
> 
> visa.log_to_screen()
> rm = visa.ResourceManager()
> my_instrument = rm.open_resource('VICP::169.254.201.2::INSTR', resource_pyclass=MessageBasedResource)
> print(my_instrument.query('*IDN?'))

OMG, you saved my life.

Why isn’t this documented anywhere?
Why isn’t the VICP resource listed by the VISA library even though it has been registered in Ni-Max?
Why would VICP natively be supported in pyvisa only by using  the pyvisa Visa library and not by the official Visa library?

I am not sure how many people use VICP but this seems to be a solid feature request to me.

jmmelko, 2025-01-17:
> Hello, when I use tcpip to communicate between LabVIEW and the Lecroy oscilloscope, this warning also appears: WARNING : CURRENT REMOTE CONTROL INTERFACE IS TCPIP;

@nine-V : 
VICP is a non official format used by Lecroy (and other manufacturers?) to communicate with their instruments through TCP-IP.

If you want to send serial text messages to a Lecroy instrument through TCPIP, you have two options:
1/ do as if it were a normal instrument, but add a special 8-byte header with each packet.
The header structure is explained in the instrument docs. The problem is that it contains, among other things, the size of the text message, so you have to calculate the header for each message, and the likeness of making a mistake is very high.

2/ use a modified protocol called VICP.

The way to use VICP is through a so called passport, which consists in additional library files to install. They are manufacturer specific, so you would have to download them from their website. If that’s Lecroy’s, good luck as their registration web form is buggy.
After that installation (which requires administrator privileges), you will have to check that the passport is enabled in NI-MAX.
Lastly, I think you have to add the resource manually in Ni-MAX. Maybe that’s not necessary when using pyvisa, but at least that will enable you to test connection and messages using the virtual instrument panel provided by NI-MAX.

If you’re are in the exact same situations as me, I can send you more info. Otherwise, juste Google the keywords VICP, VICP passport, etc and you’ll eventually find answers.