pyvisa/pyvisa #169: Read timeout with messaged based resource

mikekohlhoff, 2015-07-28:
I want to control a Quantum Composer 9520 series pulse generator via USB under W7 32bit. Line termination for this device is specified as  

    - QUERY FORMAT
        *IDN?<cr><lf>
            Will return: deviceID<cr><lf>

When testing the device in LabVIEW it works (Message Based Settings can be retrieved from property node if needed). 
Simple attempt to open the device in python

    import visa
    rm = visa.ResourceManager()
    delayGen = rm.open_resource('ASRL13::INSTR')

yields

    Traceback (most recent call last):
      File "C:/Users/tpsadmin/Desktop/scopeTestVisa/delGenTest.py", line 8, in <module>
        delayGen = rm.open_resource('ASRL13::INSTR', write_termination='\r\n', read_termination=None)
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\highlevel.py", line 1628, in open_resource
        raise ValueError('There is no class defined for %r' % ((info.interface_type, info.resource_class),))
    ValueError: There is no class defined for (<InterfaceType.asrl: 4>, None)

After fixing the problem in #168 I tried 

    from pyvisa.resources import MessageBasedResource
    delayGen = rm.open_resource('ASRL13::INSTR', resource_pyclass=MessageBasedResource, write_termination='\r\n', read_termination='\r\n')

which gives an error if some arguments are used in the constructor, e.g. `send_end` and `delay` the following error is produced:

    Traceback (most recent call last):
      File "C:/Users/tpsadmin/Desktop/scopeTestVisa/delGenTest.py", line 8, in <module>
        delayGen = rm.open_resource('ASRL13::INSTR', resource_pyclass=MessageBasedResource, write_termination='\r\n', read_termination='\r\n', delay=1)
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\highlevel.py", line 1641, in open_resource
        raise ValueError('%r is not a valid attribute for type %s' % (key, res.__class__.__name__))
    ValueError: 'delay' is not a valid attribute for type MessageBasedResource

Without these, the following script returns a **TIMEOUT** error for following code:

    delayGen = rm.open_resource('ASRL13::INSTR', resource_pyclass=MessageBasedResource, write_termination='\r\n', read_termination='\r\n')
    print(delayGen.query('*IDN?'))
    delayGen.close()

    Traceback (most recent call last):
      File "C:/Users/tpsadmin/Desktop/scopeTestVisa/delGenTest.py", line 12, in <module>
        print(delayGen.query('*IDN?'))
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\resources\messagebased.py", line 407, in query
        return self.read()
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\resources\messagebased.py", line 332, in read
        message = self.read_raw().decode(enco)
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\resources\messagebased.py", line 306, in read_raw
        chunk, status = self.visalib.read(self.session, size)
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\ctwrapper\functions.py", line 1582, in read
        ret = library.viRead(session, buffer, count, byref(return_count))
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
        raise errors.VisaIOError(ret_value)
    VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Could it be a problem with `resource_pyclass` or something just with `read()`?

hgrecco, 2015-07-29:
The `resource_pyclass` should not be necessary when viParseResourceEx works (as it does for ASRL). I have fixed a small regression that happened during the last changes. Please pull again and see if the problem persists in:
```
import visa
rm = visa.ResourceManager()
delayGen = rm.open_resource('ASRL13::INSTR')
```

mikekohlhoff, 2015-07-29:
Connection to the instrument (`SerialInstrument at ASRL13::INSTR`) works now without `resource_pyclass=MessageBasedResource`. Also, `send_end` works now in the constructor and all write operations return `(7L, <StatusCode.success: 0>)`. Any `read()` operations still produce

    Traceback (most recent call last):
      File "C:\Users\tpsadmin\Desktop\scopeTestVisa\delGenTest.py", line 16, in <module>
        print(delayGen.read())
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\resources\messagebased.py", line 332, in read
        message = self.read_raw().decode(enco)
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\resources\messagebased.py", line 306, in read_raw
        chunk, status = self.visalib.read(self.session, size)
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\ctwrapper\functions.py", line 1582, in read
        ret = library.viRead(session, buffer, count, byref(return_count))
      File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
        raise errors.VisaIOError(ret_value)
    VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

which is independent of termination characters.

hgrecco, 2015-07-29:
When you say the the write operations work, you mean that something (e.g. Frequency) is changing in the device after your command?

You should try a few things like:
- increasing the timeout
- changing read_termination and/or write_termination
- reading one byte at a time using `delayGen.read_raw(1)` to see what is comming.

(In all of them, I strongly suggest that you turn logging on as I showed you in the other issue and described in the FAQ)

mikekohlhoff, 2015-07-30:
Setting `inst.baud_rate` instead of `inst.write(<BAUD RATE>) to the value specified for the instrument solved the communication issue. Luser problem.
Thanks for the great support though!

closed 2015-07-30 by mikekohlhoff