pyvisa/pyvisa #170: ValueError: There is no class defined for (<InterfaceType.gpib: 1>, None)

MadhurashreeVN, 2015-08-03:
Hi,
I am trying to communicate with Agilent DC Source.  But, I'm getting value error. The code and the error message are as given below:

Code:
import pyvisa
rm = pyvisa.ResourceManager()
inst = rm.open_resource('GPIB::10::INSTR')

Error:
Traceback (most recent call last):
  File "<pyshell#11>", line 1, in <module>
    inst = rm.open_resource('GPIB::10::INSTR')
  File "C:\Python27\lib\site-packages\pyvisa-1.8.dev0-py2.7.egg\pyvisa\highlevel.py", line 1628, in open_resource
    raise ValueError('There is no class defined for %r' % ((info.interface_type, info.resource_class),))
ValueError: There is no class defined for (<InterfaceType.gpib: 1>, None)

I'm using NI GPIB cable to connect the device with the PC. I can query the device in NI MAX open test panel. 

Please help me. 

Thanks in Advance.

hgrecco, 2015-08-03:
I cannot reproduce. Can you update to the latest version?

hgrecco, 2015-08-24:
I am closing this. Please feel free to open another issue or reopen if you need further assistance.

closed 2015-08-24 by hgrecco