pyvisa/pyvisa #173: Can't read or query from serial device: Always Timeout

PySera, 2015-08-07:
Hello guys,

I am pretty new in this stuff but my task is to write a driver for my measurement device in the end.
I started with controlling a FUG power supply/current controller for a magnetic field and so far sending values and setting the current to e.g 0.1 A works fine. But as soon I want to recieve this value back or send a query I get the errormessage:
raise errors.visaIOError(ret.value)
pyvisa.errors.VisaIOError:VI_ERROR_TMO (-1073807339):timeout expired before operation completed.

Here is my code (I apologize for any bad style in advance..I try to get used to thsi python order)

import visa
import time
#import serial
from visa import constants
visa.log_to_screen
#Open 
rm=visa.ResourceManager()
rm.list_resources()
fug=rm.open_resource('COM1')

#Function to initialize the FUG
def func():
	#global ser
	#ser=serial.Serial()
	
	#ser.port=COMNUM-1
	#Time out
	
	fug.parity=None
	#fug.data_bits=8
	#fug.stop_bits=1
	fug.flow_control=None
	#fug.term_chars=ON
	fug.write_termination='\n'
	fug.chunk_size=30000
fug.data_bits=8
fug.stop_bits=constants.StopBits.one
#fug.flow_control=
fug.baud_rate=9600
#fug.term_chars_en=ON
fug.send_end='True'
fug.timeout=10000
fug.delay=0.2
fug.read_termination='\n'
	
fug.write("F1",time.sleep(1)) #Enable writing
fug.write("N2",time.sleep(1)) # Go to voltage channel
    
fug.write("U60",time.sleep(1)) # Set maximum voltage in device to 60V
fug.write("N1",time.sleep(1)) #Enable current writing
fug.write("I"+str(0.0),time.sleep(1))	#Set a  current value to the device
print (fug.resource_name)

#Sende die Abfrage und lese den Wert innerhalb des bestimmten Zeitintervalles aus
fug.write("?",time.sleep(0.4)) #Send the command to wait for action
fug.ask("I\n")

Closing i do still manually. I want to bring this to work first. 
Would be grad if anyone could help me with this or give me a hint where to look because i am out of ideas. Thanks.

PySera

hgrecco, 2015-08-08:
Hi,

A few things to mention regarding your code. When you do `fug.write("F1",time.sleep(1))`,
`time.sleep(1)` is called which pauses the code for 1 second. Its return value (`None`) is passed as second argument to `write`. This is not harmful because `write` happens to have `None` as default value, but is a weird way of writing it. I would recomend that you do the more usual.

```python
time.sleep(1)
fug.write("F1")
```

In your last line do not use `ask`, use `query` (`ask` is deprecated). Furthermore, why are you adding the termination character `\n`. In principle, it should not be necessary as it already provided as the `write_termination` value.

I would say:
1.- Check the manual of your device for the correct values for read and write termination, baud rate, data, bits, stop bits, etc
2.- If your device support an identification command (such as `*IDN?`) try that first. 
3.- Write the smallest script that fails.

Feel free to post your findings here, we are happy to help.

PySera, 2015-08-10:
Thanks a lot. I found the mistake. My precessor gave the wrong writing and reading termination.  :D
It was \r instead of \n. The former program has been written in LabView and somehow there the device got it right or the errror was corrected automatically.

hgrecco, 2015-08-24:
Happy to hear.

closed 2015-08-24 by hgrecco