pyvisa/pyvisa #174: Serial Connection Problem

weizenbierforever, 2015-08-11:
Hello guys,
I try to connect a simple USB to UART chip from Cypress via the Serial Connection in python3.4 64bit.
RX and TX are shorted, so I recieve a echo with Putty, that already works.

My code:
###########################
import visa
visa.log_to_screen()

rm = visa.ResourceManager()
cypresschip = rm.open_resource("COM5")

print(cypresschip.query("123"))
###########################

and I get the following Log and Error:

2015-08-11 01:51:32,086 - pyvisa - DEBUG - No user defined library files
2015-08-11 01:51:32,088 - pyvisa - DEBUG - Automatically found library files: ['C:\\windows\\system32\\visa32.dll', 'C:\\windows\\system32\\visa32.dll']
2015-08-11 01:51:32,097 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2015-08-11 01:51:32,097 - pyvisa - DEBUG - Created library wrapper for C:\windows\system32\visa32.dll
2015-08-11 01:51:32,136 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x00000000035B9AC8>',) -> 0
2015-08-11 01:51:32,136 - pyvisa - DEBUG - Created ResourceManager with session 4097
2015-08-11 01:51:32,137 - pyvisa - DEBUG - viParseRsrcEx(4097, 'COM5', 'c_ushort(4)', 'c_ushort(5)', <ctypes.c_char_Array_256 object at 0x00000000035B94C8>, <ctypes.c_char_Array_256 object at 0x00000000035B9BC8>, <ctypes.c_char_Array_256 object at 0x00000000035B9C48>) -> 0
2015-08-11 01:51:32,137 - pyvisa - DEBUG - COM5 - opening ...
2015-08-11 01:51:32,157 - pyvisa - DEBUG - viOpen(4097, 'COM5', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x00000000035B9C48>') -> 0
2015-08-11 01:51:32,158 - pyvisa - DEBUG - COM5 - is open with session 1
2015-08-11 01:51:32,159 - pyvisa - DEBUG - viWrite(1, b'123\r\n', 5, 'c_ulong(5)') -> -1073807298
Traceback (most recent call last):
  File "C:\Users\abc\Documents\PythonScripts\python3\Testfiles\SerialPort_test.py", line 17, in <module>
    print(cypresschip.query("123"))
  File "C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 401, in query
    self.write(message)
  File "C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "C:\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1867, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.
2015-08-11 01:51:32,168 - pyvisa - DEBUG - COM5 - closing
2015-08-11 01:51:32,169 - pyvisa - DEBUG - viDisableEvent(1, 1073709055, 65535) -> 1073676291
2015-08-11 01:51:32,169 - pyvisa - DEBUG - viDiscardEvents(1, 1073709055, 65535) -> 1073676292
2015-08-11 01:51:32,179 - pyvisa - DEBUG - viClose(1,) -> 0
2015-08-11 01:51:32,179 - pyvisa - DEBUG - COM5 - is closed
2015-08-11 01:51:32,184 - pyvisa - DEBUG - Closing ResourceManager (session: 4097)
2015-08-11 01:51:32,185 - pyvisa - DEBUG - viClose(4097,) -> 0

I also tried to connect directly with the National Measurement and Automation Explorer and got also a IO error.. Till I changed the buffer size for Transmit and Receive Buffer each from 0 to eg 20.
Is there a simmilar Setting in pyvisa? I searched a lot in the source files and did not find anything.
Or is problem related to something else? 

Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda3\python.exe
   Version:        3.4.1
   Compiler:       MSC v.1600 64 bit (AMD64)
   Bits:           64bit
   Build:          Jun 11 2014 17:27:11 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8.dev0

Backends:
   ni:
      Version: 1.8.dev0 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
      #2: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904

hgrecco, 2015-08-11:
As far as I know, the only attribute to control a buffer size in VISA are `VI_ATTR_RD_BUF_SIZE` and `VI_ATTR_WR_BUF_SIZE`. But they are only used by formatted I/O operations (`viPrintf`, `viScanf`, etc) which are **not** used by `query`. 

In the VISA documentation there is also a `VI_IO_IN_BUF` and `VI_IO_OUT_BUF ` which is said to control the Low-level I/O receive/transmit buffers using `viSetBuf`. But the documentation is very short about it. It might work if you add this to your code (untested):
```python
from visa import constants
rm.visalib.set_buffer(cypresschip.session, constants.VI_IO_IN_BUF, 20)
rm.visalib.set_buffer(cypresschip.session, constants.VI_IO_OUT_BUF, 20)
```

weizenbierforever, 2015-08-11:
This code works for me indeed! Thank you very much! I really appreciate your work!
Here the complete code that works on my system, in case somebody needs it.

```python
import visa
from visa import constants

#visa.log_to_screen()

rm = visa.ResourceManager()
cypresschip = rm.open_resource("COM5")

rm.visalib.set_buffer(cypresschip.session, constants.VI_IO_IN_BUF, 20)
rm.visalib.set_buffer(cypresschip.session, constants.VI_IO_OUT_BUF, 20)

print(cypresschip.query("123abcdefg"))

print(cypresschip.write("123abcdefg"))
print(cypresschip.read())
```

closed 2015-08-11 by weizenbierforever

hgrecco, 2015-08-11:
glad to help.

josecarlosmd, 2018-12-14:
Hello guys, 

I went through the same problem with SIGLENT SPD3303X. The solution was related to termination character. In my case, '\n' for reading and '\0' for writing: 

SG = rm.get_instrument('TCPIP0::192.168.1.119::INSTR', read_termination='\n',write_termination='\0')

Hope it helps

KKIII, 2019-01-16:
@josecarlosmd - How/where did you find the '\0' terminator for the SIGLENT SPD3303X? I've been struggling with this all day and your post was the solution!
Thanks much

josecarlosmd, 2019-01-16:
Sniffing with wireshark!

KKIII, 2019-01-16:
Awesome!

johndefiore, 2020-06-18:
@josecarlosmd I know this is an old thread, but I've been wrestling with the SPD3303X for two days and I finally stumbled on this thread and your solution.  Many thanks!!