pyvisa/pyvisa #175: installation problem

dschmag, 2015-08-18:
Hi,
As described on the website I entered 
pip install -U pyvisa
to install the module
What I get is:
C:\Projects\local\Python>pip install -U pyvisa
Collecting pyvisa
  Retrying (Retry(total=4, connect=None, read=None, redirect=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x0000000004538160>, 'Connection to pypi.python.org timed out. (connect timeout=15)')': /simple/pyvisa/
 ...
  Retrying (Retry(total=0, connect=None, read=None, redirect=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x0000000004538828>, 'Connection to pypi.python.org timed out. (connect timeout=15)')': /simple/pyvisa/
  Could not find any downloads that satisfy the requirement pyvisa
  No distributions at all found for pyvisa

Is it possible that the domain has changed?
I use a portable WinPython Version:
C:\ProgrammePortable\WinPython-64bit-3.4.3.2\python-3.4.3.amd64\python.exe

BR,
Daniel

hgrecco, 2015-08-18:
I have just tried this and it works fine for me. I guess is a problem with your connection or installation. Have you tried installing something else?

hgrecco, 2015-08-24:
I am closing this. Please feel free to open another issue or reopen if you need further assistance.

closed 2015-08-24 by hgrecco