pyvisa/pyvisa #176: backend VI in PYVISA

aebk2015, 2015-08-27:
I am new to Phyton and pyvisa.  I am  unable to run the simple code as below:
import visa
rm=visa.ResourceManager()
CONEX=rm.open_resource("COM10")
CONEX.timeout=2000
CONEX.write("1VE")
print (CONEX.read())

It returns the following error:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Anaconda\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 682, in runfile
    execfile(filename, namespace)
  File "C:\Anaconda\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 71, in execfile
    exec(compile(scripttext, filename, 'exec'), glob, loc)
  File "C:/Ali/PyhtonT/CONEXCC/CONEXCC-DRV.py", line 12, in <module>
    print (CONEX.read())
  File "C:\Anaconda\lib\site-packages\pyvisa\resources\messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "C:\Anaconda\lib\site-packages\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Anaconda\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Anaconda\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>> 

Here are the information about the Bckend and the Phyton

 Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda\python.exe
   Version:        2.7.9
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 18 2014 16:57:52 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136

hgrecco, 2015-08-27:
Are you sure you have the correct `read_termination` and `write_termination` for the device?

aebk2015, 2015-08-28:
Thanks for your advise.

Commands are executed as the command terminator CRLF (carriage-return
line-feed, ASCII 13 and ASCII 10) is received.  I modified the
open_resource as:
CONEX=rm.open_resource("COM10", read_termination='\r',
write_termination='\r')

but still it returns the same timeout error

Any other suggestion?

hgrecco, 2015-08-28:
If the instrument tells you to use CRLF as write termination, then you should use that. `write_termination='\r\n'` The same goes for the read_termination. What does the manual of the instrument tells you about the termination?

aebk2015, 2015-08-28:
I used write_termination='\r\n' and read_termination='\r\n' but still the
same error.

Here is the manual:
Commands are executed as the command terminator CRLF (carriage-return
line-feed, ASCII 13 and ASCII 10) is received. The controller will analyze
the received string. If the command is valid and its parameters are in the
specified range, it will be executed. Otherwise it will memorize an error.
After the execution of the command, all remaining characters in the input
string, if any, will be ignored. In particular, it is not possible to
concatenate several commands on a single string from the PC to the
CONEX-CC. Each command will handle properly the memorization of related
errors that can be accessed with the TE command. Please refer to the
command set in section 2.4 for details.

Thanks

hgrecco, 2015-08-28:
what is the purpose 1VE? does it changes anything in the device?  is it
supposed to return something?

aebk2015, 2015-08-28:
1VE is the ascii command sent to the device and it should return the
version of the device.

The device supports two-letter ASCII commands . The general command format
is:  nnAAxx

nn — Optional or required controller address.
 AA — Command name.
xx — Optional or required value or “?” to query current value.

 Most commands can be used to set a value (in that case the command name is
followed by the value “xx”) or to query the current value (in that case the
command name is followed by a “?”). When querying a value, the controller
responds with the command it received followed by the queried value. For
example, a 1VA10 sets the velocity of the controller #1 to 10 units/second.
A 1VA? sends the response 1VA10.

The controller address is 1 for this device
VE command doesn't need ? or value

Thanks,

hgrecco, 2015-08-28:
i would suggest that you try something tha produce a visible change in the
device. in that way you could see if you have problem during write or read

aebk2015, 2015-08-31:
Hello again,

I tried different syntax, different ascii command but still the same
problem.  I really appreciate your feedback.  Here is the link to the
manual for this device:
ftp://download.newport.com/MotionControl/Current/MotionControllers/CONEX/Manual/CONEX-CC/CONEX-CC%20-%20Controller%20Documentation.pdf

Thanks,

hgrecco, 2015-08-31:
It seems that the moving commands do not return a value. So my suggestions will be that you try to move the device with `1PR2.2` (for example). According to the state diagram, you might need to initialize the device before.

aebk2015, 2015-09-03:
this works when using with earlier version of pyvisa, how is it translated
to new version (1.8):

import visa

print visa.get_instruments_list()

A = visa.SerialInstrument("COM10",baud_rate=57600,data_bits=8,stop_bits=1)

A.write("1VE?\r\n")

print A.read()

Thanks

hgrecco, 2015-09-03:
Jusr give the same parameters to open_resource

closed 2015-10-06 by hgrecco