pyvisa/pyvisa #177: Urgent help please!!!!

rh111, 2015-08-31:
I am connecting a Hameg HM8143 power supply with agilent IO libraries 17. Can anyone tell me how to connect up python with aligent IO libraries so that I can talk to the power supply using python.

hgrecco, 2015-08-31:
What have you tried? What error are you getting?
 
As a suggestion, please look in the previous issues if something like this came up. Additionally, use a meaningful title for your issue.  "Urgent help please!!!" is not very indicative of the problem you are trying to solve.

rh111, 2015-09-01:
Can you kindly tell me if there is anyway I can use pyvisa to talk to Keysight IO libraries???

jonnojohnson, 2015-09-08:
@rh111 First of all are you able to talk to the power supply directly using IO Libraries? If you open the connection expert do you see the power supply in the list of instruments? If so are you able to send commands to it?
If all that is working the next suggestion I'd have is to run this at the command prompt and post your output here:
```
python -m visa info
````

hgrecco, 2015-10-06:
I am closing this for the time being. Feel free to reopen if needed.

closed 2015-10-06 by hgrecco