pyvisa/pyvisa #178: viRead()  time out error when the bytes number is big

serora, 2015-09-01:
Hi,
I am reading back the data using pyvisa, but a timing out error occurs when the given byte number is big.
The curious thing is that the maximum byte number is not constant, it may vary each time. I am not sure if this is caused by the termination issue when reading the chunk-size data back.

Here is the error 

  File "C:\Python27\lib\site-packages\pyvisa\visa.py", line 503, in ask
    return self.read()
  File "C:\Python27\lib\site-packages\pyvisa\visa.py", line 434, in read
    return self._strip_term_chars(self.read_raw())
  File "C:\Python27\lib\site-packages\pyvisa\visa.py", line 411, in read_raw
    chunk = vpp43.read(self.vi, self.chunk_size)
  File "C:\Python27\lib\site-packages\pyvisa\vpp43.py", line 841, in read
    visa_library().viRead(vi, buffer, count, byref(return_count))
  File "C:\Python27\lib\site-packages\pyvisa\vpp43.py", line 398, in check_status
    raise visa_exceptions.VisaIOError, status
VisaIOError: VI_ERROR_TMO: Timeout expired before operation completed.

serora, 2015-09-01:
Hi,

The issue occurs when the given byte number is more than 4444160 or 4366366, it varies a little each time.

Thanks

hgrecco, 2015-09-01:
It seems that you are using a **very** old version of PyVISA. I am not saying that this will be solved with the most recent version, but you should try to update first. We are not fixing old releases. Please bear in mind that you might have to change a few things in your program (albeit very few). There is a migration guide.

hgrecco, 2015-10-06:
I am closing this. Feel free to reopen if needed.

closed 2015-10-06 by hgrecco