pyvisa/pyvisa #179: Problem with import library

Eldiane, 2015-09-01:
Hello,

I have a problem when I'm trying to import Visa library error comes out:

>>> import pyvisa
Traceback (most recent call last):
  File "<interactive input>", line 1, in <module>
  File "C:\Python34\Lib\Generator\pyvisa\__init__.py", line 20, in <module>
    from . import compat
ImportError: cannot import name compat

I'm using Python 2.7.6 on win7 64x

phsdv, 2015-09-01:
Are you sure you are using the right version. 
Your Pyvisa error is from python 3.4 (File "C:\Python34\Lib\Generator\pyvisa__init__.py", line 20, in) and you are using Python 2.7.6? Is Pyvisa installed in 2.7.6 as well?

Eldiane, 2015-09-01:
I'm check that and download newest version and the error is:

>>>import visa
Traceback (most recent call last):
  File "<interactive input>", line 1, in <module>
  File "C:\Python27\lib\visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants
  File "C:\Python27\Lib\pyvisa-master\pyvisa\__init__.py", line 20, in <module>
    from . import compat
ImportError: cannot import name compat

so it's the same

Eldiane, 2015-09-01:
Im noted when import visa lib firstly comes out:

>>> import visa
Traceback (most recent call last):
  File "<interactive input>", line 1, in <module>
  File "C:\Python27\lib\visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants
  File "C:\Python27\Lib\pyvisa-master\pyvisa\__init__.py", line 45, in <module>
    from .highlevel import ResourceManager
  File "C:\Python27\Lib\pyvisa-master\pyvisa\highlevel.py", line 22, in <module>
    from . import constants
  File "C:\Python27\Lib\pyvisa-master\pyvisa\constants.py", line 29, in <module>
    class LongEnum(long, enum.Enum):
AttributeError: 'module' object has no attribute 'Enum'

And then:

>>>import visa
 Traceback (most recent call last):
 File "", line 1, in 
 File "C:\Python27\lib\visa.py", line 16, in 
 from pyvisa import logger, version, log_to_screen, constants
 File "C:\Python27\Lib\pyvisa-master\pyvisa__init__.py", line 20, in 
 from . import compat
 ImportError: cannot import name compat

maybe its something wrong wiht enum library?

hgrecco, 2015-09-01:
I doubt that this is a problem with pyvisa. It is tested against multiple python version (See [here](https://travis-ci.org/hgrecco/pyvisa) ) My guess is that you have installed incorrectly or there is something wrong with your Python installation. How was pyvisa installed in your system?

(By the way, you should really update your Python 2.7.6 to 2.7.10.  There are a few important bugs that were fixed)

Eldiane, 2015-09-01:
Ok, i will try to reinstall python and pyvisa, but tommorow becouse I end my work now and I'm going home.

About update python to 2.7.10 version, it's compicated becouse in work I must use 2.7.6 so it's imposible to update :(

I'll talk tommorow is this solve problem
So see you tommorow! :)

hgrecco, 2015-09-01:
At work you might try to reinstall pip, and then install pyvisa again.

Eldiane, 2015-09-02:
Hello hgrecco!

Thanks for advice to reinstall pip, it solve problem.
Sorry for trouble and thanks for help! :)

hgrecco, 2015-09-02:
I guess that your pip version was having some trouble with dependencies. Pip has changed a lot recently, it is a good idea to keep it up to date. Glad that you made it work. Good luck and have fun with PyVISA.

closed 2015-09-02 by hgrecco