pyvisa/pyvisa #180: 1.8 regression: pyvisa.errors.VisaIOError: VI_ERROR_TMO

greysAcademicCode, 2015-09-03:
Hello,
I have the following code:
```
import visa
rm = visa.ResourceManager('/usr/local/vxipnp/linux/lib64/libvisa.so')
inst = rm.open_resource('TCPIP::192.168.1.204::5025::SOCKET')
print(inst.query("*IDN?"))
inst.close()
```
I am using Linux.

The above code works fine with the 1.7 release, however with 1.8 I'm getting the following error:
```
File "~/src/visaTest.py", line 4, in <module>
  print(inst.query("*IDN?"))
File "/usr/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 407, in query
  return self.read()
File "/usr/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 332, in read
  message = self.read_raw().decode(enco)
File "/usr/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
  chunk, status = self.visalib.read(self.session, size)
File "/usr/lib/python2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1582, in read
  ret = library.viRead(session, buffer, count, byref(return_count))
File "/usr/lib/python2.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
  raise errors.VisaIOError(ret_value)

pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```
I get the same result with both python2 and 3.

hgrecco, 2015-09-03:
I cannot remember any change in that area. Can you do a bisect to find out
the guilty commit?

hgrecco, 2015-10-06:
Any news on this?

greysAcademicCode, 2015-10-07:
I haven't been able to spend the time to track down the offending commit. Can you at least reproduce the error?

mittal10, 2015-10-25:
I see the same error

jpl777, 2016-01-15:
I am getting the same error message while trying to connect to my instrument. 
Saw another thread with the same error (Support for TCPIP SOCKET #38, tested in June 2015). Perhaps it will narrow the timeframe of commits to search through.

MatthieuDartiailh, 2018-02-25:
Closing given that the issue is old, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh