pyvisa/pyvisa #181: list_resources() doesn't return COM

aebk2015, 2015-09-04:
Hello,

I have USB devices which show up under COM ports in the device manager.
I can initialize the device by:
rm=visa.ResourceManager()
device1=rm.open_resource("COM1", baud_rate=921600)
device2=rm.open_resource("COM2", baud_rate=57600)

but I dont want to hard code the COM numbers (COM1 and COM2) in the open_resource() and therefore I am trying to use list_resources() to get the list of all devices, save the device key and parse it in the open_resource().  However, the  list_resources() doesn't return COM1 and COM2 but, it returns ASRL1::INSTR and ASRL2::INSTR.  If I hard code ASRL1::INSTR in the open_resources() it throws an error.

hgrecco, 2015-09-06:
As a first note, if you are using the NI backend the listed resources and
how each of them is mapped to the underlying resource is completely under
the control of the NI-VISA library and therefore outside the control of
PyVISA.

In the VISA jargon, ASRLN::INSTR is a serial instrument in port N. In other
words, they are mapped to the COM ports. So what you are getting in the
list is fine. What is weird is that you are not able to open the ports. Can
you provide more information about your installation and the error message
that you are getting.

aebk2015, 2015-09-06:
Thanks Hernan, I used the wrong syntax, now the error is gone and I am able
to open the devices.  However, I am facing with another problem which is
more of python than pyvisa.  I already posted the problem on stackoverflow
and I appreciate it if you could give me your insight.  The tread is here:
http://stackoverflow.com/questions/32407942/creating-subclass-when-using-pyvisa

Thanks

hgrecco, 2015-09-06:
This is wrong:

super(Newport,self).__init__()

should be:

super(CONEX,self).__init__()

A few general (style) comments:
1. Do not create multiple ResourceManager Instances. Share one across all
your code.
2. Use lower case names for the attributes.
3. Use lower case names with underscores for functions.
4. Do not add your own termination chars. Let PyVISA do that for you

By the way, take a look at Lantz lantz.readthedocs.org We are building up
this community project to make instrumentation easier in the same line of
your classes.

aebk2015, 2015-09-08:
Thanks, it worked after I changed the syntax.  Now I am having difficulties
opening different instrument which has the same interface as the previous
one.  My suspecision is that I need to specify the flow control and
stop_bits.  It looks like stop_bits is not defined in new pyvisa.  Can you
let me know the equivalent?  Also who I specify the flow control?

Thanks

aebk2015, 2015-09-08:
Hi Hernan,

Please ignore the previous question.  I have made my program like this but
I am stuck and appreciate it if you could please help me.  As I mentioned
previously, I have parent class called MotionControllers and three sub
classes called CONEX, SMC and ESP.  The problem I have now is how to use
the DeviceKey (obtained from the return of list_resources()) and use it in
the sub classes. In the current version (please see attached), it returned:

 AttributeError: type object 'Newport' has no attribute 'DeviceKey'

import visa

class MotionControllers(object):

def __init__(self,DeviceKey):

self.rm=visa.ResourceManager()

self.DeviceKey=DeviceKey

def OpenConnectedDevices(self):

self.ListOfDevices=self.rm.list_resources()

for DeviceKey in self.ListOfDevices:

self.Base=self.rm.open_resource(DeviceKey)

def WriteRead(self, command):

self.command=command

self.Base.write(self.command)

return (self.Base.read())

def Read(self):

print (self.Base.read())

def Send(self, command):

self.command=command

self.Base.write(self.command)

def GetFirmwareVersion(self):

return self.WriteRead("1VE")

class CONEX(MotionControllers):

def __init__(self):

super(CONEX,self).__init__()

self.Base=self.rm.open_resource(MotionControllers.DeviceKey,
baud_rate=921600, timeout=2000, data_bits=8, write_termination='\r\n')

class SMC(MotionControllers):

def __init__(self):

super(SMC,self).__init__()

self.Base=self.rm.open_resource(MotionControllers.DeviceKey,
baud_rate=57600, timeout=2000, data_bits=8, write_termination='\r\n')

class ESP(MotionControllers):

def __init__(self):

super(ESP,self).__init__()

self.Base=self.rm.open_resource(MotionControllers.DeviceKey,
baud_rate=921600, timeout=2000, data_bits=8, write_termination='\r\n')

my_CONEX=CONEX()

my_SMC=SMC()

my_ESP=ESP()

print my_CONEX.GetFirmwareVersion()

print my_SMC.GetFirmwareVersion()

print my_ESP.GetFirmwareVersion()

BTW, the reason I am not following Python convention for defining the
function and variable names is because I want to create function with the
same API that are available from the controller GUI.

# -*- coding: utf-8 -*-
"""
Created on Tue Sep 08 13:48:45 2015

@author: Ali.Eftekhari
"""

import visa

class MotionControllers(object):
    def __init__(self,DeviceKey):
        self.rm=visa.ResourceManager()
        self.DeviceKey=DeviceKey

    def OpenConnectedDevices(self):
        self.ListOfDevices=self.rm.list_resources()
        for DeviceKey in self.ListOfDevices:
            self.Base=self.rm.open_resource(DeviceKey)
             
    def WriteRead(self, command):
        self.command=command
        self.Base.write(self.command)
        return (self.Base.read())
        
    def Read(self):
        print (self.Base.read())        
        
    def Send(self, command):
        self.command=command
        self.Base.write(self.command)         
            
    def GetFirmwareVersion(self):
        return self.WriteRead("1VE")

class CONEX(MotionControllers):
    def __init__(self):
        super(CONEX,self).__init__()
        self.Base=self.rm.open_resource(MotionControllers.DeviceKey, baud_rate=921600, timeout=2000, data_bits=8, write_termination='\r\n')

class SMC(MotionControllers):
    def __init__(self):
        super(SMC,self).__init__()
        self.Base=self.rm.open_resource(MotionControllers.DeviceKey, baud_rate=57600, timeout=2000, data_bits=8, write_termination='\r\n')
        
class ESP(MotionControllers):
    def __init__(self):
        super(ESP,self).__init__()
        self.Base=self.rm.open_resource(MotionControllers.DeviceKey, baud_rate=921600, timeout=2000, data_bits=8, write_termination='\r\n')

             
my_CONEX=CONEX()
my_SMC=SMC()
my_ESP=ESP()

print my_CONEX.GetFirmwareVersion()
print my_SMC.GetFirmwareVersion()
print my_ESP.GetFirmwareVersion()

closed 2015-10-06 by hgrecco