pyvisa/pyvisa #184: how to get the visa address for all devices from rm.list_resources()

aebk2015, 2015-09-16:
the following returns only the visa address of the devices connected to COM ports but not USB or  other devices.  For example list.resource() doesn't return VISA address if the devices are connected through USB and therefore showing up under "Universal Serial Bus controllers" in the device manager.  I also have devices that are connected through jungo drivers and list.resource() doesn't return the address for these devices as well.  Please let me know if there is any attribute I should use inside list.resource()
Thanks

import visa
rm=visa.ResourceManager()
print rm.list_resources()

hgrecco, 2015-09-18:
Which backend are you using?

aebk2015, 2015-09-18:
It is visa32.dll

Thanks

hgrecco, 2015-09-18:
PyVISA just returns what VISA provides. There is no filtering done.

aebk2015, 2015-09-18:
so what would i have to do so I can get usb devices.  Sorry if my question
is premature: i am new to python and pyvisa

hgrecco, 2015-09-18:
What kind of device is it?

aebk2015, 2015-09-18:
it is a power meter from Newport.  The model number is 843-R-USB

when i install the driver for the meter it shows up under jongo driver in
device manager

Thanks

hgrecco, 2015-09-18:
I do not think that you can control that power meter with VISA. It seems to have a proprietary driver

aebk2015, 2015-09-18:
My understanding was that Pyvisa should be in principle capable of
controlling any instruments whether the interface is gpib serial internet
usb.  Is it possible to modify pyvisa to control usb instruments such as
the one I mentioned.

Please note that not being able to control usb device is not only the case
for jongo driver.  I have Newport 1918- R meter, I installed NI Visa driver
but still Pyvisa is not able to return its visa address

Thanks

hgrecco, 2015-09-18:
The first question is: does ti appear in NI-MAX? If it does not show up there, then it will not show up in PyVISA. 

Going back to your question. With VISA you can send binary data to a USB port and therefore to a device. But you need to know the protocol, you need to know how the device expects the commands and how it returns the data. Without that you have to use their driver.

aebk2015, 2015-09-21:
the instrument does show up in the NI-MAX.  In fact, now I ma not using the
Jongo driver but NI-VISA drivers but still PYVISA doesn't show the
instrument.  Please see the screen shot

hgrecco, 2015-09-22:
The screen shot was not attached.

aebk2015, 2015-09-22:
[image]

Please see the screen-shot

hgrecco, 2015-09-22:
I see now. It is not an instrument, is just a RAW port. By default, only INSTR resource names are shown (See [here](http://pyvisa.readthedocs.org/en/stable/api/resourcemanager.html#pyvisa.highlevel.ResourceManager.list_resources) )  But you can easily ask for everything by doing
```python
rm.list_resources('?*')
```

aebk2015, 2015-09-22:
thanks, that worked.

Could you please look at the open issue 183 as well?

Basically, I want to implement "if" or "while" statement before self.Open=self.rm.open_resource(self.key) making sure it opens the same instrument only one time and so moves to the next available devices in self.list_of_devices but I couldn't find is_open option for the list_resources.

hgrecco, 2015-09-22:
VISA does not keep track or this, so you need to keep that list yourself.

closed 2015-10-06 by hgrecco