pyvisa/pyvisa #185: Setting DTR before opening resource

jdb78, 2015-09-19:
It seems as if one cannot set DTR=False before opening a resource.
However, pyserial supports this feature: http://stackoverflow.com/questions/15460865/disable-dtr-in-pyserial-from-code

Due to other advantages of pyvisa I would like to stick with it. Is there a workaround or are there plans to add this feature?

hgrecco, 2015-09-19:
As far as I know, in VISA you can only set an attribute (like DTR) after opening the resource. 

What is your use case? The one described in the stackoverflow thread?

jdb78, 2015-09-19:
Yes. It is the case described in the stackoverflow thread. I can prevent a restart of the arduino by using pyserial with setDTR(False). So far, I have not managed to obtain the same result with pyvisa.

hgrecco, 2015-09-19:
In principle you can't. Because VISA (not PyVISA) requires that the resource is open to set an attribute. 

However, there is an special flag (VI_LOAD_CONFIG ) to open the resource that only works with Serial INSTR and might help you.

In PyVISA, you just give use the access mode parameter to `open`:
```python
inst = rm.open(resource_name, access_mode=4)
```

The VISA docs state:
> When you open a Serial port, the VISA specification defines the default communication settings to be 9600 baud, 8 data bits, 1 stop bit, no parity, and no flow control. If you have configured the settings to a different value in the NI-VISA configuration utility (MAX on Windows, visaconf on UNIX), then you must pass the value VI_LOAD_CONFIG (4) as the AccessMode parameter to viOpen(). This parameter will cause the configured settings to be used; otherwise, if the AccessMode is 0 or VI_NULL, the default settings will be used.

So maybe this is the way to do it.

MatthieuDartiailh, 2018-02-25:
Closing given that the issue is old, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh