pyvisa/pyvisa #186: Problem with inst.query over TCPIP

Zaknife, 2015-09-24:
Hello,

I'm experiencing problems with communication with external device over TCPIP protocol. After installation of py backends I think I can connect to the device,but I cannot query any information. Below you may find the Visa INFO, error message and the code:

Code:
import visa
visa.log_to_screen()
rm=visa.ResourceManager('@py')
triton2=rm.open_resource("TCPIP0::triton::33576::SOCKET")
print(triton2.query("*IDN?\n"))

error log:

2015-09-24 12:56:04,546 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2015-09-24 12:56:04,547 - pyvisa - DEBUG - USBSession and USBRawSession were not imported No module named 'usb'.
2015-09-24 12:56:04,552 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name 'TCPIPSession'.
2015-09-24 12:56:04,553 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2015-09-24 12:56:04,553 - pyvisa - DEBUG - Created library wrapper for unset
2015-09-24 12:56:04,553 - pyvisa - DEBUG - Created ResourceManager with session 1502787
2015-09-24 12:56:04,553 - pyvisa - DEBUG - TCPIP0::triton::33576::SOCKET - opening ...
2015-09-24 12:56:04,560 - pyvisa - DEBUG - TCPIP0::triton::33576::SOCKET - is open with session 1441712
2015-09-24 12:56:04,562 - pyvisa - DEBUG - TCPIP0::triton::33576::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
Traceback (most recent call last):
  File "C:\Users\stefanowicz\Desktop\ProgramGMazur\Triton.py", line 10, in <module>
    print(triton2.query("*IDN?\n"))
  File "C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 407, in query
    return self.read()
  File "C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Anaconda3\lib\site-packages\pyvisa-py\highlevel.py", line 253, in read
    return self.sessions[session].read(count)
  File "C:\Anaconda3\lib\site-packages\pyvisa-py\tcpip.py", line 392, in read
    last = read_fun(chunk_length)
2015-09-24 12:56:04,579 - pyvisa - DEBUG - TCPIP0::triton::33576::SOCKET - closing
2015-09-24 12:56:04,579 - pyvisa - DEBUG - TCPIP0::triton::33576::SOCKET - is closed
2015-09-24 12:56:04,593 - pyvisa - DEBUG - Closing ResourceManager (session: 1502787)
Visa info:

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
   py:
      Version: 0.2
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named 'serial'
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

hgrecco, 2015-10-06:
Sorry to ask but I do not know to which instrument are you talking. Are you sure your instruments is SOCKET and not INSTR?

MatthieuDartiailh, 2018-02-25:
Closing given that the issue is old, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh