pyvisa/pyvisa #187: Anritsu power meter (ML2438A) with pyVISA 1.8

jrmagers, 2015-10-04:
I'm trying to use an Anritsu power meter (ML2438A) with pyVISA 1.8 and I have not been able to successfully make a power reading. The write() functions are successful, but receive a timeout error on the query_ascii_values() functions. Does anyone know how to get the data out of this power meter?

import visa

pm_resource = 'GPIB0::13::INSTR'
rm = visa.ResourceManager()
pm = rm.open_resource(pm_resource)
pm.write('*RST')       # reset
pm.write('CFFRQ A,10GHZ')
power_reading = pm.query_ascii_values('TR1 1')
pm.close()

MatthieuDartiailh, 2015-10-04:
Can you read anything back from the instrument ? (such as the output of *IDN?, or equivalent)
You should set  the read_termination character (instr.read_termination = '\r')? Most likely the timeout occur because the no termination character is detected. Going quickly over the manual you can try '\r', '\n' or '\r\n', the manual being not extremely clear.
It might also make sense to set the write_termination to '\n' (even if it seems that the GPIB board emitting EOS should be enough).
If everything fails try reading the response to *IDN? one char at a time using read_raw(1), but first make sure that command are correctly executed on write (setting the write_termination should take care of that).

jrmagers, 2015-10-04:
Hi Matthieu,

Thank you for getting back to me.  I can read back from the instrument:

pm.query('*IDN?')

returns

u'ANRITSU,ML2438A,98200020,3.10'

I tried the different variations of read and write terminations with no success.

pm.write('*IDN?')
print pm.read_raw(1)

returns

ANRITSU,ML2438A,98200020,3.10

Here's the traceback from query_ascii_results()

pm.query_ascii_values('TR1 1')

Traceback (most recent call last):

  File "<ipython-input-44-a12be870a1b2>", line 1, in <module>
    pm.query_ascii_values('TR1 1')

  File "C:\Anaconda\lib\site-packages\pyvisa\resources\messagebased.py", line 455, in query_ascii_values
    block = self.read()

  File "C:\Anaconda\lib\site-packages\pyvisa\resources\messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)

  File "C:\Anaconda\lib\site-packages\pyvisa\resources\messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)

  File "C:\Anaconda\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))

  File "C:\Anaconda\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

For what it's worth, I used to use this command in MATLAB successfully (although this was several years ago):

power = fscanf(pm,'%f');

jrmagers, 2015-10-05:
Ok, it turns out that if the power sensor is not connected to the power meter, then I get the timeout. This code works fine:

import visa
import numpy as np

pm_resource = 'GPIB0::13::INSTR'
rm = visa.ResourceManager()

pm = rm.open_resource(pm_resource)
pm.query('*IDN?')

pm.write('CFFRQ A,25GHZ')
power = pm.query_ascii_values('ON 1,10',container=np.array)
print power.mean()

pm.close()

MatthieuDartiailh, 2015-10-05:
Great you figured it out !

closed 2015-10-06 by hgrecco