pyvisa/pyvisa #188: how to check ResourceManager status?

aebk2015, 2015-10-06:
Hi, I am opening an instrument as below:
import visa
rm=visa.ResourceManager()
device_key=str(input("What is the COM port"))
connect=rm.open_resource(device_key)

However, the program prompts me to provide the COM port number every time I send read/write.  Is there a method that checks the status of the ResourceManager to see if it is already open?

hgrecco, 2015-10-06:
Just ask for the comport once, open it and then use it for many reads and writes.

aebk2015, 2015-10-06:
I should have written the complete code.  I am developing this in a class, basically:

import visa

class Main(object):	
    def __init__(self):
        self.rm=visa.ResourceManager()
		device_key=str(("What is the COM? "))
        self.connect=self.rm.open_resource(device_key)
		
	def send(self, command):
        self.command=command
        self.connect.write(self.command)

	def get_error(self):
        return self.write_read("FE?")

	def get_position(self):
        return self.write_read("TP?")		
		
my_class=Main()
print my_class.get_error()
print my_class.get_position()

If I send print my_class.get_error() I will be promoted to provide the COMP port.  Without closing the port, if I send print my_class.get_position(), I will be promoted to provide the COM port again.

MatthieuDartiailh, 2018-02-25:
Closing given that the issue is old, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh