pyvisa/pyvisa #190: read_raw fails (VI_ERROR_TMO (-1073807339): Timeout expired before operation completed)

PhaseShifter, 2015-10-09:
I'm trying to communicate with a device on a serial port that doesn't return a line ending. I've initialized it like this:

>K3005 = rm.open_resource('ASRL3::INSTR')
>K3005.query_delay = 1.0
>K3005.send_end = False
>K3005.end_input = visa.constants.SerialTermination.none
>K3005.write_termination = None
>K3005.read_termination = None

I can send data to it, and it returns data as well:
>K3005.write("*IDN?")
>print K3005.bytes_in_buffer
>> 19

But when I try to get the data by calling 
>data = K3005.read_raw()
I get 
> VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

By bypassing the read_raw() function like this:

>K3005.write("*IDN?")
>time.sleep(0.5)
>data, status = K3005.visalib.read(K3005.session, K3005.bytes_in_buffer)
>print data
>> VELLEMANPS3005DV2.0

I can still get the data without any error.

hgrecco, 2015-10-09:
PyVISA works on the assumption that there is a way to determine the end of the message (such as the termination character). What does the manufacturer of the instrument tells you about it? Are messages fixed in size? or you should wait a certain time and then grab everything which is in the buffer?

PhaseShifter, 2015-10-09:
Just wait a certain time and then fetch the data from the buffer (hence the sleep). I thought that setting the query_delay would take care of the delay. Anyway, the read_raw seems to fetch the data from the buffer because it is empty afterwards.

FYI it's a powersupply (Korad KA3005P), from which the protocol has been reverse engineered. But its commands are quite SCPI like.
http://sigrok.org/wiki/Velleman_PS3005D

hgrecco, 2015-10-09:
Then you would need to create a function for it. It would be find for me to add this functionality to pyvisa in MessageBasedResource. Something like:
```
def read_all(self):
    self.visalib.read(self.session, self.bytes_in_buffer)
```
or similar

AdriaanRol, 2016-01-12:
I have the exact same issue. I also have an instrument that does not return an end of line character. 
I came up with the following workaround. 
```python 
        # a workaround for a timeout error in the pyvsia read_raw() function
        with(self.visa_handle.ignore_warning(visa.constants.VI_SUCCESS_MAX_CNT)):
            mes = self.visa_handle.visalib.read(
                self.visa_handle.session, message_len)
```

I think the behaviour of the read_raw() function is confusing, especially as the argument you give it is size which I think would specify exactly the amount of expected bytes. 

Is there a particular reason against having the function return x-values if it has pulled x-values from the buffer, even if there is no end of line char?

danj7, 2016-02-10:
Hi @AdriaanRol , I was wondering how you implemented the solution you wrote there? Did you modify the messagebased.py file or did you put that in your own file?
I'm having the same error message from a function that calls other functions. The other functions work well by themselves but combined they end up with a read_raw() error.

TimTaste, 2017-02-27:
Hi,
there are several instruments on the market (especially serial interface based ones) not having a termination character. Examples are the popular climate chambers of Weiss Technik. Their protocols base on fixed length of messages. These types of messages can't be handled with current implementation of the read_raw-methode due to the time out error issue described above. 
So I would prefer to have another read_raw_bytes(num_bytes) -methode similar to  @AdriaanRol's proposal.

ppxlb, 2017-11-21:
@AdriaanRol (or even @danj7) I'm also curious how that's implemented to the messagebased.py read_raw() I can see there's a lot more going on and just editing is likely to open up a whole can of worms!

MatthieuDartiailh, 2018-02-25:
`read_bytes` has been implemented in PyVISA 1.9. Closing, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh