pyvisa/pyvisa #192: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

mittal10, 2015-10-25:
Hi I am getting timeout error everytime when I am just trying to execute the example (CALCulate2:TRANsform:HISTogram:DATA?) given in the manual. 

scpi> *CLS
scpi> CONF:FREQ 1.0E6,.001
scpi> SAMP:COUN 1000
scpi> CALC2:TRAN:HIST:RANG:AUTO ON
scpi> CALC2:TRAN:HIST:RANG:AUTO:COUN 100
scpi> CALC2:TRAN:HIST:POIN 128 
scpi> CALC2:TRAN:HIST:STAT ON
scpi> INIT
scpi> *WAI
scpi> CALC2:TRAN:HIST:DATA?
Traceback (most recent call last):
  File "shell.py", line 40, in <module>
    answer = dev.ask(cmd).strip()
  File "/Library/Python/2.7/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/Library/Python/2.7/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/Library/Python/2.7/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/Library/Python/2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "/Library/Python/2.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

VinoVino, 2016-08-12:
I'm getting the same error when I run: 
>>> my_instrument=rm.open_resource('ASRL1::INSTR')
>>> my_instrument.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Python/2.7/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/Library/Python/2.7/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/Library/Python/2.7/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/Library/Python/2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1582, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "/Library/Python/2.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2018-02-25:
This looks like issues with termination characters. Closing, feel free to re-open if there is more to discuss.

closed 2018-02-25 by MatthieuDartiailh