pyvisa/pyvisa #194: Improved backend handling

herm, 2015-11-03:
Pyvisa should not depend on the user to select the correct backend but instead choose the best available backend automatically. For example when the NI library is not found but pyvisa-py is, automatic switching to the available library should be done (unless some library is explicitly requested by the user). 

Additionally there should be a way to select the library without modifying the program. For example there could be an environment variable PYVISA_LIBRARY that is used to select the library. Many other programs (including python itself) do this.

hgrecco, 2015-11-05:
This was an option in the past but was removed during the transition. I would like to put it back but maybe within a larger overhaul regarding the configuration. Maybe we can discuss here what to do and how to it and then we try to build up a comprehensive solution. A few ideas:

**Put .pyvisarc in the right place**
For historical reasons, we are storing .pyvisarc in the wrong place. I think we should use  appropriate path based on the OS. I am proposing to put files in the app folder using the same code that is used in [click.get_app_dir](http://click.pocoo.org/5/api/#click.get_app_dir) (with *roaming=True* and maybe *force_posix=True*)

We should do it backward compatible so that the current file is still readable.

**Provide alias configurations**
It might be useful to provide some alias configuration. For example, you might want to have multiple pyvisa-sim configurations with different simulated instruments. Maybe is a good idea to put have a way to define this easily.

**A command line app for configuration**
We should provide a `config` subcommand to `python -m visa` way similar to `git config` to write this file content.

**One ResourceManager, multiple VisaLibrary**
Right now you cannot open resources from different libraries. It would be very useful to have this and in this way be able to combine real and simulated instruments.

herm, 2015-11-11:
Some comments on this:
Putting the .pyvisarc file in a standard place is a good idea. However using this file one can only select the NI library location but not the backend. This should be fixed, too. If the config file is well documented having a command line tool is not required in my opinion.

Having one ResourceManager for multiple libraries might be useful for other applications which only support a single ResourceManager (like python-ivi).

hgrecco, 2015-11-20:
Agreed.